LIFsCode/ELOC-3.0 #4: Add support for ULP and analog microphone

LIFsCode, 2023-04-20:
- [ ] Merge code from https://github.com/asanka-code/ElocNG-Firmware into Firmware
- [ ] Make analog thresholds configureable
- [ ] Add support for external analog filters (config via i2c poti?) TBD

EDsteve, 2023-05-07:
@LIFsCode 
I have done some digging and also spoke with Pille about this topic. It all seems not so easy as we thought. As it usually is :)
Pille does not like the idea of a second MCU on the board. But i just can't find a optimal solution to get the filters done without an MCU. Digital Potis are the main trouble makers and i can't find one that suits our purpose:

To set the filters correctly for elephant rumbles is no problem with hardware filters. But I would like the ELOC to be versatile. So the threshold detection should also work for other animals like birds. And for that a poti with a much higher range is needed.

Anyhow. the MCU solution seems to be superior in almost all points. The question is if the extra firmware effort is worth it? How much trouble are we talking about here?

Here a quick Pro and Con list
[image]

Couldn't find a way to format a table with colors in github. Thats way the external list. Sent you a [link](https://docs.google.com/spreadsheets/d/1Aaebn5-RBtB5sbzFFzh_BJHp-WBWmbmFLtwrL9ZgLnk/edit?usp=sharing) to it already. Edit the list as you wish.

EDsteve, 2024-04-06:
Closing this for now. And can be reopened when we have the ret of the device working

closed 2024-04-06 by EDsteve