LIFsCode/ELOC-3.0 #8: Update i2s legacy driver to current version

LIFsCode, 2023-04-23:
Currently i2s.h is used, this file is deprecated:

` This file is for the backward compatible to the deprecated I2S APIs,
 The deprecated APIs will no longer supported in the future
 Please refer to "driver/i2s_std.h", "driver/i2s_pdm.h" and ""driver/i2s_tdm.h"" for the latest I2S driver
 Note that only one set of I2S APIs is allowed to be used at the same time`

EDsteve, 2023-04-23:
Good find. That might solve some problems i got with some I2S microphones i used for testing. Looking forward to see test this new driver.

OOHehir, 2024-01-06:
This should help in solving this [issue](https://github.com/LIFsCode/ELOC-3.0/issues/18).

OOHehir, 2025-01-13:
@EDsteve I just pushed a branch to try updating the platofrm espressif32 SDK to 6.9.0. Would you mind try building it on your system? Probably need a full clean first.

@EDsteve @LIFsCode Unfortunately it seems that this latest SDK, [6.9.0](https://github.com/platformio/platform-espressif32/releases), doesn't have the updated IS2 driver, they've stuck with the older version.

EDsteve, 2025-01-13:
@OOHehir I have tried your sdk_update branch and didn't run into any problems. So far so good :)

OOHehir, 2025-01-14:
@EDsteve @LIFsCode Good news, if there's no objections I'll update master/main with this espressif32 version.

EDsteve, 2025-01-15:
Thanks. Fine with me. Too bad they didn't update the I2S driver.
Besides that i guess it fixes some bugs but nothing major we should be a aware of?

OOHehir, 2025-01-16:
@EDsteve Mostly bug fixes, details are here https://github.com/platformio/platform-espressif32/releases