LIFsCode/ELOC-3.0 #10: Define interface to the Bluetooth app

LIFsCode, 2023-04-26:
Currently the Bluetooth interface is not really clear for me.

I have written down what I found out from the code here: https://github.com/LIFsCode/ELOC-3.0/wiki 

Can you please specifiy

- [ ] What commands do exist in the app and what is there intention?
- [ ] What is the expected format of the fields in the response from the ELOC device?
- [ ] What is the intentional meaning in the fields of the ELOC response?

Currently the ELOC response looks somewhat like this:

```
statusupdate
Time:  1970-01-01 07:00:30
Ranger:  _@b$_
!0!ELOC_NONAME
!1!eloc350SDIOHighPower06Apr2023A
!2!20.00 v FULL
!3!
!4!0.04 h
!5!0.00 h
!6!0.00 h
!7!0
!8!on
!9!0
!10!0
!11!0.00
!12!ns
!13!11
!14!unknown
!15!99 m
!16!
```

Field desciption as depicted from firmware:

| Field | Description | Questions |
| ------------- | ------------- | ------------- |
|  0 | Device Name (BT Name) | |
|  1 | Firmware Version | |
|  2 | Voltage + Text | |
|  3 | Location | What is location? just a name like Device Name? GPS location?  |
|  4 | Timestamp | |
|  5 | Total Recording time | |
|  6 | Session Recording time | |
|  7 | Recording state (0/1) | |
|  8 | BT during Recording ("on"/"off") | |
|  9 | sample Rate | |
| 10 |  seconds per file | |
| 11 | free space (GB) | |
| 12 | ns  | (What is mic type?) |
| 13 | Mic bit shift|? (this seems to be some kind of gain... why not use gain?)  |
| 14 |  Location Code | This seems to be the GPS location. What format? |
| 15 | Location Accuracy | |
| 16 | Session Identifier | |

josephx86, 2023-04-27:
Hi all. Having JSON as output format to the BT stream would make it easy to understand the data from a board without the need of a look up table of sorts. 

More helpful would be a consistently named interface/API to get information to and from a board. I would suggest a getX/setX format for the interface. For example, getStatus, getSettings, etc

Also, the commands sent via BT to the board need to be better named. For example, to start recording, the current command is "setGPS", which does not make it immediately clear that the command is for recording.

EDsteve, 2023-04-28:
About the commands:
I will make a list with all commands as they are used right now to get an overview. Something like this. Will complete it in a nicer table tomorrow.

#settings#bton -> Keeps bluetooth ON while recording
#settings#btoff -> Turns BT OFF while recording 
#settings#ELOCtestsetname -> Changes the device name to ELOCtest
#settings#AnalogXXsettype -> Changes microphone type to AnalogXX

LIFsCode, 2023-05-02:
@josephx86 : I agree JSON would be a suitable format for data exchange. I would prefer having the same format for 
- Bluetooth data exchange
- internal storage
- SD-Card storage (the ELOC stores a config file with each recording)
- (optional) manual config via config file on the sd card

Also I agree, that each command should have only 1 clear meaning and things like "setGPS" and "start recording" should not be mixed up.

I think the app has a "Stop Recording" button, but I don't see any firmware code which stops recording via app. @josephx86 : What  is the command for "stop recording" is it acutally implemented?

All this leaves me with the essential question: How do we handle command interface compatibility? 

1. Are old versions of the firmware still actively in use in the field and do they have to be supported by the App?
2. Is there robust way of distinguishing the version, i.e. firmware or command interface version which can be used with existing Firmware as well?

I think this should be clarified first, so that we know which parameters we are free to modify.

From my point of view I'm pretty free in what API format to use and open to proposals. At least it should have:
- a clear API version info
- a clear termination: @josephx86 : Is there currently a command termination with '\n'?
- error codes or an error flag in the response format. So in case a command is not supported, the device has an error, etc. this can be communicated to the App.

josephx86, 2023-05-02:
"stop recording" is implemented on one line of code, line 1715 @
https://github.com/EDsteve/ELOC_may_23_2022/blob/main/src/main.cpp#L1715
It just let's the code break out of the "recording" loop.

For the termination, there is no command from the app that terminates with (or depends on the use of '\n'). 

I agree, error codes/messages will definitely make the app handle unknown commands better.

EDsteve, 2023-05-02:
@LIFsCode 
_Are old versions of the firmware still actively in use in the field and do they have to be supported by the App?_
There are "only" 30 devices out there still used in the field. Not sure how much effort it is to keep them compatible. But i guess not really easy. So for the new ELOC 3 i can just release a complete new app in the Playstore and keep the old app separated for the old ELOCs.
I know. Not the prettiest solution, but these devices have very different hardware (e.g. no PSDRAM, smaller Flash, different GPIO config, SD card interface: SPI instead of SDIO, lots of missing features). So it might not worth it to keep them compatible forever anyway. My opinion: No need to waste energy for the old devices.

About the API: I let the experts decide. But if i can do anything for that. Let me know.

LIFsCode, 2023-05-19:
@josephx86 : The question for supporting old and new Firmware interfaces in the App is best answered by you.

I suggest the following for identifying the Interface version:

1. currently (ELOC 2.x Firmware): sends this string upon connect: 
   -  `"#<SampleRate>#<secondsPerFile>#<location>\r\n"`
   -` "getClk\r\n"`
   -  e.g. 
       - `#20000#60#river`
       - getClk
2.  new Frimware (ELOC 3.x onwards): would send this upon connect: 
   -  `"{"device": <device identifier>,  "cmdVersion" : <cmdVersion>}\r\n"`
   -  e.g.` {"device": "ELOC 3.0",  "cmdVersion" : 1}\r\n`
   -  Where the parameters have this meaning:
       - "device": HW version and type of the device 
       - "cmdVersion": version of the interface API, will be incremented every time a new command is added or existing commands get modified

I think this identification method is future proof and gives us more flexibility to get rid of deprecated command & responses. From a Firmware point of view I would prefer cleaning up the interface.
@josephx86 :
Please let me know what you think about it, I'm open for any suggestions for changes. 
Especially let me know if you from an App point of view think it's worth switching to a new interface, or if we rather stay with existing interface.

josephx86, 2023-05-19:
@LIFsCode that's perfect. It is exactly the format I had in mind :)

LIFsCode, 2023-06-01:
I have a proposal for a command interface between APP and Firmware for new ELOC 3.0 devices:

https://github.com/LIFsCode/ELOC-3.0/wiki/ELOC-3.0-App-Interface-(proposal)

@josephx86 : could you have a look at it, if this suits your needs. Please feel free to comment or adjust anything.

josephx86, 2023-06-01:
So this is like defining a protocol, right? It is an excellent proposal, especially with the versioning being determined right from the start. Developing both the firmware and the app against the protocol will certainly make life easier for everyone. 

There are those commands starting (and/or ending) with underscores - I would suggest we stick to one command naming scheme. It is simpler to go without the leading and trailing underscores, unless they hold some kind of meaning.

LIFsCode, 2023-10-04:
@josephx86 ok, I adjusted the commands without the underscore.

Besides that do you have any comments about the basic response format:
`<ECODE>:<CMD>:<RESPONSE>\r\n<EOT>`

Is <EOT>(0x04) fine for you as a response termination character

And also about the parameter format I could suggest 2 possibilites:
1. parameter definition by order: `<command>#<parameter1>#<parameter2>[#<parameterN>...]\n`
2. parameter definition by key/value: `<command>#<key1>=<parameter1>#<key2>=<parameter2>[#<keyN>=<parameterN>...]\n`

Anything you would prefer or any comments on this?

josephx86, 2023-10-05:
1. The response format is fine. A question, if for whatever reason the <RESPONSE> ends up having a colon (':') in it, will it be encoded, or should the client reading the response and check for the first two colons: one after <ECODE> and the second one after <CMD? Either way is fine - it is good to have that clarified from the start.

2. (0x04) is a good termination character. 

3. Let's go with the key/value format. It is verbose but it is also self explanatory. The format will prove useful if we ever run into a case of more that 5 or so parameters. Key-value pairs will be easy to understand and order of parameters will not matter.

LIFsCode, 2023-10-05:
2. fine
3. ok, I will adjust the wiki with a key value format. Should we use that in the reponse as well?

I will also add some more examples to get a better idea of how it will look like.

concerning point 1: 
Yes the idea is to have a simple format for the response which hold the basic info (error code, command which is responed to).

E.g. The response to `getStatus `would look like this:
```
E0000:getStatus:{
  "battery": {
    "type": "no Battery",
    "state": "Full",
    "SoC[%]": 0,
    "voltage[V]": 0.186493337
  },
  "session": {
    "identifier": "",
    "recordingState": "0",
    "recordingTime[h]": "0.00"
  },
  "device": {
    "firmware": "ELOC_V0.1",
    "timeStamp": "0.01",
    "totalRecordingTime[h]": "0.00",
    "SdCardFreeSpace[GB]": "704.55"
  }
}<EOT>
```

if receiving an invalid command, e.g. `getMeMoney`

the device would respond:
`E0105:getMeMoney:ESP_ERR_NOT_FOUND<EOT>`

The idea of sending back the received command is to identify any errors in the transmission path better. But we could remove this if you think it's not necessary.

So for the App part there are 2 levels of parsing:

1. Parse the basic format separated by `:` then check the error code to see if the response is valid
2. Parse the response data, usually this will be in json format for all commands.
- How do you think about responses to simple commands as single string, not jsons? I could think about something like cmd: `getBatteryVoltage`, response `E0000:getBatteryVoltage:3.123<EOT>`, 
- Or do you prefer having the response payload always in json?

@josephx86 Is this 2 level parsing of the response fine for you? 
My idea was to have a simple format with less strict syntax than json for the basic communication to be able to transport reliable any error information, while using all advantages of json for the payload which may provice more complex data structures.
If you prefer different format or different delimiter let me know.

josephx86, 2023-10-05:
"Or do you prefer having the response payload always in json?"
Yes, let's have all responses in JSON, so we have a uniform protocol for responses.

"Is this 2 level parsing of the response fine for you?"
Yes, it is fine. As long as we have a consistent format for the responses, which from the QnA above, the response will always have JSON in it. Parsing will be simple and straight forward.

"The idea of sending back the received command is to identify any errors in the transmission path better. But we could remove this if you think it's not necessary."
Let's keep it as it is, what you have done is an excellent way to identify errors in a response.

josephx86, 2023-10-05:
That is perfect!

LIFsCode, 2023-10-26:
@josephx86 : I have pushed a early testing version for the new command interface to the branch cmdInterface:
https://github.com/LIFsCode/ELOC-3.0/commit/eb5c020e4faf08151e5a2661ec5b1331663f3ebc

Please note, that this is not fully tested and will for sure have some issues, but I think this should enable you to get started with some tests.
Not all commands are implemented yet. I have added a "getHelp" command. If you issue this, it will respond with a JSON array of all currently available commands each with a short description about their usage.

Please let me know if you encounter any errors or inconsistencies.

EDsteve, 2023-10-26:
@josephx86 I have the cmdinterface branch ready and running on an ELOC 3. Since you don't have the necessary HW yet. Let me know when you are available to connect to my PC.

LIFsCode, 2023-10-29:
### set Time functionality
@josephx86 @EDsteve : I have some questions about the setTime functionality:

The previous command was: "_setClk_G__120____32456732728"
1. What is the "G"/"P" type used for? I don't see any usage in the ESP32 code. What was the intention?
2. What is the last sync  (in the example "120" used for?
3. What is the best format for the timestamp? unix like epoch timestamp in seconds since January 1, 1970?

Currently I would suggest the following format:
`setTime#time={"seconds":1351824120, "ms" : 42, "timezone" : 6, "type" : "G"}`
Return Payload:
`{"Time[ms]" : 12345678 }`

- seconds: is mandatory UNIX epoch
- ms: optional
- timezone: currently the timezone if hardcoded, i would prefer having this configureable. @josephx86 : what's the best format? integer number as hour offset from GMT?
- "type": I personally would remove this if there is no usage for it

### Start/Stop recording:
I would suggest this command:
`setRecordMode#mode=start`

- mode: "start"/"stop" to start and stop the recording
- if used without mode parameter "setRecordMode" will toggle the current mode
- This can be extended by other potential future modes, e.g. recordOnEvent in combination with AI to record only after a trumpet event has been detected, .. TBD

This would remove the commands `recordStart `and `recordStop` as defined in the [wiki](https://github.com/LIFsCode/ELOC-3.0/wiki/ELOC-3.0-App-Interface-(proposal))
If you are fine with that I wil  update the wiki

You will find all the above changes here: a2c178f on current cmdInterface branch

josephx86, 2023-10-30:
I did not come up with the protocol so @EDsteve can correct me where I am incorrect. From working on the code I have learned the following:

1. What is the "G"/"P" type used for? I don't see any usage in the ESP32 code. What was the intention?
'G' indicates the source of the time value being used to set board time/clock is an internet time server. (G for Google, I would guess). 'P' means the source of the time is the phone that is connected to the board.

2. What is the last sync (in the example "120" used for?
I am not sure of that '120' value, I will check the old code to see how it is used.

3. For the timestamp, Unix time is perfect because it is a simple integer/long value that can be easily parsed.

In your suggested JSON response, I agree that the 'type' field is of no use. If the phone sends it, the phone would already know what the value is for.

As for the record mode, I agree that is better to use one command. However, I would suggest "on/off" instead of "start/stop". "on/off" is self explanatory and clearly conveys that recording is either on or off, but "start/stop" gives the idea that there is another operation or process that should do the actual mode setting.

EDsteve, 2023-10-30:
@LIFsCode @josephx86 

1. Wasn't really sure about it also. But yes. That sounds correct. 
2. I think that value shows how long ago the last time synch with Google was. 

The ELOC gets the time from the phone time. And phones usually get their time from the Internet service provider and these times are often 1 second off. Thats why the Google time synch feature was implemented in the first place.
But the google time synch has a bug and gives times from 1970 in some rare occastions. So Tom deactivated it and this option is not used any more. 

_"type": I personally would remove this if there is no usage for it_
Yes. That can be removed to keep things simple.

LIFsCode, 2023-10-31:
@josephx86 : sorry I have just done a minor change in protocol:

The error messages ascociated with the ecode is not anymore printed in the payload field, but as separate "error" field
This makes the payload easier and more consistant as it is always a json object, never a string.

LIFsCode, 2023-11-03:
@josephx86 Another question I would like to discuss with you

I came across multiple status indicator, which represent some kind of enumerated states. Example would be the "recordingState". Currently it's only on/off, but with future AI integration other states are possible 
what do you think about an enum representation?, e.g.
```
{"recordingState" : {
  "val" : 0,
  "state" : "IDLE"
}}
```
With this contents:

- "val": integer representation, easy to parse
- "state": string representation. human readable

The ESP firmware takes care that both will match

This would give us more flexibility  for future modes, e.g. a kind of "armed" mode which only records when a trumpet is detected by AI.

This would give you a easy parseable integer represenation to identify the state, but also gives a string representation human readable log files.
The alternative would be just a pure integer represenation of the state, which would be sufficient, but less readable.

josephx86, 2023-11-03:
Yes, the more readable and descriptive JSON is fine.

closed 2023-11-20 by LIFsCode