LIFsCode/ELOC-3.0 #17: LOW PRIORITY: Live status updates in the app

LIFsCode, 2023-06-01:
- [ ] Send bat-level and SD-card space to the app in regular intervals

When the app is connected to the ELOC it displays the ELOC status (Recording time, GPS accuracy, bat level, etc...)
For now these infos are only updated once you connect to an ELOC. 

These infos should update more frequently (every minute or so). Joseph has already implemented this in the app with status infos he which are calculated on the phone (Rec-time, up time, ...)

The firmware should send battery voltage and available SD card space in regular intervals to the app when connected with BT.

LIFsCode, 2024-01-05:
@EDsteve this should be done/duplicate or?

EDsteve, 2024-01-05:
@LIFsCode 
I think the JSON implementation made it easily possible to get these infos frequently in the app. Will close this "issue".

closed 2024-01-05 by EDsteve