LIFsCode/ELOC-3.0 #18: Deal with I2S Buffer Overruns

LIFsCode, 2023-06-06:
I2S sampler buffer overruns are not handled yet

See also: https://github.com/LIFsCode/ELOC-3.0/issues/13

OOHehir, 2023-11-23:
With the redesign of the I2S sampling as proposed [here](https://github.com/LIFsCode/ELOC-3.0/discussions/27#discussioncomment-7035230) & a thread based process implemented [here](https://github.com/LIFsCode/ELOC-3.0/blob/0802d64ed3dbde819b8b49046e6e109d21b2658a/eloc610LowPowerPartition/lib/audio_input/src/I2SMEMSSampler.cpp#L105)  & associated consumers 
1. wav writer [thread](https://github.com/LIFsCode/ELOC-3.0/blob/0802d64ed3dbde819b8b49046e6e109d21b2658a/eloc610LowPowerPartition/lib/wav_file/src/WAVFileWriter.cpp#L111) 
2. [AI inference](https://github.com/LIFsCode/ELOC-3.0/blob/0802d64ed3dbde819b8b49046e6e109d21b2658a/eloc610LowPowerPartition/src/EdgeImpulse.cpp#L83)  

I believe this issue is redundant. 

Please note as per [comment](https://github.com/LIFsCode/ELOC-3.0/blob/0802d64ed3dbde819b8b49046e6e109d21b2658a/eloc610LowPowerPartition/lib/audio_input/src/I2SMEMSSampler.cpp#L122) it is the responsibility of the consumers to ensure their respective buffer overflows do not occur/ accommodated for.

LIFsCode, 2023-11-28:
@OOHehir : No in fact not, I was actually thinking about a real overflow detection, by evaluating the [event queue](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/peripherals/i2s.html#_CPPv418i2s_driver_install10i2s_port_tPK12i2s_config_tiPv)

Like the code snippets posted [here](https://community.platformio.org/t/eps-now-audio-streaming/36925) No big deal, I just delayed it until the merge is done.

Besides that I have questions concerning the tasks and their priorities:

I'm missing a kind of overview of the different Task priorities:
What I get from a quick look is:

| Task name | priority (0 lowest) | Comments |
| ------------- | ------------- | ------------- |
| BT Task | 1 | |
| PerfMonitor | 3 | should be highest available prio, to get accurate values <br> should be executed rarely & fast enough to avoid affecting other tasks |
| WaveFileWriter | 8 | should be high enough to guarantee no buffer overflows |
| I2SMemsSampler| 10 | should have the highest priority as long as it only retrieves data from the DMA buffer <br> This Task must guarantee reacting to I2S DMA transfer interrupts fast enough. <br> if i2s_read() is executed too slow it could cause buffer overruns, which I have currently no idea how  to detect them |
| app_main | 1 | [The Main Task has a fixed priority of 1 higher than minimum ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/startup.html#running-the-main-task) |
| EI stuff | ? | unknown? does this run in separate tasks?  |

The BT tasks which are generated by the driver and all the FreeRTOS system tasks are not mentioned here.

Let's keep the discussion here and once we have the priorities for all tasks I woud move it to the wiki.

There are some [limitations ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#floating-point-usage) (see also [here](https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32/api-guides/freertos-smp.html#floating-point-arithmetic)) when it comes to the floating point unit of the ESP32. Therefore I would suggest to pin the 2 most ressource intensive realtime critcal tasks (EI & WaveWriter) to 2 separate cores.

OOHehir, 2023-12-13:
@LIFsCode I've [started the process](https://github.com/LIFsCode/ELOC-3.0/commit/784ffbaad8c7003f1b83464c60afd35d6d816591) of moving the EI inference into a thread, should be complete in two or three days.

OOHehir, 2023-12-16:
@LIFsCode @EDsteve With this [commit](https://github.com/LIFsCode/ELOC-3.0/commit/3e73585ba4ba654fce719c59377ed3d14e769389) the AI inference is now moved to thread based. I've made the AI inference process priority just less than the WaveFileWriter, this is becuase if the WaveFileWriter is delayed there's a chance the file could be corrupted, in the case of the AI thread a missed inference wouldn't be such an issue.

<p dir="auto"></p>

Task name | priority (0 lowest) | Comments
-- | -- | --
BT Task | 1 |  
PerfMonitor | 3 | should be highest available prio, to get accurate values  should be executed rarely & fast enough to avoid affecting other tasks
WaveFileWriter | 8 | should be high enough to guarantee no buffer overflows
I2SMemsSampler | 10 | should have the highest priority as long as it only retrieves data from the DMA buffer  This Task must guarantee reacting to I2S DMA transfer interrupts fast enough.  if i2s_read() is executed too slow it could cause buffer overruns, which I have currently no idea how  to detect them
app_main | 1 | The Main Task has a fixed priority of 1 higher than minimum
EI stuff | 7 | Slightly less critical than WaveFileWriter? More resource intensive

From the performance monitor it's easy to see that the AI inference process (called ei_thread) (for this & probably most AI models) takes up significantly more resources than the I2SMEMSampler & the WAVWriter:

        | Task         | Run Time   | Percentage
        |        stats |       4128 |   0%
        |         IDLE |    5095127 |  47%
        |         IDLE |    3489165 |  32%
        |         main |       1167 |   0%
        |    BT Server |      16538 |   0%
        |         ipc0 |          0 |   0%
        |         ipc1 |          0 |   0%
        |         hciT |          0 |   0%
        |     BTU_TASK |          0 |   0%
        |     BTC_TASK |          0 |   0%
        | wav_file_writer |     135478 |   1%
        |    ei_thread |    1838531 |  17%
        |    esp_timer |       1328 |   0%
        | btController |      19094 |   0%
        |     I2S read |     132983 |   1%
        |      Tmr Svc |          0 |   0%
        |       spp_tx |          0 |   0%

I haven't pinned the tasks in the code yet, perhaps it makes sense to pin the I2SMEMSampler & WaveFileWriter to the same core & the AI thread to the other?

OOHehir, 2024-01-06:
@LIFsCode @EDsteve I notice with the more recent I2S driver there is a [feature](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#how-to-prevent-data-lost) to detect buffer overflows which may help solve this issue. I'm traveling at the moment but from next weekend I should be able to update the I2S driver used & implement this feature.

LIFsCode, 2024-01-06:
@OOHehir I tried that myself last year when I took over the project, but failed on this. Since we are using Arduino as IDF component we are limited to IDF version 4.4 or something like that.

The new I2S driver was introduced in IDF 5.x. I found no way to use newer IDF functions within this framework which would compile. But maybe you are more lucky :)

OOHehir, 2024-01-06:
@LIFsCode Looks like your correct, here is the [latest release](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3) of the Arduino version of ESP IDF. It uses ESP-IDF v5.1. Unfortunately its in Alpha release. Probably worth waiting this this to hit at least Beta or beyond.

LIFsCode, 2024-01-06:
Yeah I would rather wait for a stable release. The Arduino  framework is usually not very well tested for low power management and we are using this to a significant extend. So unless we have a strong need for going to IDF 5.x I would wait for a stable Arduino release.