LIFsCode/ELOC-3.0 #28: Add UART-Programmer reset logic to HW

LIFsCode, 2023-09-25:
Currently the EN, GPIO0, GPIO2 Pins have to be set manually if the ESP should be programmed via UART.

This can be avoided by a few transisotrs on the board:
[image]

Source from here: https://esp32.com/viewtopic.php?t=1914

Requires 2 additional UART signals (DTR, RTS) on a pin header.

EDsteve, 2023-09-26:
@LIFsCode 
Thanks. I was never really sure how it would work. Now i know :)
But my USB-TTL converter does not have a RTS or DTR pin and the USB-TTL adapter you have has a no DTR pin. So we would need a different USB-TTL adapter i guess?

P.S.: Finally back home and do some timedrift and power tests now.

LIFsCode, 2023-09-26:
You could leave the buttons as is, so you always have a backup plan ^^

Anyway I think the USB TLT adapter has DTR, but simple not routed to the pin header
[image]

So maybe we just have to add another pin header and modify the housing

closed 2023-11-17 by EDsteve