LIFsCode/ELOC-3.0 #31: Build instructions or zipped binaries for download.

josephx86, 2023-10-12:
I tried to do a fresh build and I got many issues with dependencies for the C/C++ libraries. I could not find any instructions for building a firmware binary. Is it possible to add a build guide, including supported operating systems?

Alternatively, it might be easier to provide various binaries for the firmware of supported boards, so that anyone wanting to test, can just flash the firmware, without having to manually build.

LIFsCode, 2023-10-12:
Sorry I cannot reproduce this, I just did a clean checkout and it builded without errors with VS Code.

But you need a working internet connection while building, so VS Code PlatformIO downloads all required packages.

Maybe @EDsteve  has some tips for getting it working on a fresh setup

josephx86, 2023-10-12:
The build instructions are not a big deal but will you please upload the compiled binaries to be available as "releases" for the repo? If possible, two binaries - one for the new hardware and the other for the old (SPI) hardware.

EDsteve, 2023-10-13:
@josephx86 @LIFsCode 
I just realize that you still use the old ELOC 2.7 (black PCB), correct? That's my fault. I haven't send you the new board yet. I just tried to use _#define USE_SPI_VERSION_ but that spits out plenty of errors. We kind of have abandoned the SPI version and i think it would be a lot of effort to make it work again with all the new firmware features. 

Not the best solution but a workaround:
I will send you the new ELOC 3.2 today. The 60USD for shipping will be painfull, but then it hopefully arrives within a week.
You can then either wait for it to arrive or start working via AnyDesk on my computer, as we did in the past :) Which gives you access to the new HW immediately. I have a second PC here now so you can have one all for your self :)

Besides that i will make instructions for VScode/PlatformIO which should build succesfully. Or alternatively send you binaries.