LIFsCode/ELOC-3.0 #33: Integer or float/double value in JSON for consistency

EDsteve, 2023-10-31:
This is a minor/negligible issue but for consistency, the response from the getConfig command need the value of the "locationAccuracy" as in integer or double type instead of the string type that is currently returned:

"locationAccuracy": "99", -----> "locationAccuracy": 99,

EDsteve, 2023-10-31:
Other field that need to have numeric types are : 
 "recordingState": "0" -->  "recordingState": 0 
 "recordingTime[h]": "0.00" --> "recordingTime[h]": 0.0
 "timeStamp": "0.54"  --> "timeStamp": 0.54
 "totalRecordingTime[h]": "0.00",  --> "totalRecordingTime[h]": 0.0,
 "SdCardFreeSpace[GB]": "115.58"  -->  "SdCardFreeSpace[GB]": 115.58

LIFsCode, 2023-10-31:
@josephx86  fixed most of it, just a few questions:

1. what is the format of location accurarcy ant what's it measning? I now chose double as a value, no glue about precision
2. "timeStamp" what is this used for? Now it is returning the time since boot, should we maybe give it a better name? e.g. "runtime[h]"?

One thing concerning "recordingState" what do you think about an enum representation?, e.g.
```
{"recordingState" : {
  "val" : 0,
  "state" : "IDLE"
}}
```
This would give us more flexibility  for future modes, e.g. a kind of "armed" mode which only records when a trumpet is detected by AI

EDsteve, 2023-11-01:
@LIFsCode 
Joseph (who wrote that issue while working on my PC as you might have guessed already :) is not availabe today but i can answer some questions:

1. This number defines how accurate the GPS position is in meters. It's a number without decimals from 0 - 100.
2. We were looking for the uptime value, but coudn't find it. Now we know it hides behind timeStamp :) Renaming it to runTime sounds like a good idea.

For now recordingState only defines if the recording is running or not. So the value is either 0 or 1. Correct? Not so sure about the recordingState and what can be achieved with the enum representation. But at the end we need to cover the following scenarios:

1. Recording 24/7 without detecting (Without AI running) 
2. Detecting and recording 24/7 at the same time
3. Detecting with a short recordng after a trumpet was detected 
4. Detecting without recording at all

So a new proptery which is called "recordDuringDetect" which can be set to 24/7, time in seconds or OFF is necessary?

But this will be part of a seperated list of added comands and questions which I will send you via email or github issue later on. No worries. Noit too much :)

ED

LIFsCode, 2023-11-03:
> 
>     This number defines how accurate the GPS position is in meters. It's a number without decimals from 0 - 100.
>     We were looking for the uptime value, but coudn't find it. Now we know it hides behind timeStamp :) Renaming it to runTime sounds like a good idea.

Ok , then I 

- [ ] change the accuracy to integer
- [ ] change the name of "timestamp". @EDsteve  is "Uptime" a good name for it?

EDsteve, 2023-11-04:
Yes. "Uptime" sounds good to me.

LIFsCode, 2023-11-08:
I set it to "Uptime[h]"

commit 46388a55 (2023-10-31): Fix string/number interpretation of config and status

* Fix #33

* remove obsolete settings of GPS, which is not used anymore can be done
  via json config.

* add macros.hpp as a general place to store some global helper functions
commit 8bb4e56e (2023-11-08): Rework global status fields (NOTE: still not good, but at leas separated)

* Move all global status indicators to ElocStatus.cpp/hpp

* Change gRecording to enum RecState (IDLE, RECORDING) to be adjustable
  for future integration of AI

* Output Recording state as enum in JSON with val (int) and state (string)
  for better readability.

* Change "Timestamp" to "Uptime[h]"

closes #33
closed 2023-11-08 by LIFsCode