LIFsCode/ELOC-3.0 #35: Additional properties for response from ELOC

EDsteve, 2023-11-01:
Looks all pretty good already and Joseph made good progress on the app. 
While implementing the BT interface we came accoross a few things which are still missing in the response:

**"Grid ID"**
This is a value set in the app for each ELOC. It's the prefix for each recorded file. We used it before as Grid ID, but can be used for anything else. "FileHeader" might be a more universal name. 

**"SoC[%]"**
Does that work already and we get SoC from the ELOC? If yes. Then we can remove the state:FULL. If no. No problem. It can wait.

**"SdCardSize"**
Besides the "SdCardFreeSpace" we also need the actual size of the SD card to calculate free space in %. Possible?

**detectingState, detectingTime[h], totalDetectingTime, modelName, detectedSounds, ..**.
Just pointing it out for now. These and more will be needed after the AI got implemented. 

_

**These are the current responses from the ELOC:**
I have "striked through" all unnecessary properties. They should not be in use.

**greeting from device:**
{"device" : ELOC 3.0, "cmdVersion" : 1}

command: **getConfig**
{"ecode" : 0, "cmd" : getConfig, "payload" : {
  "device": {
    "location": "",
    "locationCode": "unknown",
    "locationAccuracy": "99",
    "nodeName": "ELOC_NONAME"
  },
  "config": {
    "secondsPerFile": 60,
    ~"listenOnly": false,~
    "cpuMaxFrequencyMHZ": 80,
    "cpuMinFrequencyMHZ": 10,
    "cpuEnableLightSleep": true,
    "bluetoothEnableAtStart": false,
    "bluetoothEnableOnTapping": true,
    "bluetoothEnableDuringRecord": true,
    "bluetoothOffTimeoutSeconds": 60
  },
  "mic": {
    "MicType": "ns",
    "MicBitShift": 11,
    "MicSampleRate": 26000,
    "MicUseAPLL": true,
    "MicUseTimingFix": true,
    ~"MicGPSCoords": "ns",~
    "MicPointingDirectionDegrees": "ns",
    ~"MicHeight": "ns",~
    ~"MicMountType": "ns"~
  }
}}

command: **getStatus**
{"ecode" : 0, "cmd" : "getStatus", "payload" : {
  "battery": {
    "type": "LiPo Battery",
    ~"state": "Full",~
    "SoC[%]": 4.309902668,
    "voltage[V]": 3.563073397
  },
  "session": {
    "identifier": "",
    "recordingState": "0",
    "recordingTime[h]": "0.00"
  },
  "device": {
    "firmware": "ELOC_V0.1",
    "timeStamp": "0.54",
    "totalRecordingTime[h]": "0.00",
    "SdCardFreeSpace[GB]": "115.58"
  }
}}

command: **getHelp**
{"ecode" : 0, "cmd" : "getHelp", "payload" : {
  "commands": [
    {
      "cmd": "getHelp",
      "help": "Print all available command and their help as JSON"
    },
    {
      "cmd": "setConfig",
      "help": "Write config key as json, e.g. setConfig#cfg={\"device\":{\"location\":\"not_set\"}}"
    },
    {
      "cmd": "getConfig",
      "help": "Read config as json, e.g. getConfig --> return{\"device\":{\"location\":\"not_set\"}}"
    },
    {
      "cmd": "delConfig",
      "help": "Delete the current config file. Current config is not reset to default until next reboot"
    },
    {
      "cmd": "getStatus",
      "help": "Returns the current status in JSON format"
    },
    {
      "cmd": "setTime",
      "help": "Set the current Time. Time format is given as JSON, e.g. setTime#time={\"seconds\":1351824120,\"ms\":42,\"timezone\":6,\"type\":\"G\"}"
    },
    {
      "cmd": "setRecordMode",
      "help": "Start/Stop recording. If used without arguments, current mode is toggled(start/stop). Otherwise set recording to specified mode, e.g. setRecordMode#mode=start"
    }
  ]
}}

LIFsCode, 2023-11-03:
**"Grid ID"**
I think this is already there, it is the config field "location". Sorry I had no glue about the meaning of all the config parameters couldn't get any other info as from the ESP32 source code and there it was named "gLocation" or "file header" but we could call it "Grid ID" as well, whatever you prefer.

**"SoC[%]"**
Yes and no: It is working as good as it could with the HW. It is a simple mapping of battery voltages to SoC levels (see [here](https://footprinthero.com/lifepo4-battery-voltage-charts)). However the battery voltage is among others highly dependant on:
- temperature
- current drained from, or charged into the battery
Hence without any possibility to measure the current and thus integrate the actual capacity it is not possible to get exact SoC values. Also the ADC of the ESP is very inprecise for voltage measurements.

However it is still better than simple having full/empty states which do also only rely on the measured voltage

**"SdCardSize"**
Yes it's on the list

- [ ] SdCardFreeSpace
- [ ] free space in %

> 
> detectingState, detectingTime[h], totalDetectingTime, modelName, detectedSounds, ...
> Just pointing it out for now. These and more will be needed after the AI got implemented.

Yes that's the reason why we switched to the JSON interface. It's much more simple to extend it with further parameters.

> These are the current responses from the ELOC:
> I have "striked through" all unnecessary properties. They should not be in use.

Great that will clean it up I was already wondering what the meaning of these parameters should have been ;-)

EDsteve, 2023-11-04:
**Grid ID**
Let's call it "fileHeader". Because it will be a prefix in the filename of every recorded file.

EDsteve, 2023-11-06:
@LIFsCode You can remove it as well. In case it will ever be used (small chances) We can always add it again i guess :)

commit d9ebcbc9 (2023-11-04): Moved SD card stats to SDCard class & enhance status output for SD card size

* Add fields to getStatus output
  * "SdCardSize[GB]"
  * "SdCardFreeSpace[%]"

* output session["recordingState"] as integer (TODO: this is up to changes
  to encapsulate enum types.

* remove obsolete functions from main.cpp:
  * freeSpace()
  * global variables for SD card space

* closes #35
commit a844fd48 (2023-11-08): Remove obsolete config parameter 'MicPointingDirectionDegrees'

* closes #35
closed 2023-11-08 by LIFsCode