LIFsCode/ELOC-3.0 #41: ELOC measures charging voltage and not "rest" voltage

EDsteve, 2023-11-17:
Before the battery voltage gets read from the ADC, the charger should be turned off before that. Otherwise we would read the charging voltage. In the actual master branch the voltage gets read every 5 second or so. That means the charger is turned off every 5 seconds. And that is also happening. So far so good.

**Test:**
Li Ion battery connected to ELOC - Multimeter connected to battery - No charger connected:
ELOC reads 3.5V -> OK
Multimeter reads 3.5V continuous  -> OK

Now i connect the ELOC to USB-C for charging.
ELOC reads 3.8V -> Not okay
Multimeter reads: 3.8V for two seconds and then 3.5V for two seconds in a loop 

It shows that the charger gets turned off but needs around 2 seconds to show effect? Maybe the ESP32 reads the voltage too early after the charger is turned off? Or is my Multimeter so slow? Not sure. :)
In the field the voltage readings should happen only every 15 minutes or so. I am not sure if it so healthy for the battery to start charging ever 5 seconds :)

LIFsCode, 2023-11-17:
The battery voltage is updated only every 2 seconds. See [here](https://github.com/LIFsCode/ELOC-3.0/blob/3651b7abe816ff93af3eac2d461f5a962ef93489/eloc610LowPowerPartition/src/Battery.cpp#L102C10-L102C10)

Intention was to reduce load by too many ADC readings and thus reduce power. I can make this configureable if required. 
However this will always be a tradeof between turing the charger on and off and getting up to date values.

Im not sure if I understand this:

> Now i connect the ELOC to USB-C for charging.
> ELOC reads 3.8V -> Not okay
> Multimeter reads: 3.8V for two seconds and then 3.5V for two seconds in a loop

Do you mean the charger is turned off too slow and the ESP is reading while it is still on? That could be handled but would add another delay, making voltage monitoring more difficult.

Or do you just want to tell me that the charger is turning on and off? That is known as explained above.

EDsteve, 2023-11-17:
> The battery voltage is updated only every 2 seconds. See [here](https://github.com/LIFsCode/ELOC-3.0/blob/3651b7abe816ff93af3eac2d461f5a962ef93489/eloc610LowPowerPartition/src/Battery.cpp#L102C10-L102C10)

Ahh okay. In the log it looks like 4 seconds :)
Every 2 seconds. Is there a reason why we measure the voltages so often?

> Do you mean the charger is turned off too slow and the ESP is reading while it is still on? That could be handled but would add another delay, making voltage monitoring more difficult.
> 
> Or do you just want to tell me that the charger is turning on and off? That is known as explained above.

We turn the charger off before every voltage-reading. So that we don't measure the charging voltage, which is always much higher. Correct?

Okay. I am measuring the voltage with our super Pokit. As you can see the voltage is 3.7V when charger is off and 3.9V when charger is on. So shouldnt the ELOC read the voltage when the charger is off (3.7V) instead of 3.9V? 

Sorry for wrong polarity 🙈 
[image]

LIFsCode, 2023-11-17:
ok, now I get you. 

Yes that's right. Probably the reading occurs too early, I can try adding a bit of delay.

For the voltage update: I set it to 2 seconds to get a better feedback response. So no idea about which update rate is required, I will think about it

EDsteve, 2023-11-20:
With the two seconds, the solar panel charges only 50% of the time. So that's a no go :)
For my purpose i only need every >10 minutes. Not sure how much makes sense for the Firmware.

LIFsCode, 2024-01-05:
Measurements show that ~150 ms are required at minimum to get a resting voltage. Additional averaging with larger intervals (~10-50 ms) will give better result:

```
I (236105) Battery: "0 ms": 3.557 V
I (236113) Battery: "50 ms": 3.381 V
I (236118) Battery: "100 ms": 3.340 V
I (236123) Battery: "150 ms": 3.327 V
I (236133) Battery: "200 ms": 3.321 V
I (236149) Battery: "250 ms": 3.329 V
I (236155) Battery: "300 ms": 3.333 V
I (236160) Battery: "350 ms": 3.333 V
I (236165) Battery: "400 ms": 3.336 V
I (236170) Battery: "450 ms": 3.336 V
I (236186) Battery: "500 ms": 3.327 V
```

EDsteve, 2024-01-05:
@LIFsCode 
Would a capacitor at the Analog input make things "smoother"?
See: https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/peripherals/adc.html

We still need to find a suitable voltage reading interval. What speaks against every 5 to 20 minutes?

LIFsCode, 2024-01-05:
I'm working on it. I will make the interval configurable.

commit 767dd815 (2024-01-07): Fix reading of correct battery voltage after charger is disabled

fixes #41
closed 2024-01-07 by LIFsCode