LIFsCode/ELOC-3.0 #45: Resolve differences between sdkconfig.esp32dev & sdkconfig.esp32dev-ei

OOHehir, 2023-11-24:
@LIFsCode There are some differences between the two version of sdkconfig. I think it would make more sense just to have one. I'm unsure of whats are required settings & what happens to be 'just different'.

Could you perhaps take a look & see whats needed from your code? 

I've enabled PSRAM/SPIRAM & set the flash size to 16MB in both versions.

LIFsCode, 2023-11-28:
@OOHehir I just had a look at the differences acutally all is requierd!

They affect: 
1. https://github.com/LIFsCode/ELOC-3.0/issues/26
2. Firmware update & partition layout

OOHehir, 2023-12-07:
@LIFsCode @EDsteve I've just pushed a new [commit](https://github.com/LIFsCode/ELOC-3.0/commit/76e01bc80f94740f6acd729759aedcf1ea081de2) that more closely aligns the non-EI & the EI build. I hope this won't cause any build issues for you both, please let me know if it does. 

There doesn't appear to be a straightforward way of using the same sdkconfig file for both builds so looks like we'll just have to keep them manually identical. I considered using a 'linked' file but might cause cross-platform issues.

LIFsCode, 2024-01-04:
I [combined](https://github.com/LIFsCode/ELOC-3.0/commit/988971ef7430c5b09151c2707bd02344cfa6fd5d) both sdkconfig files into a single .defaults file.

[Here ](https://github.com/espressif/esp-idf/issues/5658#issuecomment-823286894) is a short explanation about the hierarchy of sdkconfig files.
So it is still possible to add different sdkconfig options for esep32dev and esp32dev-ei build, but both now run from the same default config

OOHehir, 2024-01-06:
@LIFsCode Further to this [discussion](https://github.com/LIFsCode/ELOC-3.0/commit/988971ef7430c5b09151c2707bd02344cfa6fd5d) I added a [comment](https://github.com/LIFsCode/ELOC-3.0/commit/f7544de28920f6eec9cdda215055c504b091df96) here to clarify the handling of these generated files.

commit 988971ef (2024-01-04): Use only a single sdkconfig (default) file for ei and non-ei build

* fixes #45

* see here for better explanation: https://github.com/espressif/esp-idf/issues/5658#issuecomment-823286894