LIFsCode/ELOC-3.0 #46: Implement wav file writing control from Bluetooth

OOHehir, 2023-11-25:
Currently the recording of wav files is '[hard coded](https://github.com/LIFsCode/ELOC-3.0/blob/83c0efb46e15ac256bf49def9e2a827d9636b49c/eloc610LowPowerPartition/src/main.cpp#L1040)' as being continuous, i.e. starts at boot-up.

As previously [discussed](https://github.com/LIFsCode/ELOC-3.0/discussions/27) this needs to be set/ controllable from the config structure. 

Should this be polled from main.cpp or will the Bluetooth code set this directly on change?

OOHehir, 2023-11-25:
Additionally to this the setting "MicBitShift" seems redundant, the code uses a combination of [I2S_BITS_PER_SAMPLE](https://github.com/LIFsCode/ELOC-3.0/blob/83c0efb46e15ac256bf49def9e2a827d9636b49c/eloc610LowPowerPartition/src/project_config.h#L121) & [I2S_DEFAULT_VOLUME](https://github.com/LIFsCode/ELOC-3.0/blob/83c0efb46e15ac256bf49def9e2a827d9636b49c/eloc610LowPowerPartition/src/project_config.h#L140C17-L140C35) which is mic specific. 

Perhaps the mic volume should be 'settable' instead?

LIFsCode, 2023-11-28:
> Additionally to this the setting "MicBitShift" seems redundant, the code uses a combination of [...]

@OOHehir @EDsteve : agree, I would suggest "MicVolume2^x" to indicate it will cause a power of 2 volume change.

in config e.g. 
```
{
  "mic": {
    "MicType": "ns",
    "MicVolume2^x":3,
    "MicSampleRate": 20000,
    "MicUseAPLL": true,
    "MicUseTimingFix": true
  }
}
```

For the BT app I would hide this limitation and simply provide a drop down meue with available "volume gain" or "gain", e.g. 0, 2, 4, 8, etc.

OOHehir, 2023-11-28:
@LIFsCode Thanks. I'm just working on this BT implementation, at least an initial 'draft'. Hope to have something pushed this evening.

OOHehir, 2023-11-28:
@LIFsCode @EDsteve I've just made a [commit ](https://github.com/LIFsCode/ELOC-3.0/commit/9d779a0221314fe09335097888a8eb88e805c8c9) that is hopefully a starting point for this issue. 

The wav file 'mode' is now set in response to the button press (GPIO0). At the moment it toggle between continuous & disabled states, perhaps there is a requirement to also include 'single' (i.e. when the AI detection reaches a threshold)?

I haven't had a chance to test the Bluetooth interface yet, some mundane things to do at home here..

EDsteve, 2023-11-29:
@OOHehir I have just tested the GPIO0 button and it does toggle between continous recording and disabled. -> works
It does not stop recording right away though. It still finishes the "recording time per file" (which can be set in the app). So when this value is set to one minute. I still have to wait for it to finish  the minute. That's not really practical if the recording time per file is set to 1 or 12 hours :)
Would be much nicer, if the recording could stops right away after pressing the button.

OOHehir, 2023-11-29:
@EDsteve Sure, understandable. I wasn't sure on the functionality required for this. Do you need the switch to cycle through the available modes or just disabled & continuous?

EDsteve, 2023-11-30:
@OOHehir Actually this button is not too important for the end user. The user should use the app. It's more like a backup solution without app. Or for us to test things.
So it should start/stop recording. And with a long press (or double press) start/stop detecting. But that's not a too important thing.
But if you have other ideas about it. Let me know :)

closed 2024-01-06 by OOHehir