LIFsCode/ELOC-3.0 #48: Commands are not working any more after merge?

EDsteve, 2023-11-29:
@OOHehir @LIFsCode 

Sending serial commands over BT doesn't seem to work any more. Before the merge i could start/stop recording or get status reports with getStatus or getConfig etc. There is no response any more and i guess the merge broke that. Can't do testing if i can't control the device :)

The commands are described here. And

EDsteve, 2023-11-29:
If this issue is dealt with at "[Implement wav file writing control from Bluetooth](https://github.com/LIFsCode/ELOC-3.0/issues/46)". 
The of course my post can be deleted.

OOHehir, 2023-11-29:
@EDsteve This is no doubt as a result of the changes I pushed. Should get a chance to look at it tomorrow evening (my time).

EDsteve, 2023-11-30:
It would have been a miracle if everything worked right away after the merge. Haha
Thanks for looking into it. Only if you have the time of course.

LIFsCode, 2023-11-30:
Sorry I'm also quite busy this week, I can be of more help next week.

@OOHehir What I noticed when I saw the merge, is that there is no usage of FreeRTOS Queue, but rather polling the 'buf_ready' flag with a delay of 10 Ms (WavWriter) and 1ms (inference).
Based on the [priorities](https://github.com/LIFsCode/ELOC-3.0/issues/18#issuecomment-1830537981) this could cause the Bluetooth task to starve during processing commands. Parsing the json may take some time. Also I don't remember the priority of the BT serial driver lib. This could be a potential issue as wenn due to timing requirements.

Just a theory, but maybe it helps. 

@EDsteve Is the performance monitor active?do you have any CPU/task loads in the log?

EDsteve, 2023-11-30:
@LIFsCode 
Yes. It comes up every few seconds in the serial log. Looks like this:

        | Task         | Run Time   | Percentage
        |        stats |       3676 |   0%
        |    BT Server |     163583 |   1%
        |         IDLE |    5078727 |  47%
        |         IDLE |    3133361 |  29%
        |         main |     461375 |   4%
        |      Tmr Svc |          0 |   0%
        |         ipc1 |      10512 |   0%
        |       spp_tx |          0 |   0%
        |         ipc0 |      17763 |   0%
        | btController |    1073255 |  10%
        |    esp_timer |       1202 |   0%
        | Wav file writer | Created
        |         hciT | Created
        |     BTU_TASK | Created
        |     BTC_TASK | Created
        |     I2S read | Created
        -----------------------------------

OOHehir, 2023-11-30:
@LIFsCode I'm working on this at the moment, when I try starting recording I get the response:
```
E (294486) BluetoothServer: Invalid Command setGPS^UNKNOWN, Received setGPS^UNKNOWN
```

Not sure whats causing it, will dig in a bit more..

LIFsCode, 2023-11-30:
Are you using the ELOC app?if yes this could be part of deprecated code. The "setGPS^" is deprecated and has been removed from the code. The app must not send it anymore
Actually I don't know if there is a new version of the app available @EDsteve ?

Are we maybe mixing up multiple problems?
- Android App
- BT Command interface
- effects of the EI merge

Best just use a pure BT serial terminal.

EDsteve, 2023-11-30:
> @LIFsCode @EDsteve I'm using 4.67 AppBeta of the ELOC app, I don't see any other version of the app available.
> 
The 4.67 still uses old commands only compatible with the ELOC 2.7 hardware. So that won't work. I should write that in the description actually.

The new app still needs the new commands to be implemented (within the next few days). That's why i am using BT serial tools for now to send commands for testing. As soon as the app is ready i will send you the link.

LIFsCode, 2023-11-30:
@EDsteve @OOHehir ok so if I understand it correctly there never has been a problem with the commands except for the start/stop of the recording, it was all about the command termination in the windows terminal.

> The only thing in the code I'm unsure of is this line:
> 
> status += "{\"recordingState\" : ";
> status += String(rec_req == REC_REQ_START);  <------------------------------------
> status += "}";
> ESP_LOGI(TAG, "%s Recording...", newMode);
> Its commented out at the moment

@OOHehir this should best be set to the same as in the getStatus response field.
We have defined a [format](https://github.com/LIFsCode/ELOC-3.0/issues/10#issuecomment-1792531132) for enums in the json payload which is a tuple of the
- interger value (must not be changed without change of CMD Interface Version) this uniquely identifies the enum state
- string representation for better readability in Log Files

Sorry I have not added this to the command interface wiki yet.
I think in your last commit the tuple for the enum state is only a string. Basically the response of `setRecordMode` should be the same as in `getStatus` the field--> `{"session" : {"recordingState" : }}`

OOHehir, 2023-11-30:
@LIFsCode Thanks, I think I understand where your coming from. Unfortunately I had used a strongly typed enum for the wav writer mode so it doesn't lend itself to the template conversion. I added a method to the wav writer class to report mode in string & int form & used that in the JSON. I'm not sure what you think of that approach, if you'd rather revert no problem.

From what I can see the wav writer operation is set at boot up, from Bluetooth & GPIO0. I understand the current code uses the FreeRTOS queue. To access the enum modes of the wav writer I had to include the header (in BluetoothServer.hpp)  for wav writer & the (globally scoped) object (to enable toggling the mode). At that point it seemed more efficient to set the mode directly rather than the queue but not sure if you want to proceed in that direction? If you wish to proceed in a solely based queue manner I can modifythe code. It is similar for the AI setting (discussed below).

@LIFsCode @EDsteve I've added code to report & set the AI operation. I've tested briefly & it looks to be working as expected. The mode can be changed, similar to the wav writing, as:
setAIMode (toggles)
setAIMode on
setAIMode off