LIFsCode/ELOC-3.0 #50: Merge bugs

EDsteve, 2023-12-03:
@OOHehir @LIFsCode 
After a few tests i found a few bugs which got introduced during the merge. I compared it with the FW before the merge which does not have the following bugs.

1. Power consumption rose from 18mA to 30mA
I have done a few measurements with different devices. 
3. Every time i connect with the app to the ELOC. A folder is created on the SD card. A folder should only be created when i start recording
4. Recordings stop randomly after a few hours (due to restart?)

EDsteve, 2023-12-03:
@OOHehir @LIFsCode Okay. More details:

1. Power consumption
I will add some logs and graphs about this soon.
2. Empty folder
Correction: This folder is actuallly created with every restart. Doesn't matter if i connect with the app or not.
3. I have done three tests so far. The recording stoped between 1 and 3 hours.
This problem is difficult to "debug" without correct time stamps. That seems not implemented yet. All the recordings always get the time from the day i flashed the firmware (I think?). The ELOC should get the time from the app every time the app connects to an ELOC. Is that something we can add soon? Joseph is ready to implement the command to send the time from app to the ELOC.
Is there any tests you want me to do about it?

Hope that doesn't become a too big problem.
Cheers
ED

EDsteve, 2023-12-03:
I added the missing stand by mode in my first post.

EDsteve, 2023-12-03:
Bug 3. 
I have catched the log when the recording stops:
[Rec_stop_log.txt](https://github.com/LIFsCode/ELOC-3.0/files/13538772/Rec_stop_log.txt)

EDsteve, 2023-12-03:
@OOHehir 
1. I measured 16KHz recording without AI running.
3. I have no clue how to use the idf-monitor. But will try to read about it. In the meantime: I have catched the log file when the recordings stops. That should give some insights:
[Rec_stop_log.txt](https://github.com/LIFsCode/ELOC-3.0/files/13538787/Rec_stop_log.txt)

OOHehir, 2023-12-03:
@LIFsCode @EDsteve The bug should hopefully be resolved [now](https://github.com/LIFsCode/ELOC-3.0/commit/713123e56cb157b42c930d08c9f30a9823233230). Impressive how powerful backtrace can be sometimes to track down bugs..

I should have more in the next day or two

OOHehir, 2023-12-04:
@LIFsCode 

Do you have a link for this code? Can't seem to find it.

> * ElocSystem::pm_check_ForRecording() is not called is not called anymore prior to recording. This is required to fix [the >32 kHz noise bug](pm_check_ForRecording)

OOHehir, 2023-12-05:
@EDsteve @LIFsCode This [commit](https://github.com/LIFsCode/ELOC-3.0/commit/f8d829e1b30114679dc06d22a90eda4c057765ae) changes the wav writer from a 'polling' delay to one 'woken' from the I2S task. I don't see a huge improvement (just looking at the performance monitor output) but hopefully power consumption will be reduced somewhat.

LIFsCode, 2023-12-05:
@OOHehir  moving the HEap from PSRAM to RAM  will be the biggest effect on power consumption
keeping the CPU idle ~25% (best achieved by keeping all tasks in blocked state most of the time). Will definetely help but with a lower effect.

Since the EI will increace power consumption anyway due to higher computational performance I would suggest:
- I2S_BUFFER_IN_PSRAM --> to RAM
- WAV_BUFFER_IN_PSRAM --> to RAM

While the EI buffer should remain in PSRAM.

Thus we can have a low power mode with EI disabled and a high power mode with EI enabled.

EDsteve, 2023-12-06:
@LIFsCode Good point. I just disabled it. But can't see a difference so didn't upload the new pictures. If you need. Let me know.
Config i used:
{
  "device": {
    "fileHeader": "not_set",
    "locationCode": "unknown",
    "locationAccuracy": 99,
    "nodeName": "ELOC_SD_CONFIG_Samsung"
  },
  "config": {
    "secondsPerFile": 60,
    "cpuMaxFrequencyMHZ": 80,
    "cpuMinFrequencyMHZ": 10,
    "cpuEnableLightSleep": true,
    "bluetoothEnableAtStart": true,
    "bluetoothEnableOnTapping": true,
    "bluetoothEnableDuringRecord": true,
    "bluetoothOffTimeoutSeconds": 60,
    "logConfig": {
      "logToSdCard": false,
      "filename": "/sdcard/log/eloc.log",
      "maxFiles": 10,
      "maxFileSize": 5120
    }
  },
  "mic": {
    "MicType": "ns",
    "MicBitShift": 11,
    "MicSampleRate": 16000,
    "MicUseAPLL": true,
    "MicUseTimingFix": true
  }
}

OOHehir, 2023-12-06:
> @OOHehir moving the HEap from PSRAM to RAM will be the biggest effect on power consumption keeping the CPU idle ~25% (best achieved by keeping all tasks in blocked state most of the time). Will [...]

@LIFsCode Sounds like a good approach, I'll work on it today.

@EDsteve To implement this change the wav buffer will (very) likely needed to be once only at boot time. This means that it won't be possible to change the wav file recording time without rebooting the device. I don't think this will be an issue, but need to confirm?

EDsteve, 2023-12-06:
@OOHehir Power consumption has the highest priority. If the user must reboot to change recording time in order to achieve lower power consumption. Yes. That's ok :)

OOHehir, 2023-12-07:
@EDsteve @LIFsCode I've just pushed a new commit to master which now, by default (set in project_config.h) places the I2S & WAV buffers in RAM. Looking at the performance monitor output the change looks promising, the wav_writer task has moved from approx 7 to 10% to approx 1%.  I've only tried it using the eps32dev build, rather that the task including the EI feature.

The downside is that currently the WAV buffer appears close to its [maximum](https://github.com/LIFsCode/ELOC-3.0/blob/d4a97d88a4637b433e521a26cce44a478f94d427/eloc610LowPowerPartition/lib/wav_file/src/WAVFileWriter.h#L17) size, increasing it by any significant amount generates errors about data not fitting in particular sections.

OOHehir, 2023-12-07:
@LIFsCode @EDsteve It looks like the wav buffer size is limited to approx 9700 (int16_t), anymore & the inference appears to run out of memory & fails to run.

OOHehir, 2023-12-08:
@LIFsCode The current DMA buffer length is 1000 bytes for some reason & the current DMA buffer count is 18 which seems, well, strange also. Not sure if there's a particular reason for that? Perhaps, as you suggest, 2 or 3 DMA buffers of 2028 & matched in the WAVFileWriter would be a better fit? The only other factor I can think of is if we want the core to enter sleep during writes & the min amount of time for that to occur.

With regards the changes you suggested I've started with the [wav_writer](https://github.com/LIFsCode/ELOC-3.0/commit/740fe124c77fd7ebc267c21ff8c69df6985ec6b1).

LIFsCode, 2023-12-08:
@OOHehir sleep mode (deep & light) is [not available when I2S is active](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers), because the I2S module requires the system clock.

So the only effect should the reduced context switches.

EDsteve, 2023-12-10:
@OOHehir @LIFsCode 
Due to not enough time i was only able to do power consumption tests so far. And YES. We are back on track. It is on pair with the consumption before the merge. Awesome. :)

About stand by (sleep) - which is not working after the merge yet
When i measure the stand by consumption (using the FW before the merge). It shows around 6mA. From what i have read [here](https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/). The power consumption can be less than 1mA if the MCU is paused. Is that something we can achieve as well?

EDsteve, 2023-12-11:
@OOHehir 
Yes. That is fine. The sleep mode is used in situations if someone forgets to turn off the device, carries the ELOC days through the jungle and arrives with an empty battery :)

OOHehir, 2023-12-11:
@EDsteve OK, understandable. 

On what circumstances should the ELOC enter & exit light and/or deep sleep? E.g. 30 secs after startup & wav writer & EI inference disabled? Bluetooth status?

How should it exit this state? I think GPIO0 button press or reboot is the only option from deep sleep but @LIFsCode may know more..

LIFsCode, 2023-12-11:
Hi @EDsteve @OOHehir 
Let me intervene at this point ;-)

**deep sleep**
@EDsteve  asked for a [intruder alarm](https://github.com/LIFsCode/ELOC-3.0/issues/14#issuecomment-1839068476) based on accelerometer repeated action that will cause the buzzer to constantly beep. If that's still the case, I doubt that anyone would carry around the ELOC with forgetting to turn off.

Anyway if the deep sleep option remains a requirement we should be very careful with condition for entering deep sleep to avoid  other problems, e.g. a ranger won't notice that the device entered deep sleep and wandering why it won't respond.

Concerning wakeup source: I assume a simple power cycle by the on/off switch would be the more stable (I know and less elegant ^^)variant. It avoids accidentally turning it on by something pressing on the button in the backpack. The on/off switch is a bit more robust.

**Light sleep**
This is already enabled by the [power config](https://github.com/LIFsCode/ELOC-3.0/blob/740fe124c77fd7ebc267c21ff8c69df6985ec6b1/elocConfig.json#L12). The freeRTOS is automatically entering light sleep, once no wake locks are taken. Light sleep means that the main system clocks are stopped and only the low power RTC clock is running.

That means the following modules **prevent** light sleep:
- Bluetooth (unless provided with an [external crystal](https://github.com/LIFsCode/ELOC-3.0/issues/11)
- I2S (as mentioned above)

Besides that there is an effect of the [PSRAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#power-down-of-flash) on power during light sleep, but I would treat this as negligible as it's in the range of a few 10s uA.

Sorry it has been a long post, but the power management is quite a topic. I hope I've been comprehensible

OOHehir, 2023-12-11:
> @OOHehir @LIFsCode
> 
> After a few tests i found a few bugs which got introduced during the merge. I compared it with the FW before the merge which does not have the following bugs.
> .........
>     2. Every time i connect with the app to the ELOC. A folder is created on the SD card. A folder should only be created when i start recording
> .......

This should now be resolved in the latest [commit](https://github.com/LIFsCode/ELOC-3.0/commit/a41a3fe247dd427927a4c0f9e7e5649c60a0151b).

EDsteve, 2023-12-12:
@LIFsCode @OOHehir 

**Intruder alarm:**
I had the same thought when making the buzzer table :) Therefore I was already thinking about a setting in the app "Arm device during record. ON/OFF". That would only have the device armed while recording after it's already set up and not while setting up or carrying around for example.
It's not a mandatory feature. But I would love to test that because i am sure it will help with monkeys and humans.

**Sleep:**
I have been out in the field with rangers. They will deal with 50 and more devices. Forgetting to turn some devices off is a common mistake. Then the battery gets drained while sitting in the basecamp or on the way in the jungle. Its no fun to realize that you carry around a few KG of empty batteries for a few days. (Not all ELOCs use solar panels)

But your doubts are true too. It can be confusing for the rangers when the device doesn't respond. It's similar with the BT time-out. Easy to get confused when i don't find the ELOC in the BT list of the [app](https://play.google.com/store/apps/details?id=de.eloc.eloc_control_panel_2) any more. And wonder what happened. (Yes. The app is online, but still work in progress.)
That's because BT timout is now set to 60 sec just for testing. But later will be set to 30 minutes or more to give enough time setting up the device. (BLE could be always on and would make life easier in that regard. But that's a huge task if i see that right.)
Also the sleep mode should happen after one hour or so and not after 5 minutes as i sugested earlier for testing. That should reduce the chances of confusion.

Waking up the device from the ON/OFF switch sounds good to me. I think that is the first thing everybody does when something doesn't respond: Restart :)

Deep sleep or intruder alarm are handy features which would make a difference in my opinion, but [this feature](https://github.com/LIFsCode/ELOC-3.0/issues/55) might be more impartant which also uses deep sleep.

@LIFsCode 
Now i feel more sorry for a even longer post :)
@OOHehir 
Thanks for the new commit. I will check it out soon.

LIFsCode, 2023-12-12:
@EDsteve @OOHehir ok let's keep this out of this issue it's getting too mix up and actually has nothing to do with the AI merge.

Let's keep the deep sleep discussion all together [here](https://github.com/LIFsCode/ELOC-3.0/issues/55)

OOHehir, 2023-12-13:
I think this task is complete, there are some power/sleep issues but now in a different issue so I'll close this to tidy things up

closed 2023-12-13 by OOHehir