LIFsCode/ELOC-3.0 #51: FW crashes when start recording from BT - master branch before merge

EDsteve, 2023-12-03:
@LIFsCode 
I am just writing it down here in case this info is helpful.
The fimrware running shuld be from this commit: fe3bf6be6ff2beac278b24af8d2ef9df6f70b6aa

Problem:
I start the ELOC, connect with BT and send the setRecordMode command. -> restart

When i start recording using the GPIO0 button. There will be no restart and the recording works fine.

### Log of the crash:

```
I (38034) main: Battery: Voltage: 3.342V, 1% SoC, Temp 31 °C
W (39586) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 798, status 0x0
W (41202) BT_HCI: hcif ssr evt: st 0x0, hdl 0x80, tx_lat 798 rx_lat 798

        | Task         | Run Time   | Percentage
        |        stats |       4095 |   0%
        |         IDLE |    5294661 |  49%
        |         IDLE |    4802127 |  45%
        |         main |      10934 |   0%
        |    BT Server |      22560 |   0%
        |         ipc1 |          0 |   0%
        |      Tmr Svc |          0 |   0%
        |     BTC_TASK |       8215 |   0%
        |       spp_tx |       3600 |   0%
        |         hciT |      12818 |   0%
        |     BTU_TASK |      34345 |   0%
        | btController |     439210 |   4%
        |         ipc0 |          0 |   0%
        |    esp_timer |       1429 |   0%
        -----------------------------------
I (43051) main: Battery: Voltage: 3.353V, 1% SoC, Temp 31 °C
I (48065) main: Battery: Voltage: 3.351V, 1% SoC, Temp 30 °C
W (48106) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0

        | Task         | Run Time   | Percentage
        |        stats |       3954 |   0%
        |         IDLE |    5436400 |  49%
        |         IDLE |    5247569 |  48%
        |         main |       8043 |   0%
        |    BT Server |      12043 |   0%
        |         ipc0 |          0 |   0%
        |       spp_tx |          0 |   0%
        |         hciT |       1617 |   0%
        |     BTU_TASK |       8160 |   0%
        |     BTC_TASK |       1105 |   0%
        |    esp_timer |       2098 |   0%
        | btController |     176124 |   1%
        |         ipc1 |          0 |   0%
        |      Tmr Svc |          0 |   0%
        -----------------------------------
I (48793) BtCmds: Start Recording...
I (48808) main: REC_REQ = 1
 123442176 KiB total drive space.
 115938144 KiB available.

 110.6 GB free
I (48820) main: Sample rate = 16000
I (48834) I2S: APLL expected frequency is 8192000 Hz, real frequency is 8191999 Hz
I (48844) I2S: DMA Malloc info, datalen=blocksize=4000, dma_buf_count=18
I (48851) main: I2s REAL clockrate in record  16000
I (48858) main: Creating session folder /sdcard/eloc/not_set1700868490711
I (48901) main: Starting session with this config:
: {
  "device": {
    "fileHeader": "not_set",
    "locationCode": "6MJWMRHV+8V",
    "locationAccuracy": 99,
    "nodeName": "ELOC_SD_CONFIG_Lexar"
[... 270 lines cut ...]
I (7571) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 10, Light sleep: ENABLED
I (7596) sleep: Enable automatic switching of GPIO sleep configuration
I (7601) main: waiting for button or bluetooth
I (7620) main: Battery: Voltage: 3.350V, 1% SoC, Temp 30 °C
I (8971) BluetoothServer: SerialBT is ready ------------------------------------------------------ 

        | Task         | Run Time   | Percentage
        |        stats |       3209 |   0%
        |         IDLE |    5185611 |  48%
        |         IDLE |    4145113 |  38%
        |    BT Server |     163850 |   1%
        |         main |      18827 |   0%
        |      Tmr Svc |          0 |   0%
        |         ipc1 |      12860 |   0%
        |         ipc0 |      23733 |   0%
        |       spp_tx |          0 |   0%
        | btController |    1055853 |   9%
        |    esp_timer |        701 |   0%
        |     BTC_TASK | Created
        |         hciT | Created
        |     BTU_TASK | Created
        -----------------------------------
I (13195) main: Battery: Voltage: 3.351V, 1% SoC, Temp 30 °C
I (18205) main: Battery: Voltage: 3.352V, 1% SoC, Temp 30 °C
```

LIFsCode, 2023-12-04:
This seems to be likely a stack overflow probably caused by a combination of `"logToSdCard": true`, and the config file on SD card.

Since the task and stack structure changed significantly I would postpone this issue unless it reappears with the current version.

EDsteve, 2024-07-21:
I will close this for now, since it didn't happen again

closed 2024-07-21 by EDsteve