LIFsCode/ELOC-3.0 #56: Random reboots after stop recording

EDsteve, 2023-12-12:
I have realized that issue already one week ago. But still happening:
When the recording gets stopped. The ELOC reboots a few seconds after. I would say almost 50% of the time.
Using ELOC 3.3 HW with latest commit (a41a3fe247dd427927a4c0f9e7e5649c60a0151b)

Here is the log which hopefully helps:
[Log_Reboot_recordoff.txt](https://github.com/LIFsCode/ELOC-3.0/files/13645196/Log_Reboot_recordoff.txt)

EDsteve, 2023-12-12:
@OOHehir Sorry to add another one. But just by accident i happened to see one more reboot. This time during recording. This reboot happaned after around 80 minutes of recording without me interfering.

Log file:
[Reboot during record.txt](https://github.com/LIFsCode/ELOC-3.0/files/13647801/Reboot.during.record.txt)

Using this config:
ELOC 3.3 HW with latest commit (https://github.com/LIFsCode/ELOC-3.0/commit/a41a3fe247dd427927a4c0f9e7e5649c60a0151b)
{
  "device": {
    "fileHeader": "not_set",
    "locationCode": "unknown",
    "locationAccuracy": 99,
    "nodeName": "ELOC_SD_CONFIG_Samsung"
  },
  "config": {
    "secondsPerFile": 60,
    "cpuMaxFrequencyMHZ": 80,
    "cpuMinFrequencyMHZ": 10,
    "cpuEnableLightSleep": true,
    "bluetoothEnableAtStart": true,
    "bluetoothEnableOnTapping": true,
    "bluetoothEnableDuringRecord": true,
    "bluetoothOffTimeoutSeconds": 30,
    "logConfig": {
      "logToSdCard": false,
      "filename": "/sdcard/log/eloc.log",
      "maxFiles": 10,
      "maxFileSize": 5120
    }
  },
  "mic": {
    "MicType": "ns",
    "MicBitShift": 11,
    "MicSampleRate": 16000,
    "MicUseAPLL": true,
    "MicUseTimingFix": true
  }
}

OOHehir, 2023-12-12:
ED,

No problem,  will take a look over the next day or two.

OOHehir, 2023-12-13:
@EDsteve I think I understand this bug, but it's one of those that's hard to be certain. I think it goes like this:

1. wav_writer is finishing the job of writing to the wav file (but task still exists)
2. I2SMEMSampler sends msg (via a msg handler) to wav_write that a buffer is full & needs to be written
3. msg arrives at wav_writer to wake task but wav_writer is finished & task no longer exists -> crash

I've pushed a new [commit](https://github.com/LIFsCode/ELOC-3.0/commit/16dfb444dd91d57348abb3c7ecbbfd7b8c27abf1), I hope this catches the issue but will take some testing

EDsteve, 2023-12-14:
@OOHehir Thanks Owen. I will do tests within the next days and report back.
BTW. I am always compiling with esp32dev and not with esp32dev-ei. Is that okay?

OOHehir, 2023-12-16:
@EDsteve, sure that's fine.

EDsteve, 2023-12-16:
@OOHehir Great. No reboots any more since your last commit. Looks good so far. Will close this issue.
But will continue to test.

closed 2023-12-16 by EDsteve