LIFsCode/ELOC-3.0 #58: Option to choose between microphone channels

EDsteve, 2023-12-18:
@LIFsCode @OOHehir 

As we know the 50pcs ELOC3.3 we ordered from JLCPCB arrived with a broken microphone. Unfortunately all of them defect. 
The plan is to still use these PCBs, not with the on board microphone but with an external one, which saves money and time. The onboard microphone is the LEFT channel. And the external would be the RIGHT channel. 

Therefor there must be an option in the app to choose between LEFT and RIGHT channel for operation. (and if it's not a big thing also add the option for stereo recording (Only IF very easy)).

That should hopefully not take too much time :)

Also adding that in the config.json in the mic section is needed:

{
  "device": {
    "fileHeader": "not_set",
    "locationCode": "unknown",
    "locationAccuracy": 99,
    "nodeName": "ELOC_NONAME"
  },
  "config": {
    "secondsPerFile": 60,
    "cpuMaxFrequencyMHZ": 80,
    "cpuMinFrequencyMHZ": 10,
    "cpuEnableLightSleep": true,
    "bluetoothEnableAtStart": true,
    "bluetoothEnableOnTapping": true,
    "bluetoothEnableDuringRecord": true,
    "bluetoothOffTimeoutSeconds": 60,
	"logConfig": {
      "logToSdCard": true,
      "filename": "/sdcard/log/eloc.log",
      "maxFiles": 10,
      "maxFileSize": 5120
    }
  },
  "mic": {
    "MicType": "ns",
    "MicBitShift": 11,
    "MicSampleRate": 20000,
    **"MicChannel: Left, Right, Stereo,**
    "MicUseAPLL": true,
    "MicUseTimingFix": true
  }
}

OOHehir, 2023-12-18:
@EDsteve I can handle the required changes in main.cpp & the I2SMEMSSampler class to handle this if @LIFsCode you don't mind handling the changes required to allow me to retrieve the setting from the configuration?

LIFsCode, 2023-12-18:
@OOHehir thanks, that makes sense. 
The wavfile writer class has already quite some dependencies to the main system, so I guess it's just consequent to let it directly retrieve it's configuration from Epic config.

I plan to lock config changes while recording or detection is active, so this will guarantee that accessing the config is thread safe.
I just delayed this task until Joseph finished the App development to give him more access for testing.

OOHehir, 2024-01-05:
@EDsteve @LIFsCode I've just pushed this [commit](https://github.com/LIFsCode/ELOC-3.0/commit/8cc4b9949c1cb9bd53aed99f9485e725fab7dcb0) which allows the I2S mic channel to be selected in [project_config.h](https://github.com/LIFsCode/ELOC-3.0/blob/8cc4b9949c1cb9bd53aed99f9485e725fab7dcb0/eloc610LowPowerPartition/src/project_config.h#L108) at compilation time. This issue seems to be a temporary issue so @EDsteve do you think meets your requirements? 

@EDsteve BTW from what I remember trying to record in stereo will naturally result in double the sample data & probably lead to increase in power & processing requirements.. This is one area already [flagged](https://github.com/LIFsCode/ELOC-3.0/issues/8) for upgrading from the legacy API.

EDsteve, 2024-01-05:
@OOHehir 
The workaround changing the channel during compilation is fine for now. I actually did that already since now changing  .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT in the config.cpp :)

Still. Later when the ELOCs get firmware updates. It will become problematic because all the firmwares will have LEFT by default. So it makes sense to have it settable from the app.
On the other hand. Microphones don't last forever in the jungle. So sooner or later some will get broken due to water damage or similar. The option to connect an external microphone and therefore change the channel in the app is needed. 

Stereo recording is not needed for our elephant purposes. I just know that some wildlife acoustic projects use stereo recordings. So it is very low priority.

LIFsCode, 2024-01-05:
@OOHehir @EDsteve I can implement the config option today

LIFsCode, 2024-01-07:
should be done with [c328154](https://github.com/LIFsCode/ELOC-3.0/commit/c3281543ae49ee014e5fcce3d68f65cad389b66a)

EDsteve, 2024-01-13:
Tested and works fine. I will close this now.

closed 2024-01-13 by EDsteve