LIFsCode/ELOC-3.0 #60: Dection mode is always ON

EDsteve, 2023-12-18:
@OOHehir @LIFsCode 

Fabi has implemented the recordingmode [responses ](https://github.com/LIFsCode/ELOC-3.0/issues/53) and these show that the ELOC is always detecting. It seems not possible to turn detecting off. It also starts right away after boot. Maybe some logic error?

INFO:
The power consumtion of 60mA "confirms" that. This was not the case when i did the testing [here](https://github.com/LIFsCode/ELOC-3.0/issues/50#issuecomment-1848952425). So the bug might have been introduced after that date.

Cheeers
ED

OOHehir, 2023-12-18:
@EDsteve Ok, will take a look

OOHehir, 2023-12-18:
@LIFsCode I can see that the state of the WAVFileWriter & AI inference is gathered in [ElocCommands.cpp](https://github.com/LIFsCode/ELOC-3.0/blob/3e73585ba4ba654fce719c59377ed3d14e769389/eloc610LowPowerPartition/src/Commands/ElocCommands.cpp#L61) but I'm not sure how you'd like these states to be discoverable from the config on boot? Would you like me to extend the functionality in ElocConfig.cpp (or some other file/ class)  so these parameters can be parsed?

LIFsCode, 2023-12-18:
@OOHehir I think that's maybe a but misunderstanding. Per default the ELOC should be idle after boot, i.e. detection and recording should be off. There should be no config option to automatically start neither recording nor detection without the Bluetooth app.

There has been a new [request](https://github.com/LIFsCode/ELOC-3.0/issues/55) about deep sleep:
1. Deep sleep after inactivity: not relevant for the operation mode
2. Deep sleep due to low battery: In that case we'll have to store the current operation mode in order to resume it. I personally would prefer a simple file (definitely not the config) on the SPIFFS with the operation mode. All the rest could be recovered from config file. However I would delay that to when the deep sleep is added, as it would require separate handling of the boot process and system setup. Here are also other problems to be solved as file system integrity during brown out.

So for the first run simply keep the device idle after boot :)

OOHehir, 2023-12-19:
@LIFsCode Apologies, my misunderstanding!

@EDsteve Should be resolved now, both the wav writing & AI are idle startup (both the EI & non-EI build). The perf monitor for both builds at idle:

Non-EI build
```
        | Task         | Run Time   | Percentage
        |        stats |       2090 |   0%
        |         IDLE |    4986869 |  49%
        |         IDLE |    4783043 |  47%
        |         main |      18609 |   0%
        |         ipc0 |          0 |   0%
        |    esp_timer |          0 |   0%
        |    BT Server |          0 |   0%
        |         ipc1 |          0 |   0%
        |      Tmr Svc |          0 |   0%
        -----------------------------------
```
The EI build:
```
       | Task         | Run Time   | Percentage
        |        stats |       2330 |   0%
        |         IDLE |    5010978 |  50%
        |         IDLE |    4995127 |  49%
        |         main |      13522 |   0%
        |      Tmr Svc |          0 |   0%
        |         ipc1 |          0 |   0%
        |    BT Server |          0 |   0%
        |    esp_timer |          0 |   0%
        |         ipc0 |          0 |   0%
        |    ei_thread |          0 |   0%
        -----------------------------------
```
Please note the 'ei_thread' is a blocked state & awaits a signal from the wav writer task so consumes no cycles.

I've modified the I2SMEMSampler thread, it only runs 'on-demand' now, that is when either the wav writer or EI task is running, otherwise it terminates & will restart  when either the  wav writer or EI task starts.

If this doesn't result in the expected power savings the only possible cause I can see is that due that due to the EI task now running in a [separate task](https://github.com/LIFsCode/ELOC-3.0/commit/3e73585ba4ba654fce719c59377ed3d14e769389) the cores spend longer in the function:

```
void delay(int ms)
{
    ESP_LOGV(TAG, "Func: %s", __func__);

    vTaskDelay(pdMS_TO_TICKS(ms));
}
```

OOHehir, 2023-12-19:
@LIFsCode @EDsteve With regards further power savings while the device is NOT in deep sleep, the only 'low-hanging' fruit I can see is possibly inhibiting the I2S peripheral when now required. What power savings this would produce I don't know.

OOHehir, 2023-12-19:
@EDsteve @LIFsCode It looks like disabling the I2SMEMSampler thread has introduced some bug, intermittently the I2S buffer seems to be corrupted, working on the issue..

OOHehir, 2023-12-20:
@EDsteve @LIFsCode Still working on this particular bug.. seems like at some point the after stopping then restarting the I2S peripheral the buffer is just filled with rubbish, lots of warnings about noise clipping. I suspect it might be the

1. mic going into some sort of sleep mode & not producing valid data or
2. the i2s peripheral not starting correctly.

I'll look at the mic datasheet & see if I can narrow down the issue.

LIFsCode, 2023-12-20:
@OOHehir we didn't have such issues earlier, so I assume it has something to do with the rework of the I2S sampler

My guess would be that the driver is not correctly stopped, so the HW I2S engine is still running causing data corruption. Maybe completely uninstalling the driver would be worth a try.

If I remember correctly, in previous versions the i2s driver is completely uninstalled when stoped.

OOHehir, 2023-12-21:
@LIFsCode @EDsteve I believe I've found a reasonable working solution. I've modified the code so that the I2S only starts if the wav writer or AI requires it, but then continues, even if the 'demand' ceases. If necessary I'll continue to work on stopping & restarting the I2S peripheral but I think this will meet the immediate requirements. Ill commit the code this evening or tomorrow (it's currently in the ai-task branch).

OOHehir, 2023-12-22:
This [fix](https://github.com/LIFsCode/ELOC-3.0/commit/02a5ff45516dd2a2f3fc1f41cdbf8db7790fb35f) has been pushed, looks OK during my testing.

EDsteve, 2023-12-23:
@OOHehir @LIFsCode 
Have done some testing and you are right. No detection any more after boot. So that's good :)
Unfortunately the last commit has introcued more changes inlcuding some bugs. Not sure why that is. But i think it's better to implement things one by one?

1. The ELOC goes into a sleep mode (5mA) when BT is off
Of course i like the [sleep mode](https://github.com/LIFsCode/ELOC-3.0/issues/55#issuecomment-1852526825). Needs some adjustments though. But why is it suddenly back? Was that in purpose?
3. Seems detection is not working any more. According to the getStatus responses detection is never active. 
INFO: When i use a command which turns ON detection. Power consumption jumps to around 100mA for around 10 seconds and then back to 25mA.

|Command                               |Direct response:                                                                                                                                     |getstatus response                                                                                                                        |
|--------------------------------------|-----------------------------------------------------------------------------------------------------------------------------------------------------|------------------------------------------------------------------------------------------------------------------------------------------|
|setrecordmode#mode=recordOFF_detectOFF|{"ecode" : 0, "error" : "ESP_OK", "cmd" : "setrecordmode", "payload" : { recordingState: { val: 1, state: "recordOff_detectOff" } }}                 |  "session": {     "identifier": "",     "recordingState": {       "val": 1,       "state": "recordOff_detectOff"     },                   |
|setrecordmode#mode=recordOFF_detectON |{"ecode" : 0, "error" : "ESP_OK", "cmd" : "setrecordmode", "payload" : {   "recordingState": {     "val": 4,     "state": "recordOff_detectOn"   } }}|  "session": {     "identifier": "",     "recordingState": {    **"val": 1,       "state": "recordOff_detectOff"**    },                   |
|setrecordmode#mode=recordON_detectON  |{"ecode" : 0, "error" : "ESP_OK", "cmd" : "setrecordmode", "payload" : {   "recordingState": {     "val": 3,     "state": "recordOn_detectOn"   } }} |  "session": {     "identifier": "not_set1702641580020",     "recordingState": {      **"val": 2,       "state": "recordOn_detectOff"**    },|
|setrecordmode#mode=recordON_detectOFF |{"ecode" : 0, "error" : "ESP_OK", "cmd" : "setrecordmode", "payload" : {   "recordingState": {     "val": 2,     "state": "recordOn_detectOff"   } }}|  "session": {     "identifier": "not_set1702641580020",     "recordingState": {       "val": 2,       "state": "recordOn_detectOff"     },|
|setrecordmode#mode=recordOnEvent      |{"ecode" : 0, "error" : "ESP_OK", "cmd" : "setrecordmode", "payload" : {   "recordingState": {     "val": 5,     "state": "recordOnEvent"   } }}     |  "session": {     "identifier": "not_set1702641580020",     "recordingState": {      **"val": 0,       "state": "recInvalid"**   },    |

OOHehir, 2023-12-23:
@EDsteve My understanding is that deep sleep is available now because the I2S isn't running.

With regards to the detection, I assume you're using the 'espdev-ei' build? It won't be available in the 'espdev' build.

EDsteve, 2023-12-23:
@OOHehir Okay. Thanks. I get it with the sleep mode.

Detection:
I was using espdev32 build in my last post.
Now i have changed to espdev-ei and detection "works" but spits out errors. Sometimes it does make Classification.
Just for information: During recording, the power consumption is around 2.5mA higher with espdev-ei compared to espdev32 build.

A few logs which hopefully make it more easy to find the devil inside the code :)

- [RecordOFF_detectON](https://github.com/LIFsCode/ELOC-3.0/files/13758531/RecordOFF_detectON_Error_log.txt)

- [RecordON_detectON](https://github.com/LIFsCode/ELOC-3.0/files/13758529/RecordON_detectON_Error_log.txt)
 
- [Reboot after BT timed out](https://github.com/LIFsCode/ELOC-3.0/files/13758528/Reboot.after.BT.turns.off.txt) (could not reproduce, but happened once)

OOHehir, 2023-12-23:
@EDsteve I had a look through the logs & they all seem to have one thing in common, lack of memory. This was touched on here [briefly.](https://github.com/LIFsCode/ELOC-3.0/issues/50#issuecomment-1845652521). As proposed by @LIFsCode ideally all the objects/ buffers would be [statically allocated](https://www.geeksforgeeks.org/difference-between-static-and-dynamic-memory-allocation-in-c/) (at boot up) so we would have a better idea of the RAM usage at compile time (rather than runtime). I'll create an issue to move the SD card & wav writer to static allocation but given that you want to deploy devices soon I think its better to limit changes to bug removal at the moment...

With that in mind I've just added a [commit](https://github.com/LIFsCode/ELOC-3.0/commit/29325f68d7c831e7b754bf81b368c9930524c5eb) that:
1. Reduces the wav buffer size
2. Makes the edge-impulse classifier a [static allocation](https://docs.edgeimpulse.com/docs/run-inference/cpp-library/deploy-your-model-as-a-c-library#static-allocation). (It results in a approx 1% RAM usage change.)

I've tried the espdev-ei build & tested using a BT connection, toggling wav writing & AI inference, no issues so far for me.

EDsteve, 2024-01-05:
Sorry for the late response. But i have have done some testing with it and seems all is working now :)

closed 2024-01-05 by EDsteve