LIFsCode/ELOC-3.0 #62: PCB HW: SD Card detect

LIFsCode, 2024-01-04:
The wiring of the SD card detect pin is wrong. It should be Pin **9**. It is a bit lisleading in the datasheet.

This drawing makes it clear:
[image]

Currently it is wired to pin 2 which is connected to the SD card itself.

EDsteve, 2024-01-05:
Oh i see. Didn't realize there are two CD pins.
It seems that both pins (2 and 9) have a Card Detect purpose. Not sure how it works with pin2 but pin9 is a mechanical switch to GND from the SD card holder. Which is the way to go.
I have implemented the changes. Luckily an easy fix. 

[image]
[image]

Regarding the 50 ELOC PCBs with wrong wiring (ELOC 3.3). Would it work if i solder a wire between pin2 and pin9?Mmaybe i will just try and see :)

LIFsCode, 2024-01-05:
@EDsteve the new schematic looks good.

Concerning shortening pin 2&9 for testing: I don't think it's a good solution.

This is may switch the SD card into SPI mode (see [here](https://community.nxp.com/t5/i-MX-Processors/SD-card-detection-with-multiplexed-CD-DAT3/m-p/234466)) so I guess this could cause more problems and also some hidden issues which may arise only sporadically.

There could be a solution adapting the [SDIOMmc driver[(https://esp32.com/viewtopic.php?t=15604) but this would take me a while and I think time could be better invested on other topics.

@EDsteve if you want this feature on ELOC 3.3 HW cut the wire to the pin 2 (use a scalpel und cut the wire on the board between the via und the pin2 of the SD card socket) and wire it from the via to pin 9.
Instead you can also try to lift the pin 2 of the SD card socket and thus isolate it from the pad. 

Leaving the pin unconnected will most likely cause problems at least makes it vulnerable to EMI issues, so for testing the additional pullup must be wired as well.

EDsteve, 2024-01-05:
@LIFsCode 
The via is not acessable because it's underneath the SD card holder. The only other option i see is to connect it directly to the IO expander. But that's not easy to solder. It seems the ELOC 3.3 has to live without SD Detect.

I guess now is the right time to get [NVS](https://github.com/LIFsCode/ELOC-3.0/issues/37) running. Not really sure how to do that yet but i will take a look.

LIFsCode, 2024-01-05:
@EDsteve : yes that's exactly why I wanted this NVS information.

Actually in a basic way it is already prepared. Only the serial number increment is not implemented. But the hw revision & generation fields are static and this is already implemented to be generated, which is the mandatory stuff :)

EDsteve, 2024-01-16:
After flashing the NVS.bin it works as expected. 
I (1938) ElocSystem: Reading values from NVS done - all OK
I (1942) ElocSystem: Factory Data: Serial=1240100001, HW_Gen = 3, HW_Rev = 3

closed 2024-01-16 by EDsteve