LIFsCode/ELOC-3.0 #63: Strange LED and buzzer behaviour

EDsteve, 2024-01-06:
@LIFsCode 

- [ ] The blue status LED is flashing for recording. Which is correct. But it also flashes for detecting (recordoff_detecton). 
Only the battery LED should flash for detecting. 
- [ ] During recording/detecting: The knock sensor activates the LEDs. That's good. But sometimes (maybe 5% of the time) it takes more than 10 seconds for the LEDs to show the status after knocking was detected, which makes the user think the device is OFF.
- [ ] Random beep and random LED flashes during recordon_detecton or recordon_detectoff. This does not happen right away. Takes a while. Still investigationg it further if it is connected to BT.
   - recordon_detecton ([Video](https://photos.app.goo.gl/kajQr8fNEnL2MhM68))
The buzzer beeps when Audio samples clipping. But there should not be any clipping because there is no loud sound -> another issue involved?
   - recordon_detectoff ([Video](https://photos.app.goo.gl/rfj6xxZ3f8hhcW2z8))

I think i will also rename these LEDs on the PCB to Blue and Red. That's more universal and avoids confusion. Open for name suggestions though :)

Compiled with esp32dev-ei-windows
Harware: ELOC 3.2
This is the boot log:
```
I (29) boot: ESP-IDF 4.4.4 2nd stage bootloader
I (29) boot: compile time 13:37:56
I (29) boot: chip revision: v3.0
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (77) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (84) boot:  2 partition0       OTA app          00 10 00010000 007e0000
I (92) boot:  3 partition1       OTA app          00 11 007f0000 007e0000
I (99) boot:  4 spiffs           Unknown data     01 82 00fd0000 00030000
I (107) boot: End of partition table
I (111) boot_comm: chip revision: 3, min. application chip revision: 0
I (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=51fb0h (335792) map
I (248) esp_image: segment 1: paddr=00061fd8 vaddr=3ff80000 size=00004h (     4) load
I (248) esp_image: segment 2: paddr=00061fe4 vaddr=3ffbdb60 size=0492ch ( 18732) load
I (261) esp_image: segment 3: paddr=00066918 vaddr=40080000 size=09700h ( 38656) load
I (278) esp_image: segment 4: paddr=00070020 vaddr=400d0020 size=e7894h (948372) map
I (622) esp_image: segment 5: paddr=001578bc vaddr=40089700 size=15388h ( 86920) load
I (672) boot: Loaded app from partition at offset 0x10000
I (673) boot: Disabling RNG early entropy source...
I (685) psram: This chip is ESP32-D0WD
I (685) spiram: Found 64MBit SPI RAM device
I (685) spiram: SPI RAM mode: flash 40m sram 40m
I (690) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (697) cpu_start: Pro cpu up.
I (701) cpu_start: Starting app cpu, entry point is 0x400851b8
I (0) cpu_start: App cpu up.
I (1573) spiram: SPI SRAM memory test OK
I (1606) cpu_start: Pro cpu start user code
I (1606) cpu_start: cpu freq: 80000000
I (1606) cpu_start: Application information:
I (1609) cpu_start: Project name:     idf-wav-sdcard
I (1614) cpu_start: App version:      8cc4b99-dirty
I (1620) cpu_start: Compile time:     Jan  6 2024 13:36:26
I (1626) cpu_start: ELF file SHA256:  23e42b52c3822a4f...
I (1632) cpu_start: ESP-IDF:          4.4.4
I (1637) heap_init: Initializing. RAM available for dynamic allocation:
I (1644) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1650) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1657) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1663) heap_init: At 3FFD3A08 len 0000C5F8 (49 KiB): DRAM
I (1669) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1675) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1682) heap_init: At 4009EA88 len 00001578 (5 KiB): IRAM
I (1688) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1698) spi_flash: detected chip: gd
I (1701) spi_flash: flash io: dio
I (1716) sleep: Configure to isolate all GPIO pins in sleep state
I (1717) cpu_start: Starting scheduler on PRO CPU.
[... 191 lines cut ...]
I (9997) main: Test model predictions:
I (10005) main:     (DSP: 1400 ms., Classification: 44 ms., Anomaly: 0 ms.)
I (10009) main:     background: 0.207031
I (10012) main:     trumpet: 0.792969
W (10018) main: Test of trumpet sample appears to be poor, check model!
I (10022) main: Running test category: other
I (10301) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (10329) main: Test model predictions:
I (10334) main:     (DSP: 246 ms., Classification: 45 ms., Anomaly: 0 ms.)
I (10338) main:     background: 0.996094
I (10342) main:     trumpet: 0.000000
I (10448) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 10, Light sleep: ENABLED
I (10453) sleep: Enable automatic switching of GPIO sleep configuration
I (10460) I2SMEMSSampler: i2s_port = 0
I (10464) I2SMEMSSampler: i2s_sampling_rate = 16000
I (10469) I2SMEMSSampler: mBitShift = 11
I (10474) I2SMEMSSampler: mListenOnly = 0
I (10480) I2SMEMSSampler: m_fixSPH0645 = 1
 123442176 KiB total drive space.
 120850464 KiB available.

 115.3 GB free
I (10490) WAVFileWriter: Allocating 2 buffers of 6144 bytes in RAM
I (10997) main: Battery: Voltage: 3.332V, 1% SoC, Temp 36 °C
```

LIFsCode, 2024-01-06:
@EDsteve

1. Yes sorry I enabled the wrong led for detection only. My fault I'll change chat

2. It's not beeping because of clipping but it's it's clipping because of beeping. It's beeping because, e.g. BT gets enabled during a knock.
This is a issue addressed earlier and requires a conceptional solution. I would doubt you're happy with clipping only just because a monkey is knocking on the ELOC. --> I would rather limit the buzzer behavior to only the least possible use cases.

3. Big problem: this is due to the thread priority architecture. Detection and WAV writing run with highest priority as this is the most essential functionality. But this blocks the status update from execution depending on system load. This is what I told you earlier when discussing the LED behavior that timing inaccuracy may occur.
This can be changed to guarantee faster responses, but comes with the price of potentially loosing samples. Unless we have a reliable [buffer overflow detection](https://github.com/LIFsCode/ELOC-3.0/issues/18) this is not what I would recommend. But I've you prefer it the other way I can change it it's up to you

EDsteve, 2024-01-06:
Yes. I remember now. You mentioned the problem with buzzer and clipping and i agree to keep the buzzer at a minimum.
On the other hand. Anything that is knocking on the case creates unusable recording. When a monkey or anything else is playing with the ELOC. The recording won't be usable due to the knocking. With or without the buzzer. 

>    This is what I told you earlier when discussing the LED behavior that timing inaccuracy may occur.
This can be changed to guarantee faster responses, but comes with the price of potentially loosing samples. Unless we have a reliable [buffer overflow detection](https://github.com/LIFsCode/ELOC-3.0/issues/18) this is not what I would recommend. But I've you prefer it the other way I can change it it's up to you

Hmm. I see what you mean. Sorry that i forget things easily sometimes. Too often actually.
According to my tests. The LEDs work almost normal right after starting recording/detecting.
But after a few hours recording/detecting, the LED and buzzer indicator show random behaviour (flashing for more than one hour randomly,  continuously ON for a minute or beebing for 15 minutes in 8 seconds intervalls) and i was knocking only once. It's not only inaccurate. It seems random and instead of indicating a status. It confuses the user and me too :)

So if i understand you corectly. We either turn off all indicators during recording because it does not make sense to keep it as is. Or we loose sound samples when the knock sensor gets triggerd. With a few sound samples you mean a few milliseconds during the time the LEDs are flashing, correct?

In that case i would go with loosing samples. Because:
There are a lot of things which creates useless recordings. Strong wind, ants trying to get into the microphone, birds picking foam from the protection, heavy rain,.... Compared to that, loosing a few samples due to the indicators is an okay trade off. 
Hope that doesn't involve too much changes.

LIFsCode, 2024-01-07:
I have this working here without any problems. Sorry cannot reproduce.

EDsteve, 2024-01-08:
@LIFsCode 
Hmm. That is strange. I have done three long term tests (between 5 and 12 hours) with two devices showing the same behaviour. 
Have you had recordon_detecton running for more than 5 hours in your test? And did a knock in between?

I will do more tests and document all steps with log files and videos.

EDsteve, 2024-01-09:
@LIFsCode 
I think i figured out the reason for the many peeps and the flashes at long term recordings. 
It's connected to low battery status. Because the peeps and flashes were not consistent (came and go. Still not sure why) and the battery indication should only happen when triggered by the knock sensor. I didn't realize that the beep is connected to the low battery.

So it seems the status LEDs work fine. Only change needed is that the battery indication should only show when knocking on the case. Otherwise it would beep every few seconds on low battery.
But Fabi mentioned we have buffer problems which definitely higher priorities than the buzzer.