LIFsCode/ELOC-3.0 #64: Modification: add  "detectingTime[h]": 0.00,  "detectedEvents": 133, "aiModel": "Trumpets2.4", "totalDetectingTime[h]": 0.00,

LIFsCode, 2024-01-07:


OOHehir, 2024-01-11:
@LIFsCode Home tomorrow evening. Should get a chance to look at this latest Monday

OOHehir, 2024-01-11:
@LIFsCode @EDsteve With regards the parameters detectingTime[h]" & "totalDetectingTime[h]", say the inference requires 3 secs of audio for each sample to run. Characteristic sounds (trumpet or rumble or other) are detected as follows:

13:01:00 (1 event of characteristic sound)
13:05:30 (2 events of characteristic sound)
13:15:45 (1 event of characteristic sound)

What should the parameters report?

EDsteve, 2024-01-11:
@OOHehir 
We have the same for recording time already implemented.
This information is for the rangers, who check the device every two weeks, so they can see in the app how long the device has been in detection/recording-mode already. 

- "detectingTime[h]": Represents the time the device has been in detection mode since detecting was started the last time. 
- totalDetectingTime[h]: Represents the total time the ELOC was in detecting-mode since boot. (This time should equal to: Total ELOC runtime since boot minus the time the device was not detecting)

LIFsCode, 2024-01-12:
@josephx86 have you already implemented these status fields? if yes what is the exact spelling of the fields you are parsing?

if not, I would prefer structured  fields. Are you fine with this:

```
    "detection": {
      "detectingTime[h]": 0.00,
      "detectedEvents": 133,
      "aiModel": "Trumpets2.4",
      "aiState": {
         "val":1,
         "state": "on"
      }
    }
```

which would look like this in the whole status message:

```
{
  "battery": {
    "type": "BAT_LiFePo Battery",
    "state": "Full",
    "SoC[%]": 95,
    "voltage[V]": 3.34
  },
  "device": {
    "firmware": "ELOC_V0.1",
    "timeStamp": 0.01,
    "totalRecordingTime[h]": 0.00,
    "totalDetectingTime[h]": 0.00,
	"SdCardSize[GB]": 1.83,
    "SdCardFreeSpace[GB]": 1.74,
    "SdCardFreeSpace[%]": 94.86
  },
  "session" : {
     "identifier":"",
     "recordingState": {
        "val":1,
        "state":"recordOff_detectOff"
     },
    "recordingTime[h]": 0,
    "detection": {
      "detectingTime[h]": 0.00,
      "detectedEvents": 133,
      "aiModel": "Trumpets2.4",
      "aiState": {
         "val":1,
         "state": "on"
      }
    }
  }
}
```

If you prefer something else, or have already implemented something please let me know

josephx86, 2024-01-12:
It looks like I have not implemented some of those fields... I think. I am on my phone right now but the fields I have implemented can be seen at https://github.com/EDsteve/ELOC-Control-Panel/blob/94b9d822b7f21d5ee5ae9951038e3d4efa05e58c/app/src/main/java/de/eloc/eloc_control_panel/driver/DeviceDriver.kt

They are the constants that start with KEY_

I will double check when I get on the computer.

OOHehir, 2024-01-14:
@EDsteve @LIFsCode I've added some of these parameters 

    "detectingTime[h]": 0.00,
    "totalDetectingTime[h]": 0.00,

with this [commit](https://github.com/LIFsCode/ELOC-3.0/commit/7ed20d087724f2a730a133a391c8ae27fc38a8a7). The time is currently returned as a float of hrs. I'll work on the other two tomorrow.

LIFsCode, 2024-01-15:
@josephx86 

With this [this commit](https://github.com/LIFsCode/ELOC-3.0/commit/86020f7f5e012621e81e006d16cca5f1e6529acd) the status response looks like this:
```
{
  "battery": {
    "type": "no Battery",
    "state": "Full",
    "SoC[%]": 95,
    "voltage[V]": 3.34
  },
  "session": {
    "identifier": "",
    "recordingState": {
      "val": 1,
      "state": "recordOff_detectOff"
    },
    "recordingTime[h]": 0,
    "detection": {
      "state": false,
      "detectingTime[h]": 0,
      "detectedEvents": 0,
      "aiModel": "trumpet trimmed"
    }
  },
  "device": {
    "firmware": "ELOC_V0.1",
    "Uptime[h]": 0.012889775,
    "totalRecordingTime[h]": 0,
    "SdCardSize[GB]": 1.83,
    "SdCardFreeSpace[GB]": 1.7,
    "SdCardFreeSpace[%]": 92.98
  }
}
```

is that ok?
@josephx86 : what is your prefered way of handling the status response if a non-AI version of the ELOC is running (AI not part of the firmware) should we remove it or fill it with empty values?

josephx86, 2024-01-15:
@LIFsCode the status response is okay. For the non-AI responses, let's fill it with empty values. And my understand of 'empty values' is 0 for integers, 0.0 for floats/doubles, false for booleans, and "" for strings. Did I understand that the way you meant?

LIFsCode, 2024-01-16:
@josephx86 perfect yes that's the way it's implemented 👍