LIFsCode/ELOC-3.0 #65: AI may fail EI_IMPULSE_DSP_ERROR

LIFsCode, 2024-01-07:
@OOHehir I just run into some problems with the EI running in parallel with the recording.

1. mode=recordOn_DetectOff--> all works fine, except of some wav buffer overruns at the beginning
2. mode=recordOff_DetectOn --> all fine
3. mode=recordOn_DetectOn --> I get these errors ```
ERR: MFE failed (-1002)
ERR: Failed to run DSP process (-1002)
I (187203) main: Cycles taken to run inference = 971009
E (187218) main: ERR: Failed to run classifier (-5)
```

I attached the [log file](https://github.com/LIFsCode/ELOC-3.0/files/13854488/device-monitor-240107-191306.log)

and the [config file](https://github.com/LIFsCode/ELOC-3.0/files/13854493/not_set1704653169181.config.txt)

Do you have any idea about what coud has caused these errors? 
I have not seen these errors with commit c3281543a, but since I merged the battery calibration stuff I see them (commit d093f7476)

But I don't see what the battery stuff has to do with the EI, the battery is only read in the main loop. So I guess it is more a sideeffect which triggered this behavior.

OOHehir, 2024-01-07:
@LIFsCode The output
ERR: Failed to run classifier (-5)

Is normally a sign of insufficient memory.

LIFsCode, 2024-01-07:
Ok that's a good hint. I'll check for unreleased memory or heap fragmentation.

OOHehir, 2024-01-11:
@LIFsCode A quick & simple check to see if memory is the issue is to modify the WAVFileWriter buffer size [here](https://github.com/LIFsCode/ELOC-3.0/blob/cf51c0378c7afd5d93d96bc6fc62fe502a3b9fc0/eloc610LowPowerPartition/lib/wav_file/src/WAVFileWriter.h#L18). You could reduce if from 1024 * 3 to 1024 * 2 which would free 2048 bytes (it uses a double buffer scheme). I haven't checked if it would impact the wav file performance.

LIFsCode, 2024-01-11:
I guess it's related with the STD::map structure I'm using for storing the calibration values. I'll replace that with something statically allocated.

LIFsCode, 2024-01-12:
Unfortunately it was not the use of std::map, I will do some more debugging, currently I have no guess what has caused this.

OOHehir, 2024-01-14:
@LIFsCode I've noticed there can be a significant time between sending a command via BT to the ELOC & it registering on the terminal when toggling on & off the AI/ EI status. I believe it could result in a race condition with starting & stopping the associated thread. I've tried to improve the code to make it more resilient with this [commit ](https://github.com/LIFsCode/ELOC-3.0/commit/7ed20d087724f2a730a133a391c8ae27fc38a8a7)& looks like its improved but will do more testing tomorrow.

OOHehir, 2024-01-15:
I've reduced the size of the I2S DMA buffers in the commit above from 18 * 1000 to 8 * 1024. This frees 9808 bytes & appears, at least on initial testing, to resolve this issue, supporting the theory that its memory exhaustion issue.

I've just carried out sound recording at 16kHz & the performance seems unaffected. The worse case scenario is probably at max sampling rate (44kHz?) which needs to be tested.

LIFsCode, 2024-01-15:
Yes actually that solved the issue with the DSP_ERROR. So I guess the memory footprint should be fine at least for the moment.

I created a separate [issue ](https://github.com/LIFsCode/ELOC-3.0/issues/69) concerning the potential performance issue to not mix things up.

commit 92497792 (2024-01-15): Optimise (reduce) size of I2S DMA buffers
May help to resolve #65
closed 2024-01-15 by LIFsCode