LIFsCode/ELOC-3.0 #68: EI_Results.csv adjustments

EDsteve, 2024-01-15:
@OOHehir 
It's time to give the CSV file a new home :)

- [ ] Insted of one CSV file in `/eloc`. It would be better if every session-folder get's it own CSV file.

- [ ] The Prediction threshold should be increased to a higher (adjustable) value before we get too big CSV files on the SD card.
 For example: Only predictions >0.60 are getting written to the CSV file.
- [ ] Question: Are these irregular time stamps in the CSV file normal? Or should the predictions happen in a more constant interval? Not sure :)
Example using this commit 92497792e591c9f4b4a28a8f8410080cade57051
[EI-results-ID-290754-DEPLOY-VER-14.csv](https://github.com/LIFsCode/ELOC-3.0/files/13942018/EI-results-ID-290754-DEPLOY-VER-14.csv)

OOHehir, 2024-01-15:
@EDsteve 

- [ ] Will do
- [ ] The prediction threshold can be set in [project_config.h](https://github.com/LIFsCode/ELOC-3.0/blob/92497792e591c9f4b4a28a8f8410080cade57051/eloc610LowPowerPartition/src/project_config.h#L190) as required.
- [ ] I believe the irregular time stamps were bug related. With the latest commit they're occurring ever second, e.g. 
[EI-results-ID-290754-DEPLOY-VER-14.csv](https://github.com/LIFsCode/ELOC-3.0/files/13943150/EI-results-ID-290754-DEPLOY-VER-14.csv)

OOHehir, 2024-01-16:
@EDsteve 

1. The EI-results.. file is now created inside the session folder
2. The EI-results.. file is only created when it is determined that a characteristic sound has exceeded the required threshold, so if the AI detection is running but no characteristic sound is deemed to have been detected this file won't exist
3. The session folder is only created if the above file is created OR a wav file is created

closed 2024-02-02 by OOHehir