LIFsCode/ELOC-3.0 #71: Missing sound samples and recordings are a bit too long

EDsteve, 2024-01-28:
@OOHehir @LIFsCode 
There are two issues with sound recordings. Hope this is an easy one to fix :)

### 1. Recordings too long
A 60 second recording does record 60 seconds plus a few ms (See yellow arrows in point two). This results in these inconsistent time stamps.

| Should look like this | That's how it is now |
|------------|------------|
| 18:27:53   | 18:27:53   |
| 18:28:53   | 18:28:53   |
| 18:29:53   | 18:29:54   |
| 18:30:53   | 18:30:55   |
| 18:31:53   | 18:31:55   |
| 18:32:53   | 18:32:56   |

[image]

### 2. Missing sound samples
With this commit (d2dd26c8f8e37bfb0c3e9861ef266469df3d2101) i get missing sound samples between each recorded sound file. This didn't happen with this commit (cf51c0378c7afd5d93d96bc6fc62fe502a3b9fc0).

I recorded an interval of peeps for > 2 minutes and opened the first and second minute next to each other in a sprectrogram viewer to see if the sound has missing samples. In the spectrogram you can see the difference between these two commits when you look at the beeping interval. (4 beeps and a pause).

[image]

EDsteve, 2024-01-28:
The spectrogram shows the problem. But if you want to see the sound files yourself:
[not_set1706443569745_2024-01-28_12_06_10.zip](https://github.com/LIFsCode/ELOC-3.0/files/14075159/not_set1706443569745_2024-01-28_12_06_10.zip)

LIFsCode, 2024-01-28:
@EDsteve just to make sure I understand it correctly: 

missing samples means you are missing a whole block of sound data, not some sporadic spread missing samples, as we have seen with the 44kHz recording and [apll use with power management](https://github.com/LIFsCode/ELOC-3.0/issues/30#issuecomment-1816570875) the later would result in distorted spectrum. 
From what I see in the screenshots it looks more like a block of data missing but the spectrum is clean.
Can you specify the length of data which is missing?

Do you see any buffer overruns in the log?

EDsteve, 2024-01-29:
@LIFsCode @OOHehir 
1.  It doesn't matter if it's exactly one minute or not as long the time stamps show the right time. So let's leave it as it is. No more complexity for now :)

2. You are right Fabi. It seems to be connected to a wav buffer overrun. I have just done more tests. And evry time there is a `wav buffer overrun` in the log. There is also the missing sound block. 
The missing sound block is around 390ms long.

   This problem is not happening all the time. But i would say 90% of the time.

   Here is the log:
[Missing_blocks.txt](https://github.com/LIFsCode/ELOC-3.0/files/14080682/Missing_blocks.txt)

OFFTOPIC:
In this log file i got also an error with the time synch. Which i have never seen before. Is that a firmware issue or an app issue?
`I (44932) BtCmds: updating time with {"seconds":1706509716, "ms" :setConfig`
`E (45047) BtCmds: Parsing time config failed with InvalidInput!`
Might this be connected to the issue? 
`[ 22254][E][BluetoothSerial.cpp:342] esp_spp_cb(): RX Full! Discarding 23 bytes`

EDsteve, 2024-01-29:
The wav buffer overrun seem to be happening only when a new files is written. Which is bad if the file size is 1 minute.
But the rangers will use 1h sound files. So if that issue is too complex to solve and/or you have more important improvements to do. It's not a too big deal if 390ms are missing every hour - for now. :)

OOHehir, 2024-01-29:
@EDsteve I concentrate on solving the wav buffer overrun first, then come back to the file length.

EDsteve, 2024-01-29:
@OOHehir I think the [reboot issues](https://github.com/LIFsCode/ELOC-3.0/issues/66#issuecomment-1914410673) should be solved first before we deal with the file length, which adds complexity :)

OOHehir, 2024-01-29:
@EDsteve @LIFsCode I've made some changes which should improve the wav buffer overrun issue. An outstanding issue to resolve is to continously check the SD card free space as each file is created. There's a few functions in main.cpp that could be moved into SDCardSDIO.cpp which should allow this to be achieved in a cleaner method.