LIFsCode/ELOC-3.0 #73: Rework Logging to sd card to a RAM buffered mode

LIFsCode, 2024-02-03:
Direct writing to SD card has 2 major drawbacks:

1. it adds a log of stack & performance penalty to all functions due to integration in ESP_LOG()
2. it has a significant perfformance impact due to a lot of small size file writing

This can be reworked:

- [ ] Use only a single printf() function and output the printed buffer to both static RAM for storage and stdio
- [ ] Buffer log messages in a static RAM buffer and write it only to SD card by explicitly calling a store command which can be run from a low priority status task
- [ ] Use a [external stack](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/system/esp_function_with_shared_stack.html) for the printf() to avoid stack effects on other functions

OOHehir, 2024-02-03:
@LIFsCode I'm sure you've considered this.. 

Is the driving factor behind logging to the SD tracing bugs? If the output is buffered it won't be written to the SD card when a reset occurs so this output will be lost.. It seems [RTC memory](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-guides/memory-types.html?highlight=rtc#rtc-fast-memory) is persistent during resets but I don't know if thats a viable option.

As I say, I'm sure you've considered this, but just in case!

LIFsCode, 2024-02-03:
@OOHehir I know that is a problem, either getting reliable and full log or getting good performance. That's why currently after every log message the buffer is flushed.

Regarding the RTC memory, thanks that's a really good hint but so far I don't have a proper idea how to use it since it is only accessible from core 0 (PRO CPU) which makes it kind of blind for core 1. 

This is something to balance out.

OOHehir, 2024-04-09:
@LIFsCode Might be of some use,[ 'noinit' DRAM](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/memory-types.html#noinit-dram) should retain its value from resets.