LIFsCode/ELOC-3.0 #74: SD - mount state & free memory

OOHehir, 2024-02-03:


OOHehir, 2024-02-03:
@LIFsCode Please let me know if you think this is worthwhile

LIFsCode, 2024-02-04:
@OOHehir great we have the same thoughts about that.

Actually I already had prepared that for making the sdcard a member of ElocSystem class so that it holds all the external HW peripherals. I just delayed the implementation until the SD card detect is [correctly wired](https://github.com/LIFsCode/ELOC-3.0/issues/62).

I wouldn't add another task just for the SD card handling, rather add it to [ElocSystem::handleSystemStatus()](https://github.com/LIFsCode/ELOC-3.0/blob/master/eloc610LowPowerPartition%2Fsrc%2FElocSystem.cpp#L307) this is currently called from low priority wakeup_task (I know the name is not suited anymore) but can be also called from main.
This would keep all the system status updates centralized in a single low priority update task.

But if you prefer having the SD card separatly handled we can do that as well, I just wanted to keep the task architecture simple. I think this already gave us some problems when it comes to reproduceability and stack utilization.

OOHehir, 2024-02-04:
@LIFsCode OK, I'll start on this enhancement, I'll add it to ElocSystem::handleSystemStatus.

OOHehir, 2024-02-05:
@LIFsCode @EDsteve This enhancement looks complete within this [branch](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/sd_central_task).  I can wait to merge it into main but the WAVFileWriter uses it to check the SD card status so I think it's worth merging soon.

OOHehir, 2024-02-06:
@LIFsCode Its probably fairly obvious but ElocSystem::handleSystemStatus, when the system is not loaded, is called constantly. I thought accessing the SD card constantly might slightly increase power usage. I'll take a look to check.

LIFsCode, 2024-02-06:
@OOHehir : the task is run with [100ms interval](https://github.com/LIFsCode/ELOC-3.0/blob/ac2d9c6fc5871b4533e1f822643daf9c6c26b6c5/eloc610LowPowerPartition/src/Commands/BluetoothServer.cpp#L272) when BT is off and everything is idle.

For ELOC 3.3 or newer only the [IO expander must be read to check if a SD card is present](https://github.com/LIFsCode/ELOC-3.0/issues/62), so I guess power impact for the future is neglectable.

OOHehir, 2024-02-06:
@LIFsCode Yes, looks OK, I can't see much difference in power consumption..

OOHehir, 2024-03-08:
@LIFsCode Any issue if I pull this into master?

LIFsCode, 2024-03-09:
@OOHehir From my point it's ok, but maybe it makes sense not to change too much on the master as long as we still have the analyzing of the timestamps running.

We have currently 2 things open which should be done first before any major changes on master:
- #77: Analyze of time drift source
- #86 the change in the BT CMD interface for the App to better keep track of commands

I guess it will make the life easier for @EDsteve  if we keep the master clean until this is complete. Or alternatively fork from the master with a testing branch and @EDsteve works on this.

Sorry I'm not at home at the moment so I don't have an overview of what's the actual master branch state.

EDsteve, 2024-03-10:
@LIFsCode @OOHehir 
All the time-drift-test i do with d1e8350819970620c633369ac651091fbdd2fe1c 
I am not using newer commits in my tests. So it's fine with me if you make new commits.

OOHehir, 2024-04-09:
@EDsteve @LIFsCode This seems like a good time to introduce this change to master.

OOHehir, 2024-04-11:
Tested briefly

closed 2024-04-11 by OOHehir