LIFsCode/ELOC-3.0 #77: Recording time value is off

EDsteve, 2024-02-21:
I have realized that already a while ago but didn't want to interrupt the Guru hunt :)

After **61 minutes uptime** and **60 minutes recording/detecting** i get the following times from getstatus:
[image]

The recording/totalrecording time time needs an adjustment of the decimal point. And the uptime seems off a few minutes?

It would be good to know the recording time in the app. So the rangers know what is going on when they check the devices. Should be an easy fix.
But if this fix is not easy or if this fix might interefer with the overall stability of the firmware. It's also okay to resolve that issue with the next firmware update.

OOHehir, 2024-02-21:
@EDsteve @LIFsCode OK, I'll take a look at this later today.

OOHehir, 2024-02-21:
@EDsteve @LIFsCode I suspect what's happened in this case is documented here [ArduinoJSON](https://arduinojson.org/v6/how-to/configure-the-serialization-of-floats/). The JSON library has switched to scientific notation to represent the number. You'll probably find that the number is actually correct.

At this point, so close to deployment, it's probably better not to change things. We can look at rounding floats for the next release.

EDsteve, 2024-02-21:
@OOHehir Ok. Thanks for looking into it. I just find it strange that these times (recording time, total recording time, uptime, detecting time) are so different from each other eventhough they should show all the same time because i started recordOn_detectOn right after i started the ELOC.

@LIFsCode 
As Owen already said. Let's implement that with the next release. I think you should get some ELOC holidays after the last few weeks of hard work. :)

OOHehir, 2024-03-02:
@EDsteve @LIFsCode This should solve this issue, the times will now appear round to two decimal places. I'm testing now & looks good.

@EDsteve If this doesn't solve the issue please re-open

EDsteve, 2024-03-06:
@OOHehir @LIFsCode 
The time shows in the app now. First step done :)

Unfortunately the time is not accurate. The time from th ELOC runs much faster than real recording time. And it's kind of random?

Here are a few tests i made:

- After 5 minutes recording (measured with stopwatch), the app shows 7 minutes
- After 15 minutes recording, the app shows 18 minutes
- After 60 minutes recording, the app shows 76 minutes
- After 42 hours recording, the app shows 65 hours
- After 11 hours recording it showed more than 50 hours

I couldn't see a correlation between recording time from the ELOC and real recording time. Not sue what is going on. But hopefully you already have an idea why that is :)

EDsteve, 2024-03-07:
> @EDsteve When you say the 'time', which time is it? Uptime, recordingTime, detectingTme or TotalRecordingTime, or all of them?

In my experience all three times running with the same "wrong speed". So if you find a fix for one time. That should also fix the other times.
Will try your latest commit and report back :)

OOHehir, 2024-03-07:
@EDsteve @LIFsCode  I can recreate this issue. After approx 1hr 20min of running (with recording & inference in progress) the Uptime is reported as 1.6 (should be 1.33), looks like a 20% error.

The code uses this [library](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/system/esp_timer.html) for time. From a quick look online it appears it should be accurate to around 5%. 

The SDK includes:
```
CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256=y
``` 
@LIFsCode unless any solutions springs to mind I'll continue looking at a solution. 

I see a related issue #26 

I've pushed a commit to master to improve the precision of the times in the status message.

LIFsCode, 2024-03-07:
@OOHehir @EDsteve  Does this time drift occurs in the timestamp of the ESP_LOG() messages as well?

I want separate any HW related time drift from SW APIs influence.

If yes it is likely that it's caused by the RTC clock source accuracy.
I have not found any datasheet entries about RTC accuracy in the ESP how do you get the 5% precision?

If there is a time drift in the log timestamps as well it is related issue #26. Otherwise I suspect a SW API issue.

20% drift is far too high drift for the internal oscillator. That's more what I expect for the internal RC oscillator (see #26) this has also shown large drifts in the past but that should have been fixed with
 `CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256=y`

EDsteve, 2024-03-07:
@LIFsCode @OOHehir 
The actual recordings do have the correct recording-time when i check the length. So when i record for exactly 6 hours and open the wav-file in Audacity. It shows exactly 6 hours.
This also relies on the internal RC oscillator, correct? Doesn't that tell us that the internal RC oscillator is accurate enough?

LIFsCode, 2024-03-07:
@EDsteve Yes that's what I meant. The RTC should be accurate enough.

BTW. it is **not** the internal RC oscillator. This has been changed in [ffbfa34](https://github.com/LIFsCode/ELOC-3.0/commit/ffbfa34a93b67261220cdbeb4eb522a036f60025) it is changed to the [internal 8.5 MHz oscillator ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html) which is more accurate.

If the timestamps of the recording are accurate it is something with the API or its usage. 

With [this ](https://github.com/LIFsCode/ELOC-3.0/blob/59b967a9e5da93e39e4a161979bfef873cf50b02/eloc610LowPowerPartition/sdkconfig.defaults#L700C1-L700C36) the ESP32 timers are configured to only use the RTC clock and not its timers to avoid inaccuracies due  dynamic frequency scaling. 

Maybe the use of this [library](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/system/esp_timer.html) has something to do with DFS? Could be checked by setting Min & max Freq equal to 80 in the [PM config](https://github.com/LIFsCode/ELOC-3.0/blob/59b967a9e5da93e39e4a161979bfef873cf50b02/elocConfig.json#L11).

EDsteve, 2024-03-07:
@LIFsCode Just started a test with Min and max of 80 and will run it through the night.

OOHehir, 2024-03-07:
@LIFsCode 
_I have not found any datasheet entries about RTC accuracy in the ESP how do you get the 5% precision?_

From online discussions, so treat with caution.

@EDsteve @LIFsCode For info WAVWriter doesn't actually use 'clock time' to set the file size, it uses the sample rate, so say for a 16kHz sample rate, 16000 * 60 = 1 min file size.

LIFsCode, 2024-03-07:
@EDsteve @OOHehir 
> @EDsteve @LIFsCode For info WAVWriter doesn't actually use 'clock time' to set the file size, it uses the sample rate, so say for a 16kHz sample rate, 16000 * 60 = 1 min file size.  

Yes you have to check the timestamps on the filesystem and the timestamp in the filename to get an idea about accuracy.
The time in audacity only means that there are the correct number of samples with the given sample rate for 60s.

Regarding overall clock accuracy this can also be observed by applying a defined sine tone (e.g. 1kHz) as input and analyzing the resulting WAV file in Audacity in the frequency domain. An inaccuracy of 10-20% should be visible in the signal. But I don't remember ever seeing any problems here, so I guess it is not a general clock issue, but a timestamp/timer/API issue.

EDsteve, 2024-03-08:
@LIFsCode @OOHehir 
The time stamps seem to be also device specific. But still on test runs and report back when i have all results.

One thing to note: The [old firmware from Tom](https://github.com/tbgilson/-elocNew--low-power-espidf-partition) using ELOC2.7 does not have this issue (Just tested). Recording times showing correct values in the app and also the timestamps in filename and filesystem didn't have that problem.

LIFsCode, 2024-03-08:
@EDsteve could you please run tests with the same suspicious device and firmware version [ffbfa34](https://github.com/LIFsCode/ELOC-3.0/commit/ffbfa34a93b67261220cdbeb4eb522a036f60025). Just to make sure the later merges did not change anything.

> One thing to note: The old firmware from Tom using ELOC2.7 does not have this issue (Just tested). Recording times showing correct values in the app and also the timestamps in filename and filesystem didn't have that problem.
@EDsteve yes that's clear. It doesn't use any power management.

I'll try to explain a bit how the ESP clocks work for timestamp:

# clock sources and time base

There are 2 different clock sources for the system time:
- the internal system clock 80 MHz (+- 10 ppm)
- the RTC clock (3 options: internal RC (low power poor accuracy, internal 8.5 MHz oscillator (med power med accuracy), external quartz (best accuracy, requires external component))

By default the time base uses the system clock which is the most accurate (Tom's firmware uns with this) but has 2 drawbacks:
- it will not preserve time during deep sleep (not critical for us currently)
- it is highly affected by power management and dynamic frequency scaling !!!

That's why I changed it in [ffbfa34](https://github.com/LIFsCode/ELOC-3.0/commit/ffbfa34a93b67261220cdbeb4eb522a036f60025) to the RTC clock and the internal 8.5 MHz clock.

# Conclusion 
Since there is nothing about the internal RTC clock accuracy in the datasheet the only option to reliably get a stable time base **and** power management is to use an external RTC Quartz (see also #11)

OOHehir, 2024-03-08:
@EDsteve I added some commits over the last day or two. Would you mind checking your on the latest master? If the issue persists could you please comment with the build log onto issue #88.

EDsteve, 2024-03-14:
@LIFsCode @OOHehir 
This takes more time than i expected. No smooth sailing here. Sometimes the times are more accurate than other times. And sometimes they are off by a lot. It seems kind of random and i can't find any regularities. And of course i found some new issues during all these tests which didn't make things easier.

So yeah. Still in testing mode, but one thing i can say for sure:
The timestamps in the log file (ms since boot) do not represent "real time" and are contrary to the timestamps of the recorded files.

Example:
I have an ELOC which is set to record 1 minute files. I boot it up, press the GPIO button to start recording and once the wav file is created (which i can see in the log) i start the stopwatch.
Theoretically after 60 seconds have passed, the second wav file should be created. And in most cases that is true (more to that later). But when i take the timestamp from the log file (ms since boot) and calculate the time difference between the first wav file and the second wave file was created in the log. Then the result is much more than 60 seconds (for example 1133325ms - 1059897ms = 73428ms). I have seen numbers between 68 and 77 seconds, but never less than that. 
[Here](https://docs.google.com/spreadsheets/d/1khbRgweA-oDa-n73EJ0IwOPv4qLqriZYRWleOVxIy2w/edit?usp=sharing) is an example of the time differences in the log file of a >40 minute recording session.

This happens with the older firmware (ffbfa34a93b67261220cdbeb4eb522a036f60025) before the merge and also with the newer firmware d1e8350819970620c633369ac651091fbdd2fe1c.

_"Theoretically after 60 seconds have passed the second wav file should be created. And in most cases that is true (more to that later)."_
The last 12h i had a drift of only 1 second. That was good. But with the new test it starts drifting already a few seconds within 10 minutes. I suspect the SD card now because that is the only thing i changed. And it seems with that card i get more errors. But yeah... that's how it has been the last days. Can't really give you more "solid" results for now :(

I just hope now that this log file timestamp problem does give you some ideas.

OOHehir, 2024-03-14:
@EDsteve @LIFsCode The code used to access time on the ESP32 is based on this [library](https://github.com/fbiego/ESP32Time). It seems that there are (unresolved) issues reported that are not too dissimilar from the issues we're observing. I'll try running some test code to see if I can recreate the issues separate from our code base.

EDsteve, 2024-03-15:
After a call with @LIFsCode we have decided to test the firmware before the merge. With this FW there is a lot less going on which should make things easier.
I still want to post the "time drift" of my last tests here which also includes the time drift of the FW before the merge (last row). Log files and recordings for each test are available if needed.
If some of these tests seem suspicous to you. I am happy to redo them. 

| min | max | FW                                     | Time between start/Stop recording | Windows Explorer time stamps of first file created and last file modified | Time difference in seconds |
|-----|-----|----------------------------------------|-----------------------------------|--------------------------------------------------------------------------|---------------------------|
| 80  | 80  |                                        | 16h15min0s                        | 16h14m6s                                                                 | 54                        |
| 80  | 80  | 7090f504b99a19d17d354611547ddd5405d6fcf7 | 12h59m0s                          | 12h58m44s                                                                | 16                        |
| 10  | 80  | 7090f504b99a19d17d354611547ddd5405d6fcf7 | 12h37m0s                          | 12h37m52s                                                                | 52                        |
| 10  | 80  | d1e8350                                | 12h15m0s                          | 12h15m3s                                                                 | 3                         |
| 10  | 80  | ffbfa34                                | 12h37m16s                         | 12h36m33s                                                                | 43                        |

OOHehir, 2024-03-16:
@LIFsCode @EDsteve If we agree, the issue here is primarily that is that the internal clock is drifting. There are other symptoms (mainly inaccurate timestamps) but these are most likely related to this issue.

I've seen some discussion [online ](https://esp32.com/posting.php?mode=quote&f=12&p=25836)around the use of the RTC & its accuracy, such as:
_My problem is very similar to yours. I'm also doing light sleep and seeing very large and unpredictable time drifts (around +3-10ms/s)._
There's other similar reports but as we can't examine their config and/ or code it's difficult to trust such reports. One [interesting post](https://github.com/espressif/esp-idf/issues/6687#issuecomment-1079609214) I did find & maybe a possible solution, is a method of calibrating the 150kHz RTC. 

If we believe that this issues needs to be explored further (will this particular ELOC design be abandoned in favor of one featuring an external crystal?):
1. Measure (using NTP or an external RTC) the drift over a significant time period in the current setup
2. Examine & implement a solution such as the one above
3. Confirm it solves the issue using step 1.

EDsteve, 2024-03-16:
@OOHehir Not sure. Maybe we are dealing with two seperate issues?

1. The Timestamps from the recorded files (Date created from first file and date modifed from last file) are off only a few seconds after a longer period of recording (My guess: due to RTC drift) And that drift would be okay
2. The timestamps in the log file (ms since boot) and the uptime/recording-time (from getStatus) are off by few hours or days after the same period of recording (My guess: Not related to the RTC drift)

But maybe my guess is totally wrong.

@LIFsCode 
Just got the data from another ELOC, which shows that. Running with 1h filesize for almost 4 days using d1e8350819970620c633369ac651091fbdd2fe1c (Recording has been started and stopped from the app, but no other BT connection during recordings):

- Actual real recording time: **3d19h38m??s**
- Time between first file created and last file modified: **3d19h38m??s**
- Recording time in app (from getStatus): **5d3h25m**

If i understand all correctly. The timestamps in the filename show that the time drifts <1 second per hour in both diretions?

[image]

I started the same test again to check if i can repeat the same results (without any connection from app this time).

OOHehir, 2024-03-16:
@EDsteve, yes, there seems to be something else here as you say.

@LIFsCode @EDsteve So there appears to be two or three issues involved here:

1.  Drifting RTC, perhaps this is within a reasonable tolerance. Further investigation required.
2. Timestamps in the system console are incorrect (perhaps related to this [issue](https://github.com/espressif/esp-idf/issues/700#issuecomment-316258237), perhaps not) 
3. esp_timer_get_time() used by getStatus reports the incorrect time, quite possibly related to the previous issue.

I'll investigate issue 2, & perhaps solving it will solve issue 3.

OOHehir, 2024-03-18:
@EDsteve @LIFsCode Something that may help make this clearer. It seems that the system console uses the [FreeRTOS tick count](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/system/log.html#_CPPv417esp_log_timestampv), rather than system time for the time stamps.

So from what I can see so far:
SD card & WAVWriter -> system clock (appears correct)
Logging -> FreeRTOS tick count (appears incorrect)
esp_timer_get_time() -> should be system clock (appears incorrect)??

LIFsCode, 2024-03-18:
@EDsteve @OOHehir  I just run into this
https://github.com/espressif/esp-idf/issues/9297
Though it sounds related this would imply that there is an effect of the pm settings (minCPU & maxCPU) settings if they are set identical.

@OOHehir I know I had problems with the RTOS tick count that's why I added 
 ```
CONFIG_ESP32_TIME_SYSCALL_USE_RTC=y
```
But it seems that this only affects the systime not the log time or the esp_timer API

OOHehir, 2024-03-18:
@LIFsCode That sounds promising, doesn't seem to have got any attention from Espressif. I wonder is it just because it doesn't included a reproducible example?

@LIFsCode @EDsteve 
I still cant understand how the SD files have (fairly) correct timestamps & how the system clock seems off is happening. I started looking through the espressif code to see where the SD card gets its time, not luck so far.

Might be worth adding further debug output to console comparing all time sources (e.g. esp_timer_get_time(), FreeRTOS sys tick count & clock time).

OOHehir, 2024-03-19:
@LIFsCode @EDsteve I've added some more output to test the various times including:

esp_timer_get_time 
esp32Time.getEpoch
FreeRTOS xTaskGetTickCount
esp32Time.getTime

& so far I haven't observed any big issues, after 8hrs all the times were within 3 min or so. Those tests were simply letting the board boot & observing the results. I'm now performing again with continuous recording. 

Once I have results I'll update this, but in the meantime I came across [this](https://forum.arduino.cc/t/esp32-settimeofday-functionality-giving-odd-results/676136) which may be of importance.

OOHehir, 2024-03-20:
@EDsteve @LIFsCode Juts finished a test of around 8hrs with continuous recording. At the end 

esp32Time.getEpoch
esp32Time.getTime

Were identical (naturally enough) & had an error of 54 secs

esp_timer_get_time was incorrect by 132 sec
FreeRTOS xTaskGetTickCount was incorrect by 390 sec

A little surprising! I don't understand why esp_timer_get_time is not as accurate as esp32Time. 

My suggested solution is then:
1. Uptime source -> use the most accurate time source which seems to be esp32Time. 
2. Uptime calculation -> get the  difference between  esp32Time.getEpoch at boot (stored as static variable) & current esp32Time.getEpoch. 
3. I suggest setting UTC/ GMT on the ELOC & then setting the local time on the device using
```
setenv("TZ", "WIB-7", 7);  // Western Indonesian Time
tzset();
```

OOHehir, 2024-03-21:
@LIFsCode @EDsteve As per the above commits I've added support for setting the system time using Unix style time & a timezone. I've also added a method of getting of uptime from ESP32Time. For evaluation I've added some terminal output (temporary disabled heap info output). 
```
I (25431) main: Battery: Voltage: 3.382V, 2% SoC, Temp 27 °C
I (25439) main: FreeRTOS xTaskGetTickCount = 23650
I (25442) main: esp_timer_get_time = 25283423
I (25446) main: esp32Time.getUpTimeSecs = 21
I (25458) main: esp32Time.getEpoch = 1711056484
I (25462) main: esp32Time.getTime = 04:28:04           // <-------- This returns local time as per timezone
```
I'll let it run overnight & see how it performs. 

Please note the time source for the Bluetooth uptime, recording & inference time hasn't changed.

EDsteve, 2024-03-22:
@OOHehir 
Awesome. You did a lots of time digging :)
Sorry for late reply. On the road right now but back on 27. Surely i brought two ELOCs with me. So I will also test the last commit on my side and report back with results tomorrow.

OOHehir, 2024-03-22:
I've tested for a 20,450 seconds  & after that:
esp32Time.getUpTimeSecs = 20447 secs
FreeRTOS xTaskGetTickCount = 20243 secs
esp_timer_get_time = 20584 secs

Its still annoying to not know the reason behind the differences but for the moment its fairly clear that esp32Time.getUpTimeSecs is more accurate. I can modify Bluetooth 'uptime' with this (rather than esp_timer_get_time) for further evaluation & then see if its worth modifying recording & inference time to use this for calculation also.

EDsteve, 2024-03-23:
@OOHehir My ELOC didn't make it. It rebooted after BT turned on. But i had another ELOC running last night which i tested the phone time synch with. The last time-synch happened around 10 hours before i took the results below. So the results should still be representable for the time drifts.

This is result after 59750 seconds uptime:

```
I (70285300) main: FreeRTOS xTaskGetTickCount = 70283478
I (70285304) main: esp_timer_get_time = 79758196570
I (70285307) main: esp32Time.getUpTimeSecs = 59817
I (70285313) main: esp32Time.getEpoch = 1711175197
I (70285319) main: esp32Time.getTime = 14:26:37  <------ Only 23 seconds off. Real time was 14:27:00
```

I just started two more tests (without any time synchs from the app):
1. Start the ELOC without any interaction
2. Start the ELOC with recording using GPIO button

Will report results in a few hours.

@LIFsCode If you want me to do specific tests with the old firmware. Let me know.

EDsteve, 2024-03-24:
@OOHehir @LIFsCode Two more tests running for >11 hours: using this commit: 5a1a0ce65a6f6aee87629f0a7a9c96a124ad3610
Not sure if it helps. But i hope so :)

### ELOC_00052
(**with** recording from GPIO - without any time synch from phone)

**ELOC was running for 41639 seconds** (Started the stopwatch when i saw the first time-log-entry). 

The results where calculated from first visible time-log-entry to the last one:
**esp32Time.getUpTimeSecs = 41642 secs
FreeRTOS xTaskGetTickCount = 50726 secs
esp_timer_get_time = 41642 secs**

First time log entry:
```
I (20667) main: FreeRTOS xTaskGetTickCount = 18845
I (20670) main: esp_timer_get_time = 20103975
I (20673) main: esp32Time.getUpTimeSecs = 17
I (20677) main: esp32Time.getEpoch = 1711093742
I (20683) main: esp32Time.getTime = 14:49:02
```

Last time log entry:
```
I (50746902) main: FreeRTOS xTaskGetTickCount = 50745080
I (50746906) main: esp_timer_get_time = 58973323206
I (50746909) main: esp32Time.getUpTimeSecs = 41659
I (50746914) main: esp32Time.getEpoch = 1711135384
I (50746920) main: esp32Time.getTime = 02:23:04
```

Here the complete log file from start to end: [eloc1.log](https://github.com/LIFsCode/ELOC-3.0/files/14734615/eloc1.log)

### ELOC_00056
(**without** recording from GPIO - without any time synch from phone):

**ELOC was running for 40789 seconds** (Started the stopwatch when i saw the first time-log-entry). 

The results where calculated from first visible time-log-entry to the last one:
**esp32Time.getUpTimeSecs = 40761 secs
FreeRTOS xTaskGetTickCount = 40375 secs
esp_timer_get_time = 40761 secs**

First time log entry:
```
I (10657) main: FreeRTOS xTaskGetTickCount = 8835
I (10662) main: esp_timer_get_time = 8891122
I (10675) main: esp32Time.getUpTimeSecs = 8
I (10679) main: esp32Time.getEpoch = 1711112483
I (10683) main: esp32Time.getTime = 20:01:23
```

Last time log entry:
```
I (40386420) main: Battery: Voltage: 3.838V, 50% SoC, Temp 33 °C
I (40386427) main: FreeRTOS xTaskGetTickCount = 40384605
I (40386439) main: esp_timer_get_time = 41054460230
I (40386443) main: esp32Time.getUpTimeSecs = 40769
I (40386446) main: esp32Time.getEpoch = 1711153244
I (40386450) main: esp32Time.getTime = 07:20:44
```

Here the log file from start to end: [eloc2.log](https://github.com/LIFsCode/ELOC-3.0/files/14734622/eloc2.log)

@OOHehir In my tests the getUpTimeSecs and getTime show the same accuracy. That was different in your test and i wonder why that is.

EDsteve, 2024-03-25:
@OOHehir 
I just tested the last commit (362215d47791dd5bec05a7fbceabd9c765155a05)
Test procedure: 
Start the ELOC -> Connect with app (time-synch) -> start recording -> wait 1.5h -> connect with BT serial and send getStatus

After 1.5h (real time) i get these results:
getStatus gave me these times:
Uptime: 1.5h  -> Correct
Recording time: 1.9h -> To long

@LIFsCode @OOHehir 
If that get's too difficult to solve. We can just keep it as is and deal with it later.
Most important are the timestamps in the filenames. And they seem to be close enough to the real time.

OOHehir, 2024-03-25:
@EDsteve @LIFsCode The recording time is still using the esp_timer so it's likely that's the reason for the error in the recording time. It seems uptime (which uses the ESP32Timer) is correct. It shouldn't be too difficult to change the wav writer & inference to use the ESP32Timer. I'm away until the weekend, will make the changes then.

OOHehir, 2024-03-25:
> @OOHehir @LIFsCode Two more tests running for >11 hours: using this commit: 5a1a0ce65a6f6aee87629f0a7a9c96a124ad3610
> Not sure if it helps. But i hope so :)
>
> ### ELOC_00052
> (**with** recording from [...]

Is this test using anything different? The original board you sent me & the default config? I've never found the two to be the same. Whatever is different in this test could help explain the underlying issue.

EDsteve, 2024-03-25:
@OOHehir Ohh. I mistakenly switched esp32Time.getTime with esp_timer_get_time. Here are the corrected values which should make more sense. Sorry for the confusion.

**ELOC was running for 41639 seconds**
FreeRTOS xTaskGetTickCount = 50726 secs
esp_timer_get_time = 58953
esp32Time.getUpTimeSecs = 41642 secs
esp32Time.getTime = 41642 secs

**ELOC was running for 40789 seconds**
FreeRTOS xTaskGetTickCount = 40375 secs
esp_timer_get_time = 41045
esp32Time.getUpTimeSecs = 40761 secs
esp32Time.getTime = 40761 secs

OOHehir, 2024-04-05:
@EDsteve I've just tried a quick check of the above commit & looks promising.  When you get a chance could you please try also & close this if it checks OK?

EDsteve, 2024-04-06:
@OOHehir Thanks. I have started a test 2 days ago. But just now realized that i forgot to flash the latest FW. haha.
Will start again with the latest commit and report back.

EDsteve, 2024-04-06:
This seems to be fine now. But the this [issue](https://github.com/LIFsCode/ELOC-3.0/issues/93https://github.com/LIFsCode/ELOC-3.0/issues/93) might have been introduced from the fix?

closed 2024-04-06 by EDsteve

LIFsCode, 2024-12-20:
@EDsteve @OOHehir I'm sorry I missed a bit during my absence this year. First of all thanks for the debugging this was really a tough one.

Just to verify that I have the correct conclusion, since there was a lot going on:

1. your tests showed that `esp_timer_get_time `is less accurate than esp32Time.getTime (`gettimeofday`)
2. DFS (Min. & max. CPU Frequency settings) did not have a major impact on timing accuracy
3. The Timestamps of the files on SD card were accurate (probably they have been created based on gettimeofday()
4. No Config/RTC Changes were needed to fix time accuracy.
5. If I understand it right, it is still unclear why `esp_timer_get_time  `and `gettimeofday ` show different accuracies. Correct?

Please correct me if I'm wrong in any of my above interpretations.

OOHehir, 2024-12-20:
@LIFsCode @EDsteve We never got to the reason for the discrepancies. It's been a while since I looked at this but, yes, your summary seems correct.