LIFsCode/ELOC-3.0 #79: Changing the AI model results in DRAM compiling error

EDsteve, 2024-02-23:
@OOHehir 
We would like to test some models in the jungle, therefore i just tried to implement a new model (using this commit 8339ced2ec9f617414e5dc075a878fdef896f7a1  ) using esp32dev-ei-windows. But getting a compiling error. Haven't tried to change the model for a while so i am not sure since when this issue is happening. If the error doesn't point you in the right direction. I can try to find out which commit introduced that issue.

```
Archiving .pio\build\esp32dev-ei-windows\libe32\libaudio_input.a
Archiving .pio\build\esp32dev-ei-windows\lib882\libedge-impulse.a
Linking .pio\build\esp32dev-ei-windows\firmware.elf
c:/users/edste/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev-ei-windows/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/edste/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/edste/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/edste/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 12088 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev-ei-windows\firmware.elf] Error 1
```
Let me know if you need more infomration about it. The model i tried to implement is this one:
[test_Trumpet_Trimmed_1.0.54.zip](https://github.com/LIFsCode/ELOC-3.0/files/14383445/test_Trumpet_Trimmed_1.0.54.zip)

EDsteve, 2024-02-24:
@OOHehir 
After doing a fresh github clone. The model import works. Not sure what actually caused it.
I am sorry if you have wasted time to investigated this issue.

OOHehir, 2024-02-24:
@EDsteve No problem,  I didn't get a chance to look at it. I think different EI models might pull in different files, so perhaps could be the issue.

EDsteve, 2024-02-28:
I closed this mistakenly. I still can't change the model. 
@OOHehir if you can take a look. That would be great. So i can start testing models and hopefully deploy some in the jungle for testing soon.

OOHehir, 2024-02-28:
@EDsteve Its difficult to examine the model from the files but I presume its quite complex? The compiler is complaining that there's too much data to store in the assigned region. I added this [build flag](https://github.com/LIFsCode/ELOC-3.0/blob/7090f504b99a19d17d354611547ddd5405d6fcf7/eloc610LowPowerPartition/platformio.ini#L90) for the AI build, you could comment it out & see if the code compiles. It's not an ideal fix as it will result in the memory getting more fragmented during runtime.

OOHehir, 2024-03-02:
@EDsteve Additionally to this you could try uncommenting
```
#define EI_BUFFER_IN_PSRAM
```
in project_config.h to a allocate the bufer in PSRAM & see if that solves the problem.

EDsteve, 2024-03-02:
@OOHehir Haven't had much luck with the build flag. But thanks for the new tip. I will try that hopefully tomorrow.
The models i am using are the same as i used before. Not complex at all i would say. But let's see if #define EI_BUFFER_IN_PSRAM solves that problem :)

OOHehir, 2024-03-08:
@EDsteve Just tried this. It fails to build as you described but commenting out this [line](https://github.com/LIFsCode/ELOC-3.0/blob/10d83ee1663e22cf0433ff0c63fb836f47336d3d/eloc610LowPowerPartition/platformio.ini#L90) allows a build. Probably best to do a 'full clean' first to ensure there's no left over build artifacts.

Please keep in mind the caution above:
_It's not an ideal fix as it will result in the memory getting more fragmented during runtime._

I'll close this, please re-open if it persists for you.

closed 2024-03-08 by OOHehir