LIFsCode/ELOC-3.0 #83: Firmware Updates from SD card not always working

EDsteve, 2024-02-29:
@LIFsCode @OOHehir 
Just realized that the update process from SD card doesn't seem to always work. The LEDs are flashing as they should indication an update. But the log file shows errors as you can see in this log:
[Failed_Update-Log.txt](https://github.com/LIFsCode/ELOC-3.0/files/14443234/Failed_Update-Log.txt)

We updated all firmwares for the ELOCs from SD card without checking the serial monitor. So this issue might be the reason for the reboots in Jambi as discussed [here](https://github.com/LIFsCode/ELOC-3.0/discussions/82)
That might also explain the older dates from the log files shown in the pictures from the discussion?

After 10 tests I haven't figured out yet when it works when it doesn't. Because it definitely works sometimes and sometimes it doesn't. But maybe you can find it out faster than i do looking at the log file? :)

This is the content of the SD card i used to update the ELOCs for Jambi. I made the doUpdate.txt file Read-Only so we were able to use it without adding the txt file after every update.
[Jambi_firmware.zip](https://github.com/LIFsCode/ELOC-3.0/files/14443350/Jambi_firmware.zip)

OOHehir, 2024-02-29:
@EDsteve @LIFsCode OK, is this a priority issue or should we incorporate it into the upcoming discussion about updating via Bluetooth? It seems it'll be overlapping, so unless it's necessary, it might be fixing that could possibly be replaced.

EDsteve, 2024-02-29:
@OOHehir @LIFsCode 
Definitely priority. It's a lot of effort to get the ELOCs out in the jungle. So they should work. If it takes less than one week to make the BT update work. I would choose the BT route. But thats wishfull thinking i guess? :)

Let's see what @LIFsCode says about this issue and then we decide what we do.

EDsteve, 2024-03-01:
@LIFsCode 
> hat I assume has happend here is: Did you maybe flashed the ELOCs via VS Code and then updated them via SD card with a previous generated .bin file?

You are correct. That's what happend. So yea. That explains it. Good to know this savety feature.
Was hoping that would explain also the reboot issues in Jambi. But no luck. :)

I will close this now.

closed 2024-03-01 by EDsteve