LIFsCode/ELOC-3.0 #84: BT connection set's weird time stamps / recordings missing

EDsteve, 2024-03-02:
There is a new issue, which might be serious if it happens more often. It seems on some devices recordings are missing and timestamps are wrong after a BT connection.

### How it happens:
This ELOC was running under a tree near my house for 8 days. Almost every day i connect to it via BT from the app just to see if it is still recording. All seems fine in the app. Still shows recording. I never stopped the recording. I only connected with the app to check the status. 
When did i connect to the ELOC:
24 Feb: 10:36
25. Feb: 11:21
25. Feb: 13:59
27. Feb: 14:29
28. Feb: 13:20
29. Feb: 11:16
@LIFsCode @OOHehir 
These times corelate with the timestamp-change in the screenshots below.

### Problem
When i look at the recordings from this ELOC after 8 days (when the battery was empty). I realized that it didn't record all the time and the timestamps don't make sense. 

1. One hour recording is actually >61 minutes according to the timestamps in Windows explorer. But when i open one recording in Audacity it shows exactly 60 minutes. This accumulates over time which is a lot after one month recording. Might be connected to [this issue](https://github.com/LIFsCode/ELOC-3.0/issues/71)?
2. The timestamp of each file seem to be using the time of the last BT connection. Untill i connect again.
3. Day 26 of February is missing completely

This is the /eloc content as txt file: [BAd_TimeStamp.txt](https://github.com/LIFsCode/ELOC-3.0/files/14468629/BAd_TimeStamp.txt)
And the config: [not_set1708625762254.txt](https://github.com/LIFsCode/ELOC-3.0/files/14468667/not_set1708625762254.txt)

Here the screenshots (colored by day):
[image]
[image]
[image]
[image]

Not sure if that is connected. But this ELOC always showed Battery state empty during these 8 days. Here is a getStatus: 
[image]
[image]

### ELOC without this issue:
And of course to make it more difficult. This does not always happen. Another ELOC also running outside did not show this issue. When i am testing ELCOs I normally only check if it has recordings and listen to a few of them, but i normally don't count them. That's why i can't say how often this happened already.

This is the folder content as txt from another ELOC without issues: [Good_TimeStamps.txt](https://github.com/LIFsCode/ELOC-3.0/files/14468627/Good_TimeStamps.txt)
And the config: [black case1709055550864.txt](https://github.com/LIFsCode/ELOC-3.0/files/14468668/black.case1709055550864.txt)

Would it make sense to fix the [recording time, uptime issue](https://github.com/LIFsCode/ELOC-3.0/issues/77)? That would maybe also make it easier to realize these kind of problems?

LIFsCode, 2024-03-02:
@EDsteve @OOHehir 

This is indeed a very complicated issue and I assume here are multiple issues mixing up in this:

- [ ] 1. There are obviously missing files --> problem 1
- [ ] 2. There is a (potentially) large timedrift in the timestamps, this may relate to https://github.com/LIFsCode/ELOC-3.0/issues/26 As mentioned in [ffbfa34](https://github.com/LIFsCode/ELOC-3.0/commit/ffbfa34a93b67261220cdbeb4eb522a036f60025) the time drift of the oscillator is not yet evaluated. So maybe you just did the evaluation --> problem 2
- [ ] 3. in blocks all files on the sd card have the same timestamp even though having different timestamps in their filename: This seems more a file system issue --> problem 3
- [ ] 4. There might be an issue with the updating of the timestamps during runtime --> unproved so far

@EDsteve do you have any log files please?

### Further steps to track the issue:

@EDsteve Can you redo the tests with larger logfiles. I have increased the default file size of the rotate files to 5 MByte [2633684](https://github.com/LIFsCode/ELOC-3.0/commit/26336849be52f68cbb0e667246df5fbd63a192fa) the 5 kByte end up to be far too small for effecient analysis.

1. @EDsteve Could you start the ELOC (which showed this errors) only once and not connect via BT and simple let it run for 1-2 days and then connect and stop it. Just to see how the timestamps drifted and keeping out all other issues.
2. Can you also redo the same test for multiple days with multiple BT reconnect with larger logfiles.

EDsteve, 2024-03-03:
@LIFsCode 

2. I have done some tests before about the timedrifts and could not find any timedrift within a few days. So the timedrift problem from this issue is either specific for only this one ELOC or some other reason. - > But will do more tests and report back

3. The timestamp in the filename is the time of creation and the time in windows explorer is date modified. So they should be different. Date modified should be 60 minutes behind date of creation in a working scenario.
Not really sure if i understand you correctly. But in the screenshot they do have the same timestamp in filename and explorer (see picture below).
[image]

**Log File:** Unfortunately i tested this device with logs off. 

I will do all tests soon and will report back.

And thanks for the recordingtime-fix. Will check it out as well @OOHehir

LIFsCode, 2024-03-03:
@EDsteve  Ah sorry I have read it wrong. So the files are 1 second apart even though they have samples for 60 seconds. That's strange

EDsteve, 2024-04-06:
Got hopefully solved with #86

closed 2024-04-06 by EDsteve