LIFsCode/ELOC-3.0 #91: Handle different sample rates for AI & Recording correctly

LIFsCode, 2024-03-10:
AI and recording may have to run With different sample rates. 

Current code has 2 major issues:
- sample rates must be integer multiplies of each other. Thus the available audio sample rates which are useable with AI are limited! E.g. Ai sample rate =4kHz -->, available  audio sample rates: 4, 8, 12, 16, 20, ... 44 kHz @EDsteve: question is up to you: Is this a problem or can we live with this limitation? I would strongly suggest accepting the limitations as it will make life much easier and I don't see any significant drawback
- second thing is more critical: the [current code](https://github.com/LIFsCode/ELOC-3.0/blob/10d83ee1663e22cf0433ff0c63fb836f47336d3d/eloc610LowPowerPartition/lib/audio_input/src/I2SMEMSSampler.cpp#L280) only removes every e.g. 2nd or 3rd sample but does not do any filtering. This adds aliasing artifacts to the downsampled data which can have severe disturbance in the signals spectrum. @OOHehir I don't know about the robustness of the AI model concerning this, but I strongly suggest adding at least a Moving average filter or better any kind of FIR or IIR filter.

OOHehir, 2024-03-10:
@LIFsCode @EDsteve You're quite right, I know we discussed something similar before with regards to dynamically changing the volume, I should have handled this with the inference also. From my understanding, the AI models looks at the overall 'shape' of the sound sample & should be relatively robust to variations, but it's worth trying to verify this effect. 

What I can suggest is to implement a test that will run a target sound recorded (e.g., trumpet) through the model at its expected sample rate & record the results. By then, comparing this result against results from lower sample rates, we can guage the effect. (There is something similar now in main.cpp that I was planning to remove to a unit test.)

EDsteve, 2024-04-20:
@OOHehir @LIFsCode 
Sorry for my late reply to this topic. Somehow it slipped my mind.

1. Yes. let's keep it as is. Sample rates from 4, 8, 12, 16, 20, ... 44KHz should work fine.

2. Maybe that is the reason why i can't get a model working on the ELOC. I have made models with Edge Impulse which run on my phone or computer and can recognize trumpets correctly. But for now none of these models work on the ELOC. @OOHehir any update on your test? My focus right now is the AI. I hope we can have one for the next batch of ELOCs :)

OOHehir, 2024-04-20:
@EDsteve I have a test setup in place (test/test_target_edge-impulse). It runs a sample sound through the model & reports results, its similar to what was in main.cpp. Building on this we can run different sample rates of  the same audio through the model & compare results. So for example if you have a model that uses 4Khz sample rate over 1 sec we can take the same 1 sec audio sample &:

- sample it at 16kHz, pick every 4th sample & run it through the model
- sample it at 8kHz, pick every 2nd sample & run it through the model
- sample it at 4kHz, pick every sample & run it through the model
By comparing the results we can see what effect this has. It won't be conclusive but should be useful to indicate the effect.

Can you provide the audio samples? We would need the target sound (e.g. trumpet) & one 'other' sound sampled at 18, 8 & 4kHz for whatever length the model expects (1 sec?).  I think your using audacity which should be able to handle the resampling. 

Also if you have the best candidate model I can integrate it into the main branch.

OOHehir, 2024-04-22:
@EDsteve Thanks, I'm surprised Audacity doesn't do a good job, it seems like a mature application so strange this hasn't been caught before. [VLC](www.videolan.org) might be another option to convert sample rates, I think there is also a command line option in Linux, I'll take a look.

With regards the WAV -> C array I think a better approach will just to use a WAV reader library & have the ESP32 read the data in. This way the wav files can be changed in a straightforward manner in the future as testing continues. I think there should already be such libraries available, I take a look. I can structure the project to only pull this library in for the testing, so won't effect the main application.

I'll let you know when I have something.

OOHehir, 2024-04-24:
@EDsteve Well, it turns out there wasn't really a decent library to read wav files. Most of them were more focused on streaming & playing the sound over I2S.

 I found some code & fleshed it out to get the number of samples in the wav file etc, it's included now in the 'wav_file' library. Using this hopefully I'll be able run the sounds through the model over the next few days.

I used [sox](https://opensource.com/article/20/2/linux-sox) to convert the 16k to 4k sounds, you might check to see if the quality is decent. I imagine given the sample rate it's inevitable to lose some of the higher frequencies.

[16bit_16Kand8Kand4k_1sec.zip](https://github.com/LIFsCode/ELOC-3.0/files/15100110/16bit_16Kand8Kand4k_1sec.zip)

EDsteve, 2024-04-26:
@OOHehir 
I just checked the 4K sound file and it seems to have the same strange edge as the Audacity version. Also when i listen to it, it sounds like clipping/distorted? I know the quality of the sound will be much less with 4K. But the clipping sound shouldn't be there i think.
Not sure what kind of impact that has on the AI though. 

I see you have added a way to run inference on WAV files on the ELOC. That is awesome. Looking forward to test this with a few models. That will be an important step.
I will also train some models using 16K sound files. That way we can avoid the unknown things that might happening during downsampling.

Here the spectrogram from 4K (SOX), 8K and 16K
[image]

EDsteve, 2024-04-26:
@OOHehir I just realized that i can downsample the wav files using EI. And download it from there. Worked well.
Also i was listening to your 4K samples again and somehow there is no distortion any more. Before i could clearly hear it. I listened to it a few times actually. Now it's gone. No idea why. Must be something on my setup.

Anyway. Here is the complete sound set without strange edges:
[16bit_16K-8K-4K_Test-Sounds.zip](https://github.com/LIFsCode/ELOC-3.0/files/15131915/16bit_16K-8K-4K_Test-Sounds.zip)

OOHehir, 2024-04-27:
@EDsteve Thanks for that.

I've run some tests (after correcting some bugs) & am using wav files I've copied across to the SD card, listed [here](https://github.com/LIFsCode/ELOC-3.0/blob/cc25c5de8a8c6d73239a6eb184e1b1217dc8b11f/eloc610LowPowerPartition/test/test_target_ai_model/test.cpp#L265). Eventually I plan that the code will be able to scan the /sdcard/wav_test_files/ directory & run any wav files found through the AI model, but for the moment the wav files are hard coded.

To run the test you need to have the board ready to be flashed & then select:

target_unit_selected_tests > Advanced > Verbose Test

It'll build the code, flash & display the results.

I got surprisingly good correlation across the 3 different sample rates, almost exactly the same.

EDsteve, 2024-04-28:
@OOHehir Great to hear that the results are very similar. And it does recognize the trumpet as trumpet?
I tried to test it as well but getting an upload error. I guess it does not find my COM port. Not sure how to fix that. If you have an idea. Let me know.

FYI: The models i am training mostly have a window size of 2 seconds. If the wav file for testing has only 1 second. Not a problem? What if the wav file has 10 seconds?

OOHehir, 2024-04-28:
@EDsteve @LIFsCode [This script](eloc610LowPowerPartition/tools/setUploadMonitorPort.py) now runs at the build stage & sets the upload & monitor port depending on the OS. It simplifies the 'Project Task' list & reduces the number of tasks. It should address your issue ED.

With regards to the models yes, they appear to produce the same results but I'll have to do more work to very. They detected the trumpet sound with over 95%.

The way that code works is simply to start at the beginning of the wav file & feed in samples until the ai model buffer is full, then run the inference. I'd suggest you make sure:
1. the file your using has the target sound in the model window size (after that the file contents are ignored)
2. the file your using is at least as long as the model window size, otherwise I'm sure an error will result.

EDsteve, 2024-04-28:
@OOHehir 
Thanks a lot. Upload works now. But i am getting Guru error with reboot.
Here the log: [Guru.txt](https://github.com/LIFsCode/ELOC-3.0/files/15142965/Guru.txt)

Does this WAV-test implementation change a lot of code? I don't really understand it. Haha. But we should keep the master branch tidy and stable since we will deploy a new release in three weeks. But i am sure you have it under control :)

Just for your information:
@LIFsCode will be busy with his job and won't have much time for the ELOC for the next weeks (hopefully not months).

OOHehir, 2024-04-29:
@EDsteve Its not immediately obvious whats causing the error, do you have the three samples files on the SD card? Its possible its trying to load a wav file that doesn't exist. I still have to improve the functionality to dynamically load in wav files, hopefully have it in a day or two.

This wav test doesn't change the production code significantly, most of it is in /test/test_target_ai_model/test.cpp. I need to extend the functionality of some of the libraries (e.g. ffsutil for the wav file list) but I'll try & keep this to a minimum.

With regards the production code I don't think there's any big changes required for the next deployment so should be OK.

EDsteve, 2024-04-29:
@OOHehir In my understanding the three files should be in the folder /wav_test_files of the SD card:
[image]
I am using the model which is on the github master branch right now. That should work right?

But i will just wait for you to improve it before we start debugging my issues which are probably caused by me somehow :)

EDsteve, 2024-05-02:
I can't make it work. Always results in the Guru error.
INFO: If i remove the folder wav_test_files. Then i get the same Guru error.

@OOHehir Something i can try? Or maybe you want to connect to my PC using Anydesk?

OOHehir, 2024-05-02:
@EDsteve Do you mind if we keep this under the discussion, #99, its moving off this topic.