LIFsCode/ELOC-3.0 #93: Watchdog error since a few commits

EDsteve, 2024-04-06:
@OOHehir @LIFsCode 
I have seen this twice already. Only recently (if i would have to guess: commits after 12th of March).

### What and How it happens:
boot ELOC, connect with app, start recording,.. all looks good for a while but the i gets this repeated error in serial log:
```
E (9504546) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9504546) task_wdt:  - IDLE (CPU 0)
E (9504546) task_wdt: Tasks currently running:
E (9504546) task_wdt: CPU 0: I2S read
E (9504546) task_wdt: CPU 1: IDLE
E (9504546) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E2266:0x3FFBE590 0x40086195:0x3FFBE5C0 0x4010FC96:0x3FFF48B0 0x40112F7D:0x3FFF48D0 0x4011BCCB:0x3FFF4920 0x4011BF75:0x3FFF4970 0x400F988C:0x3FFF49A0 0x400F98BA:0x3FFF49C0 0x400F9736:0x3FFF49E0 0x400F5345:0x3FFF4A00 0x400F7AB2:0x3FFF4A40 0x400FA7F4:0x3FFF4AA0 0x400FA8CD:0x3FFF4C00 0x4011E641:0x3FFF4C20 0x401C8C09:0x3FFF4C40 0x400D3FDA:0x3FFF4C60 0x400D8500:0x3FFF4CC0 0x400D8B1B:0x3FFF4D50 0x400D447F:0x3FFF4D80 0x401E2B35:0x3FFF4DB0 0x4009CBD5:0x3FFF4DE0 0x400D9853:0x3FFF4E30 0x400D99B5:0x3FFF4E80

  #0  0x401E2266:0x3FFBE590 in task_wdt_isr at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\esp_system/task_wdt.c:183 (discriminator 3)
  #1  0x40086195:0x3FFBE5C0 in _xt_lowint1 at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\freertos\port\xtensa/xtensa_vectors.S:1114
  #2  0x4010FC96:0x3FFF48B0 in sdmmc_host_start_command at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\driver/sdmmc_host.c:246 (discriminator 1)
  #3  0x40112F7D:0x3FFF48D0 in sdmmc_host_do_transaction at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\driver/sdmmc_transaction.c:148
  #4  0x4011BCCB:0x3FFF4920 in sdmmc_send_cmd at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\sdmmc/sdmmc_cmd.c:23
      (inlined by) sdmmc_send_cmd_send_status at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\sdmmc/sdmmc_cmd.c:345
  #5  0x4011BF75:0x3FFF4970 in sdmmc_get_status at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\sdmmc/sdmmc_cmd.c:507
  #6  0x400F988C:0x3FFF49A0 in ff_sdmmc_card_available at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\diskio/diskio_sdmmc.c:23
  #7  0x400F98BA:0x3FFF49C0 in ff_sdmmc_initialize at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\diskio/diskio_sdmmc.c:38
  #8  0x400F9736:0x3FFF49E0 in ff_disk_initialize at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\diskio/diskio.c:62
  #9  0x400F5345:0x3FFF4A00 in find_volume at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\src/ff.c:3258  
  #10 0x400F7AB2:0x3FFF4A40 in f_stat at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\src/ff.c:4610       
  #11 0x400FA7F4:0x3FFF4AA0 in vfs_fat_stat at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\vfs/vfs_fat.c:597
  #12 0x400FA8CD:0x3FFF4C00 in vfs_fat_stat at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\fatfs\vfs/vfs_fat.c:590
  #13 0x4011E641:0x3FFF4C20 in esp_vfs_stat at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\vfs/vfs.c:580 (discriminator 3)
  #14 0x401C8C09:0x3FFF4C40 in stat at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/syscalls/sysstat.c:11
  #15 0x400D3FDA:0x3FFF4C60 in ffsutil::getFileSize(char const*) at lib/utils/src/ffsutils.cpp:46
  #16 0x400D8500:0x3FFF4CC0 in RotateFile::needsRotate() const at lib/utils/src/RotateFile.cpp:97
      (inlined by) RotateFile::needsRotate() const at lib/utils/src/RotateFile.cpp:96
      (inlined by) RotateFile::rotate() at lib/utils/src/RotateFile.cpp:108
  #17 0x400D8B1B:0x3FFF4D50 in RotateFile::vprintf(char const*, __va_list_tag) at lib/utils/src/RotateFile.cpp:236
  #18 0x400D447F:0x3FFF4D80 in Logging::_log_vprintf(char const*, __va_list_tag) at lib/utils/src/logging.cpp:55
  #19 0x401E2B35:0x3FFF4DB0 in esp_log_writev at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\log/log.c:200     
  #20 0x4009CBD5:0x3FFF4DE0 in esp_log_write at C:\Users\edste\.platformio\packages\framework-espidf@3.40404.0\components\log/log.c:210      
  #21 0x400D9853:0x3FFF4E30 in I2SMEMSSampler::read() at lib/audio_input/src/I2SMEMSSampler.cpp:342 (discriminator 3)
  #22 0x400D99B5:0x3FFF4E80 in I2SMEMSSampler::start_read_thread() at lib/audio_input/src/I2SMEMSSampler.cpp:405
      (inlined by) I2SMEMSSampler::start_read_thread_wrapper(void*) at lib/audio_input/src/I2SMEMSSampler.cpp:416

E (9504546) task_wdt: Print CPU 1 backtrace`
```
The knock sensor still works and responds in serial monitor. But i can't connect with the app any more. 

### Interesting:
When i reboot the ELOC from theis stage using the EN push button. It can't mount the SD cards:
```
I (7117) SDC: Initializing SD card
E (8120) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (8168) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109
E (8168) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
E (8170) SDC: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
```

But then i restarted with On/OFF. I can bood nofrmally. And it did recognize the SD card. Coincidence or not? Not sure.

Let me know if i should test something specific to find out more.

OOHehir, 2024-04-09:
@EDsteve Seems the SD became 'unavailable' somehow & the resulting blockage caused the watchdog to trigger.

Is the SD old? Could be bad sectors?

EDsteve, 2024-04-10:
@OOHehir 
Okay. So that seems to be most probably hardware related if i understand you correctly. Hmm. All SD cards i use for testing are only a few months old. Using different brands (Lexar, Samsung and SanDisk). Not sure any more which one i was using when this error occured. I guess we can just hope it won't happen again. But for sure will pay attention if i encounter this again.

EDsteve, 2024-04-20:
I will close this now untill it happens again .)

closed 2024-04-20 by EDsteve