LIFsCode/ELOC-3.0 #95: Adding new Edge Impulse model builds with error

EDsteve, 2024-04-19:
@OOHehir 
I have made several models over the past days. They can predict trumpets from my phone and laptop so i want to try them on the ELOC
Unfortuntately i get compiling errors. 
I did full clean between the model change. Can you take a look?

This is the model i try to use (select.cpp already modified): [ei-trumpet-trimmed-arduino-1.0.68-modified.zip](https://github.com/LIFsCode/ELOC-3.0/files/15033420/ei-trumpet-trimmed-arduino-1.0.68-modified.zip)

Here the full compiling log: [new_EI_model_error.txt](https://github.com/LIFsCode/ELOC-3.0/files/15033425/new_EI_model_error.txt)

This is the error i am getting:
```
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_bitreversal.c.o
In file included from lib/edge-impulse/src/edge-impulse-sdk/classifier/ei_run_dsp.h:25,
                 from lib/edge-impulse/src/edge-impulse-sdk/classifier/ei_run_classifier.h:24,
                 from lib/edge-impulse/src/trumpet_trimmed_inferencing.h:49,
                 from lib/edge-impulse/src/EdgeImpulse.cpp:10:
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h: In member function 'virtual int flatten_class::extract(ei::signal_t*, ei::matrix_t*, void*, float)':
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h:183:31: error: 'mean' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             means[axis][head] = mean;
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h:90:19: note: 'mean' was declared here
             float mean; // to use with moving average
                   ^~~~
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_bitreversal2.c.o   
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_bitreversal_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_f64.c.o       
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_init_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_init_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_init_f64.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_init_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_init_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_init_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_init_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_init_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_init_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_init_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_init_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_init_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_init_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix8_f16.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix8_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_init_f32.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_init_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_init_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_q15.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_dct4_q31.c.o
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_mfcc_f16.c.o
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h: In function 'EI_IMPULSE_ERROR {anonymous}::process_impulse(ei_impulse_handle_t*, ei::signal_t*, ei_impulse_result_t*, bool)':
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h:183:31: error: 'mean' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             means[axis][head] = mean;
lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h:90:19: note: 'mean' was declared here
             float mean; // to use with moving average
                   ^~~~
Compiling .pio\build\esp32dev-ei-windows\lib882\edge-impulse\edge-impulse-sdk\CMSIS\DSP\Source\TransformFunctions\arm_mfcc_f32.c.o
```

OOHehir, 2024-04-19:
@EDsteve The above commit fixed this build error, I've documented [here](eloc610LowPowerPartition/README-ai.md) also in case you need it.

The model is in a new branch (using model deploy ver).

The model looks bigger than the current model & the buffers won't fit in standard RAM, so to enable the buffers to be stored in PSRAM you need undefine this [line](https://github.com/LIFsCode/ELOC-3.0/blob/646ec400960b52803d044e1c548d1545c672788e/eloc610LowPowerPartition/include/project_config.h#L212)

OOHehir, 2024-04-19:
Please reopen if it doesn't solve the issue

closed 2024-04-19 by OOHehir

EDsteve, 2024-05-25:
Just adding more info.

To avoid this or similar errors find the file the error results from. In the above case it is lib/edge-impulse/src/edge-impulse-sdk/dsp/ei_flatten.h and then change the line
`float mean; `
to
`float mean = 0;`