LIFsCode/ELOC-3.0 #96: BAT LED is always ON during RecordON_DetectOFF

EDsteve, 2024-04-19:
@LIFsCode @OOHehir 
When the ELOC is in RecordON_DetectOFF mode it won't turn off the Bat LED. To save some battery power it would make sense to turn it off. 
That is hopefully a quick fix :)

Cheeers
ED

OOHehir, 2024-05-15:
@EDsteve Just taking a look at this noew, apologies, forgot about it. What is the required functionality of the LEDs in the various recording/ detection modes?

EDsteve, 2024-05-15:
@OOHehir No worries. No worries :) The LEDs are normally LIFsCode corner. But he is not awaylable now. So hope it doesn't take too long for you to find it. If it takes too long or getting too complicated. We can leave it as it is and deal with it later. Not a big issue. :)

To answer you question:
When connected with the app. Both LEDs (Status and Bat LED) are continously on. 
When recording is being started from the app, the Status LED should flash for 30 seconds and then go off. The Bat LED is still On if the app is still connected. But when the app disconnects. The BAT LED should go off as well. But that never happens.

There might be still some logig error in this table, but that's the idea of the status LEDs:
https://github.com/LIFsCode/ELOC-3.0/wiki/Status-LEDs-and-Buzzer

OOHehir, 2024-05-15:
Tested briefly, looks OK on my side.

closed 2024-05-15 by OOHehir