LIFsCode/ELOC-3.0 #104: Regular reboots when PCU running at 240Mhz

OOHehir, 2024-05-19:
Following on from #101 is desirable to temporarily increase CPU clock frequency from 80 to 240 Mhz. Running the CPU at 80 & 160 Mhz appears satisfactory but running it at 240Mhz results in regular reboots. The messages occuring when the reboots are generally fairly conistent, interrupt WDT timeout on either cores:

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Backtrace: 0x40094973:0x3ffbc910 0x40095913:0x3ffbc930 0x400972b1:0x3ffbc950

  #0  0x40094973:0x3ffbc910 in cpu_ll_waiti at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/hal/esp32/include/hal/cpu_ll.h:183
      (inlined by) esp_pm_impl_waiti at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/esp_pm/pm_impl.c:828
  #1  0x40095913:0x3ffbc930 in esp_vApplicationIdleHook at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/esp_system/freertos_hooks.c:63
  #2  0x400972b1:0x3ffbc950 in prvIdleTask at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/tasks.c:3987
```

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Backtrace: 0x40099b00:0x3ffbc8f0 0x40097124:0x3ffbc930 0x400972b4:0x3ffbc950

  #0  0x40099b00:0x3ffbc8f0 in spinlock_acquire at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/esp_hw_support/include/soc/spinlock.h:124
      (inlined by) xPortEnterCriticalTimeout at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/port/xtensa/port.c:301
  #1  0x40097124:0x3ffbc930 in vPortEnterCritical at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) prvGetExpectedIdleTime at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/tasks.c:2468
  #2  0x400972b4:0x3ffbc950 in prvIdleTask at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/tasks.c:4010

```

```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Backtrace: 0x40094973:0x3ffbd080 0x40095913:0x3ffbd0a0 0x400972b1:0x3ffbd0c0

  #0  0x40094973:0x3ffbd080 in cpu_ll_waiti at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/hal/esp32/include/hal/cpu_ll.h:183
      (inlined by) esp_pm_impl_waiti at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/esp_pm/pm_impl.c:828
  #1  0x40095913:0x3ffbd0a0 in esp_vApplicationIdleHook at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/esp_system/freertos_hooks.c:63
  #2  0x400972b1:0x3ffbd0c0 in prvIdleTask at /home/owen/.platformio/packages/framework-espidf@3.40404.0/components/freertos/tasks.c:3987
```

OOHehir, 2024-05-19:
Increase CONFIG_ESP_INT_WDT_TIMEOUT_MS to a large value (3000) does not improve the situation

OOHehir, 2024-05-22:
@EDsteve It seems that the resets at 240Mhz centers around the CPU spending most of its time at idle at sleep. As the CPU is running faster it accomplishes its tasks quicker & hence spends more time asleep. Hence the interrupt WDT doesn't get fed/ reset & times out. This seems to be particularly because CONFIG_FREERTOS_USE_TICKLESS_IDLE was set to y (to enable enhanced power saving). 

I've just created another [branch](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/ai_increase_cpu_freq) . Will you try that? No resets/ crashes on my side so far. Please remember to delete sdkconfig.esp32dev-ei so that it gets rebuilt with the new sdkconfig values.

On another important point, it appears that the max CPU freq is not limited by the config settings:
```
  "config": {
    "secondsPerFile": 60,
    "cpuMaxFrequencyMHZ": 80,
    "cpuMinFrequencyMHZ": 10,
```
From my tests the "cpuMaxFrequencyMHZ" is disgregarded by the CPU & the CPU runs at whatever CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ is set to. I'll have to do more testing to confirm.

However at least you should be able to test models at 240Mhz with the current build.

LIFsCode, 2024-05-23:
@EDsteve @OOHehir 

I'm not too sure but sometimes the esp implementations of freeRTOS get messed up withe the timer configs: sometimes they represent ticks sometimes ms and thus this gets affected by CPU frequency.

CONFIG_FREERTOS_USE_TICKLESS_IDLE can be also disabled using the 

  `"config": {
    "cpuEnableLightSleep": false
}`

Did the problems also arise when setting min & max freq to the same value?

OOHehir, 2024-05-24:
@LIFsCode @EDsteve I've had mixed success changing the CPU freq.

1. When trying to change the CPU frequency (from something different to that defined in the sdkconfig) during boot it appeared to fail
2. When changing the CPU freq during 'routine' program execution I was able to cycle repeatedly from 80Mhz -> 160 -> 240 -> 80 etc without issue. 
3. If CONFIG_FREERTOS_USE_TICKLESS_IDLE=n was set no crashes/ resets were observed. 

I'll have to repeat the above to ensure I hadn't made an error somewhere but perhaps there's some issue changing the CPU freq during boot?

@LIFsCode The [documentation](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/system/power_management.html#configuration) is a little vague about automatic light sleep. It states:

Automatic light sleep is based on FreeRTOS Tickless Idle functionality. If automatic light sleep is requested while the option [CONFIG_FREERTOS_USE_TICKLESS_IDLE](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/kconfig.html#config-freertos-use-tickless-idle) is not enabled in menuconfig, [esp_pm_configure()](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/system/power_management.html#_CPPv416esp_pm_configurePKv) will return the error ESP_ERR_NOT_SUPPORTED.

So automatic light sleep is **only** available when CONFIG_FREERTOS_USE_TICKLESS_IDLE=y ?

_Did the problems also arise when setting min & max freq to the same value?_

I haven't tried this yet, will hopefully get a chance over the weekend.

EDsteve, 2024-07-02:
@OOHehir A restart after changing the frequency is totaly acceptable if that makes things easier :)
(My test results above are with a restart in between each change)

OOHehir, 2024-07-11:
@EDsteve  I made a few changes to investigate this. sdkconfig.defaults now contains only the changes to the default sdkconfig. I checked this a few times & fairly sure I captured all the custom chnages to sdkconfig that we're using. This file gets overlaid onto the sdkconfig during the build process. Just having our changes makes it much clearer what has been customised.

Please note the warning in the file:
```
# WARNING: If any changes are made to this file the other sdkconfig
#          files must be deleted to ensure changes are used
```
I created a new [branch](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/cpu_240) which changes the CPU frequency to 240Mhz. Disabling power management (CONFIG_PM_ENABLE=n) in conjunction with this appears to produce a stable build, I only ran the ELOC for a few minutes but didn't notice any reboots. Not ideal to lose the power management but if the AI feature is running at 240Mhz I don't think will be much to be gained from having the PM enabled.

It should be possible to try enabling the PM again & perhaps one-by-one disabling the PM configs (below) & see if it's possible to isolate if one in particular is causing the issue.
```
CONFIG_PM_SLP_IRAM_OPT=y
CONFIG_PM_RTOS_IDLE_OPT=y
CONFIG_PM_SLP_DISABLE_GPIO=y
```

EDsteve, 2024-07-11:
@OOHehir Nice solution with the sdconfig. Makes things much easier :)

Will do all the tests and report back when i return from my trip (not before the 18th unfortunately).

EDsteve, 2024-07-22:
@OOHehir I have tested all the PM settings as mentioned above. Unfortunately without much luck.

**Procedure:**
I used [this branch](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/cpu_240)
- modified the PM parameters
- removed all sdkconfigs besides default
- full clean
- build
- upload
- run the ELOC with RecordON_DetectON

CPU frequency for all tests was set to min 10MHz and max 240MHz

**Result:**
I have not tested every combination. If you think it could make a difference. Let me know and i am happy to go through different combinations.

| Config | Settings | RecordON_DetectON |
|--------|----------|-----------------------------|
| 1 | CONFIG_PM_ENABLE=n<br>CONFIG_PM_SLP_IRAM_OPT=y<br>CONFIG_PM_RTOS_IDLE_OPT=y<br>CONFIG_PM_SLP_DISABLE_GPIO=y | Runs without problems |
| 2 | CONFIG_PM_ENABLE=y<br>CONFIG_PM_SLP_IRAM_OPT=y<br>CONFIG_PM_RTOS_IDLE_OPT=y<br>CONFIG_PM_SLP_DISABLE_GPIO=y | Reboot after few minutes |
| 3 | CONFIG_PM_ENABLE=y<br>CONFIG_PM_SLP_IRAM_OPT=n<br>CONFIG_PM_RTOS_IDLE_OPT=y<br>CONFIG_PM_SLP_DISABLE_GPIO=y | Reboot after few minutes |
| 4 | CONFIG_PM_ENABLE=y<br>CONFIG_PM_SLP_IRAM_OPT=n<br>CONFIG_PM_RTOS_IDLE_OPT=n<br>CONFIG_PM_SLP_DISABLE_GPIO=y | Reboot after few minutes |
| 5 | CONFIG_PM_ENABLE=y<br>CONFIG_PM_SLP_IRAM_OPT=n<br>CONFIG_PM_RTOS_IDLE_OPT=n<br>CONFIG_PM_SLP_DISABLE_GPIO=n | Reboot after few minutes |

Example reboot log of Config 2: [Config2.txt](https://github.com/user-attachments/files/16332781/Config2.txt)

OOHehir, 2024-08-06:
@EDsteve Finally got a few minutes to look at this again, using master ee641e63fc5576bad173dbc3be8c5f048bceccb9.

The only changes I did to above was:
sdkconfig.defaults:
```
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
```
ElocSystem.cpp:
```
    esp_pm_config_esp32_t cfg = {
        .max_freq_mhz = 240, // max_freq,
        .min_freq_mhz = 240, // getConfig().cpuMinFrequencyMHZ,
        .light_sleep_enable = getConfig().cpuEnableLightSleep
    };
```

Its run for 20+ min now without issue. Would you like to try on your side?

The power consumption at 240Mhz with recording & detecting active:

[image]

EDsteve, 2024-08-06:
@OOHehir Thanks for testing in your busy times.

**Frequency**
Maybe you missed one of my [previous posts](https://github.com/LIFsCode/ELOC-3.0/issues/104#issuecomment-2200594160)? I have done similar tests with this result:

Ai and recording runs without issues using the following ELOC config:
1. Min 240MHz, Max 240MHz
2. Min 10MHz, Max 160MHz

**Power consumption**
125mA is a big number :)
If i remember correctly. I have tested the power consumption a while ago (in Sumatra) with the second setting (min10, max160) and it was way below 100mA. Of course that also depends on the model and i can't remember any more which one i used.

OOHehir, 2024-08-08:
ED,

Apologies, forgot about that. Strange how those combinations work.

Yes, the power consumption is certainly significant,  not sure if it'll be
realistic for you to run at those levels. Might get a chance to start
looking at the circular buffer this weekend.

Regards,