LIFsCode/ELOC-3.0 #107: RecordOnEvent doesn't record the event

EDsteve, 2024-07-01:
Hi @OOHehir @LIFsCode,

Hope you are doing fine. I know that both of you are busy lately and frankly i didn't have much time either for the ELOC the last weeks. But i am back :) And since we have the AI running on the ELOC we should also retrain the AI with false positive sounds to improve it. If you find time to look into this issue. Would be great. No time pressure though :)

**Problem:**
RecordOnEvent starts recording AFTER the triggered sound. So the trumpet is not on the recorded wav file. Is possible to save the sound which actually triggered the recording? That would be helpful.
I tested with commit 92fd571c95e073c87770c4a5d6384af422614b18

If both of you are too busy. Maybe an idea where to look at?

Cheeers
ED

OOHehir, 2024-07-02:
Hello @EDsteve, hope you're enjoying been at home, especially while all the football is on!

I think capturing the actual target sound will involve some compromise. I think the only way to do so so will be to continuously store sound into the wav buffer & once the AI threshold has been exceeded to start writing to the SD card. To capture any significant length of sound (i.e. greater than 1 sec) will require a significantly sized buffer, & probably use of PSRAM will the resulting higher power consumption. Also because the ESP32 will need to write out the previously captured sound (say 3 or 4 secs worth) & keep up with storing the current sound without overwriting the buffer the situation is slightly complicated. Currently two buffers are used (1 to store incoming sound whilst the other is been written to the SD card), but this setup may require 3 or more large sized buffers, or a single large ring buffer (which is perhaps the optimum solution). @LIFsCode had started work on a factory setup for the EI & WAV functionality, but if a large code change isn't an issue I can start looking at this over the coming days/ weeks.

EDsteve, 2024-07-02:
@OOHehir Thanks for clarifying. I didn't realize it would be a bigger job.
For the next weeks/months the ELOCs in the jungle will use recordON_detectON. So it's not urgent at all and we can put this enhancement on hold for now.