LIFsCode/ELOC-3.0 #108: External Edge Impulse model - compiling errors

EDsteve, 2024-08-08:
@OOHehir @MWieland

Owen, i am sure you remember that Shyam tried to make a model last year which should be imported to EdgeImpulse ([BYOM](https://docs.edgeimpulse.com/docs/edge-impulse-studio/bring-your-own-model-byom)) and then use EdgeImpulse to deploy it for the ESP32, so we can avoid the paywall in Edge Impulse. But he was never sucessful.
After some long nights our friend Marc made it work, which wasn't a piece of cake. Thanks again Marc. Lot's of rocks in the way. But now we have a pipeline, which produces a model that can be imported to Edge Impulse. :)

The next step is to bring it to the ELOC. So i deployed Marc's model using Edge Impulse, added it to our firmware, but i can't compile it without errors.
Marc made three models with different window sizes (1, 3 and 5 seconds). All models have three features (background, rumble, trumpet). I tested two models with our firmware.

### My testing procedure:

1. Deploy the model using EON Compiler in Edge Impulse
2. Unzip the model
3. Remove the old model from the firmware folder and copy the `src` folder of the new model to the firmware
4. Adjust [select.cpp](https://github.com/LIFsCode/ELOC-3.0/blob/master/eloc610LowPowerPartition/README-ai.md), EdgeImpulse.cpp (line 10) and [flatten.h](https://github.com/LIFsCode/ELOC-3.0/discussions/99#discussioncomment-9308705) accordingly 
5. Full Clean in PlatformIO
6. Build

### One second window size model
Download the model here: [ei-byom-arduino-1.0.8_1sec.zip](https://github.com/user-attachments/files/16549749/ei-byom-arduino-1.0.8_1sec.zip)

Last lines of the compiler log:
```
Linking .pio\build\esp32dev-ei\firmware.elf
c:/users/ed/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev-ei/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/ed/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/ed/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.      
c:/users/ed/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 31424 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev-ei\firmware.elf] Error 1
```

DRAM too small it seems. Not sure if we can use PSRAM? Or is that too slow?
According to this screenshot from Edge Impulse. The imported model uses 66KB RAM and a bit over 200KB Flash (without being optimized)

[image]

### Three second window size model
Download the model here: [ei-byom-arduino-1.0.6_3sec.zip](https://github.com/user-attachments/files/16550427/ei-byom-arduino-1.0.6_3sec.zip)

Last lines of the compiler log:
```
lib/edge-impulse/src/edge-impulse-sdk/dsp/../porting/ei_classifier_porting.h:285: warning: "EI_PORTING_ARDUINO" redefined
 #define EI_PORTING_ARDUINO        0

lib/edge-impulse/src/edge-impulse-sdk/dsp/../porting/ei_classifier_porting.h:262: note: this is the location of the previous definition
 #define EI_PORTING_ARDUINO      1

src/main.cpp: In function 'void ei_callback_func()':
src/main.cpp:509:40: warning: missing initializer for member 'ei_impulse_result_t::bounding_boxes_count' [-Wmissing-field-initializers]
         ei_impulse_result_t result = {0};
                                        ^
src/main.cpp:509:40: warning: missing initializer for member 'ei_impulse_result_t::classification' [-Wmissing-field-initializers]
src/main.cpp:509:40: warning: missing initializer for member 'ei_impulse_result_t::anomaly' [-Wmissing-field-initializers]
src/main.cpp:509:40: warning: missing initializer for member 'ei_impulse_result_t::timing' [-Wmissing-field-initializers]
src/main.cpp:509:40: warning: missing initializer for member 'ei_impulse_result_t::copy_output' [-Wmissing-field-initializers]
src/main.cpp: In function 'void app_main()':
src/main.cpp:724:55: error: static assertion failed: TEST_SAMPLE_LENGTH must be at least equal to EI_CLASSIFIER_RAW_SAMPLE_COUNT
         static_assert((EI_CLASSIFIER_RAW_SAMPLE_COUNT <= TEST_SAMPLE_LENGTH),
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:734:40: warning: missing initializer for member 'ei_impulse_result_t::bounding_boxes_count' [-Wmissing-field-initializers]
         ei_impulse_result_t result = {0};
                                        ^
src/main.cpp:734:40: warning: missing initializer for member 'ei_impulse_result_t::classification' [-Wmissing-field-initializers]
src/main.cpp:734:40: warning: missing initializer for member 'ei_impulse_result_t::anomaly' [-Wmissing-field-initializers]
src/main.cpp:734:40: warning: missing initializer for member 'ei_impulse_result_t::timing' [-Wmissing-field-initializers]
src/main.cpp:734:40: warning: missing initializer for member 'ei_impulse_result_t::copy_output' [-Wmissing-field-initializers]
In file included from lib/audio_input/src/I2SMEMSSampler.h:4,
                 from src/main.cpp:14:
lib/wav_file/src/WAVFileWriter.h: At global scope:
lib/wav_file/src/WAVFileWriter.h:24:23: warning: 'wav_static_buffers' defined but not used [-Wunused-variable]
   static signed short wav_static_buffers[2][wav_static_buffer_size];
                       ^~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32dev-ei\asio\asio\asio\src\asio.o
*** [.pio\build\esp32dev-ei\src\main.o] Error 1
```
Full log: [3_sec_all.txt](https://github.com/user-attachments/files/16550191/3_sec_all.txt)
And also for the three second model the screenshot from Edge Impulse after the import:
[image]

@OOHehir August is the busy month for family. So no pressure. But if you have a free minute and feel like it. Maybe you can take a look. Hopefully an easy fix on the firmware side. Or if something need to change on the model architecture?

I have the feeling we are very close to make an external model work, which would give us much more freedom.

OOHehir, 2024-08-09:
@MWieland Welcome to the project! Great work!

@EDsteve  I haven't had a chance to look at this properly but a quick response:

1. I think we've come across this error before (#79), you might try commenting out the line:
```
-DEI_CLASSIFIER_ALLOCATION_STATIC=1
```
In platform.ini.

2. I assume your using quite a long sample length time in the model? The code starting at line 716 in main.cpp runs a test sample through the EI inference process as a test at startup. It seems that this test sample length is shorter than the model input & hence the error. You could, just to get the code compiling, comment out lines 716 to 783.

I should be able to look at this tomorrow for a while.

EDsteve, 2024-08-09:
@OOHehir @MWieland
You have a good memory Owen. I have already forgotten about solution for problem number one. Sorry.
But now both problems are solved. That was pretty good for a quick look.  👍  :)
I think that was the last AI-pipeline milestone to solve. Your sleepless nights were worth it, Marc :)

@OOHehir
Not sure how much work it is to adjust the testing at boot to the models sample length. But it's not high priority. So no rush :)

MWieland, 2024-08-10:
That sounds great so far! Thanks for the quick and efficient reply :+1:  With the current setup, we would be fully flexible regarding sample length from the model side as now we package the whole audio processing part directly into the tflite model. We had better initial results with longer samples, but can adapt to hardware requirements. I guess then it is time now to actually work on improving the model accuracy.

OOHehir, 2024-08-11:
@EDsteve No problem, I'll close this issue with this comment. Please re-open if necessary.

@EDsteve @MWieland The test samples I created are [here](https://github.com/LIFsCode/ELOC-3.0/blob/master/eloc610LowPowerPartition/include/test_samples.h). The file consists of two test samples, one the target sound & the other background noise, etc. They're useful at verifying that the code, at least to some degree, is working correctly. I exported the test samples from Edge Impulse, the length is sufficient to fill the TensorFlow buffer (i.e 16000 samples = 1 sec at the samples rate of 16000 samples/ sec for a 1 sec buffer).

@EDsteve You may wish to remind @MWieland how the crude downsampling works in the code, it's discussed here (#99) & can have a significant impact on model performance.

closed 2024-08-11 by OOHehir