LIFsCode/ELOC-3.0 #109: Modify behaviour of file header in file names

EDsteve, 2024-11-27:
@OOHehir 
When recording. The file name consist of file header, sessionID and date of first recording. (see [WAVFileWriter.cpp](https://github.com/LIFsCode/ELOC-3.0/blob/master/eloc610LowPowerPartition/lib/wav_file/src/WAVFileWriter.cpp))
Two thing should be changed:

**Underscore after file header:**
This is how it looks right now when using Grid02 as file header:
`Grid021732700450700_2024-11-27_16-40-52.wav`

It would make things much easier on our side to add a underscore between file header and session ID. So it should look like this:
`Grid02_1732700450700_2024-11-27_16-40-52.wav`

**No file header if file header is set to "not_set":**
If "not_set" is set as file header. The filename should not have a file header and should look like this: `1732700450700_2024-11-27_16-40-52.wav`

OOHehir, 2024-12-04:
@EDsteve  Can you try this please

EDsteve, 2024-12-07:
@OOHehir Thanks so much. I have tested it and it works great. 
A restart of the ELOC is required after a change of the file header. Is there an easy way to make it work without restart?
If it is too complex. We can use it as is. No problem.

OOHehir, 2024-12-13:
@EDsteve When you say updating the session folder do you mean that the location should change, the time or both?

EDsteve, 2024-12-14:
@OOHehir 
I think i wasn't really clear of what i meant :) My last post was about the file header (the beginning of the file name of each recording).
Hope that will explain it better:
When i go to the ELOC app and change the file header from "not_set" to "Grid02" for example. Then the app sends the information of the new file header to the ELOC device. But when i start recording it still uses "not_set". 
I have to restart the ELOC. Only then Grid02 is used as file header for file names.

OOHehir, 2024-12-20:
@EDsteve Say for your example, changing from  "not_set" to "Grid02", should the remained of the file header (e.g. 1732700450700_2024-11-27_16-40-52) also change or be retained?

EDsteve, 2024-12-21:
@OOHehir 
Not 100% sure what you mean. Hope we are talking about the same thing :)
The remaining part of the file name (1732700450700_2024-11-27_16-40-52) should not be influenced/changed if the file header changes. The file header only changes the front part of the file name. See:
[image]

OOHehir, 2025-01-08:
@EDsteve Looking at the code it seems like the folder is created on the first time that the I2S sampling occurs (i.e. inference or recording). After this point it would get a bit messy to change the folder so unless it becomes particularly important we'll leave this as is?

closed 2025-01-08 by OOHehir

EDsteve, 2025-01-09:
I just tested this again. And it all works as it should be. You implemented it as intended. 
Not sure why i had to restart the ELOC last time.