LIFsCode/ELOC-3.0 #110: Add gain control to setconfig

EDsteve, 2024-11-29:
@OOHehir 
No pressure. I am just adding things which needs to be implemented. Sooner or later :)

Adding the gain control into the JSON setconfig file as [discussed here](https://github.com/LIFsCode/ELOC-3.0/issues/46#issuecomment-1830431158) would be great. So we can easily adjust it from the app.

OOHehir, 2024-12-20:
@EDsteve I'll take a look at this over the coming days. I've a feeling this will introduce bugs during testing so I'll create another branch to try it out

EDsteve, 2024-12-21:
Okay. Yeah. Better play safe :)
I hope it's not too much effort to add it.

OOHehir, 2024-12-28:
@EDsteve @LIFsCode Along with implementing this should I clean up the setconfig by removing:

- listenOnly - marked in code as unused
- MicBitShift - passed into the I2S function but unused

I assume by changing this it might break the functionality from the Android app?

LIFsCode, 2024-12-28:
@EDsteve @OOHehir 
From the ESP32 source code this should be irrelevant if the app sets this parameter or not. But I don't know if the App requires certain fields in the response:
@josephx86  Maybe you can answer this

OOHehir, 2025-01-01:
@EDsteve @LIFsCode Just made the changes to [this branch](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/elocConfig)

As the commit says it removes redundant items from the config:
- bitShift
- fixSPH0645

Implements
- volume gain

I've done some brief tests & seems OK. 

@EDsteve To test you may need to erase the ESP32 first, to ensure the config in SPIFFS is removed. As the volume gain is not available via Bluetooth you may need to edit the code or modify the config on the SD card to test it. I2S_DEFAULT_CHANNEL_FORMAT_RIGHT is set in project_config.h, for my board to work I had to change it to I2S_DEFAULT_CHANNEL_FORMAT_LEFT.

@LIFsCode If there's no significant issues I'll merge the changes into master/main the next few days.

josephx86, 2025-01-02:
> @EDsteve @OOHehir From the ESP32 source code this should be irrelevant if the app sets this parameter or not. But I don't know if the App requires certain fields in the response: @josephx86 Maybe you can answer this

It is okay to make name changes, add/remove items. The only requirement is that the names should be compliant with the JSON standard, otherwise the app's JSON parser with fail.

EDsteve, 2025-01-03:
@OOHehir 
I have tested https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/elocConfig and the gain works fine.
You have also removed the MicUseTimingFix. Correct?

@OOHehir @josephx86
The changes introduced some communication problems with the app. Changing settings (e.g. recording time) takes forever and sometimes there is missing status information in the status page of the app. What is the best way to find out about it? 

This is how the JSON config for the microphone looks right now:

```
  "mic": {
    "MicType": "ns",
    "MicVolume2_pwr": -4,
    "MicSampleRate": 20000,
    "MicUseAPLL": true,
    "MicChannel": "Right"
  }
```

josephx86, 2025-01-03:
The best way to find out about any changes is to use the command line option to manually run commands.  

I can start patching the app's code right away. I do need a binary file of the firmware that I can flash on to my board to use during tests

EDsteve, 2025-01-03:
@josephx86
That's great. Thanks. This is the bin file from the [elocConfig ](https://github.com/LIFsCode/ELOC-3.0/tree/OOHehir/elocConfig)branch:[firmware.zip](https://github.com/user-attachments/files/18301146/firmware.zip)
Rename to firmware.bin

And in case you forgot the [flashing procedure](https://github.com/user-attachments/assets/d1e46250-a124-4353-9073-d77cfb9457a5) :)

OOHehir, 2025-01-07:
@EDsteve @LIFsCode I'll merge this branch unless there's any objections?

OOHehir, 2025-01-08:
@josephx86 One thing that's probably bearing in mind, the MicVolume2_pwr value should be an integer & limited in range from -5 too +5 I think. @EDsteve Will have more user experience on this range.

I'll close this as merge is complete.

closed 2025-01-08 by OOHehir

EDsteve, 2025-01-09:
@OOHehir @josephx86 

With my tests (some time ago):
0 seems already extremely loud
-4 is the same level as the microphone we used before.

So i told joseph to set the range in the app from -8 to 0. Didn't know that it maxes out at 5 and -5. But i will test again to be sure about it :)

Question: Do the values have to be integers or can also be decimals? For example -3.2

OOHehir, 2025-01-09:
@EDsteve It was just a guess from my end so whatever experience you have is more valid.

Yes needs to be a whole/ integer as its basically just ends up as a bit shift (critical for performance reasons).