LIFsCode/ELOC-3.0 #113: Define LoraWAN Config interface

LIFsCode, 2024-12-28:
@josephx86 @EDsteve 

Just to define a first shot for the config interface for the loraWAN part of the ELOC functionality:

### Configuration:
I suggest the following extension of the config (including default values):
```
  "config": {
    "lorawan": {
      "loraEnable": false,
      "upLinkIntervalS": 60,
      "loraRegion": "EU868"
    }
  }
```
With this [default config](https://github.com/user-attachments/files/18268027/elocConfig.json)

### Status Information

1. Any status information required via BT App?
2. Should we include this in getStatus? or in a separate command?
3. How to read the devEUI (derived from the MAC address of the ESP32) and required for TTN?

### Lora Sending information

1. What information should be sent via Lora?
2. Should this be configureable?
3. What Information should be received via Lora?
4. Should received information be configureable?

EDsteve, 2024-12-29:
@LIFsCode  Looks good to me
### Configuration
What is the meaning of `upLinkIntervalS`?
LoRa should only be activated and send data, when an elephant sound was recognized. Or maybe once a day so we know the ELOC is still "alive". Is that what upLinkIntervalS is for?

### Status Information
2. I don't see a downside of using getStatus
3. Not sure what you mean. The devEUI is stored in NVS. Is that not accessible? I also have a list of all devEUIs and keys to register the ELOC on TTN.

### LoRa Sending Information
1. Most importantly: 
   - Date/Time
   - Device Name
   - Detected Event (e.g. Trumpet)
   - Confidence level
   - File Header
2. I can imagine that more data would be handy to have. (e.g. battery level or location accuracy). But more data means also more data loss if the signal is not good enough? So to keep that flexible would be good. Is that a much bigger job to make it configurable? Short answer: Not really necessary. But would be great.
3. Nothing we can think of for now. Future thoughts: Changing settings (setConfig) via LoRa if possible
4. Can not think of anything besides sending setConfig

LIFsCode, 2024-12-29:
@EDsteve 

### Configuration 

Yes, the uplink interval is basically the once a day status sending interval. I strongly recommend not to reduce it below 1/day in the beginning, otherwise we will have no chance to identify when a node went offline, looses Lora connection, etc.

### Status information 

Maybe there's a misunderstanding concerning the devEUI. I adjusted the code, so that the devEUI is created on the ESP itself based on its MAC address as discussed [here](https://github.com/LIFsCode/ELOC-3.0/discussions/112#discussioncomment-11638273).
So it does not use the NVS entry. Sorry that was my fault at the start I was a bit confused about what's the best way to handle it.
It's easy to change it to the NVS entry, but then the NVS entry has to be unique when generated. Using the MAC address solves this as it is already defined to be unique.

Anyway I think it is useful to be able to read the devEUI as it is required to identify a ELOC in the TTN app.

### Lora sending information 

1. Date can be sent as timestamp. Other information like device name and header can be assumed static (unless config changes) hence it would make more sense to store this in a database and match it with the devEUI, instead of sending it via Lora Everytime.