LIFsCode/ELOC-3.0 #116: LoRa: After some time the ELOC can't send message any more

EDsteve, 2025-02-26:
@LIFsCode 
After some amount of messages. Didn't find a pattern yet. But if i send more messages in faster intervals, the error seems to happen faster.  To make it send message again i have to restart the ELOC. 

Log:
```
I (542187) main: Cycles taken to run inference = 97899484
I (542188) main: (DSP: 395 ms., Classification: 12 ms., Anomaly: 0 ms.)
I (542191) main:     background: 0.003906
I (542195) main:     trumpet: 0.996094
I (542200) main: Target sound detected: trumpet
I (542217) LoraWAN: Sending uplink with detected Event
I (542217) LoraWAN: Sending data (16 bytes): type = 1, time = 1740234926, Events=1
E (542277) RADIOLIB_DBG: Error in sendReceive - RADIOLIB_ERR_UPLINK_UNAVAILABLE (-1108)
I (543211) main: Cycles taken to run inference = 98026732
```

[Radiolib ](https://jgromes.github.io/RadioLib/group__status__codes.html)defines this error as:

```
#define | RADIOLIB_ERR_UPLINK_UNAVAILABLE   (-1108)
| User requested to start uplink while still inside RX window or under dutycycle.
```

LIFsCode, 2025-02-26:
@EDsteve ok I get the point, that requires a dead time after an event message. I didn't expect it to trigger so fast.

I'd rather define the dead time a bit higher to increase stability. Would 5 minutes sound good? 
I mean there is no point in notifying about an elephant in shorter windows. Guess it's not the fastest kind of animal ;-)

EDsteve, 2025-02-26:
@LIFsCode 
They run faster than humans, if they want to. But yes. Elephants are normally relaxed travellers :) 
On the other hand we still need to train our model with more elephant sounds and also with false positive sounds to improve it. So of course as more triggers (timestamps) are known, the more we can include into the existing database for training.

5 minutes sounds good for now. If needed, we can always adjust that value i guess?

EDsteve, 2025-02-28:
@LIFsCode 
Correction about more data for AI training:
The elephant detrctions time stamps are getting saved to a CSV file on the SD card as well. 
So we actually don't rely on Lora messages. 

Long story short. 5 minutes is totally fine.